OE4T/meta-tegra #1261: gstreamer1.0-plugins-nvv4l2camerasrc-1.14.0-r32.7.3-r0 do_compile: oe_runmake failed

hmen5656, 2023-05-31:
I am trying to build "core-image-sato-dev" for jetson-nano board on kirkstone-l4t-r32.7.x  branch but i'm getting some errors on someof gstreamer1.0 layers .

Here is part of error code :
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: gstv4l2allocator.o: in function `gst_v4l2_allocator_alloc_mmap':
/usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2allocator.c:889: undefined reference to `NvBufferGetParams'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: gstv4l2allocator.o: in function `gst_v4l2_allocator_dqbuf':
/usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2allocator.c:1469: undefined reference to `NvBufferGetSize'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: gstv4l2bufferpool.o: in function `gst_v4l2_buffer_pool_copy_buffer':
/usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2bufferpool.c:291: undefined reference to `NvBufferTransform'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: /usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2bufferpool.c:281: undefined reference to `ExtractFdFromNvBuffer'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: /usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2bufferpool.c:231: undefined reference to `NvBufferMemMap'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: /usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2bufferpool.c:249: undefined reference to `NvBufferMemUnMap'
/home/sermed/Desktop/Yocto/build-jetson-nano/tmp/work/armv8a_tegra210-poky-linux/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/ld: gstv4l2bufferpool.o: in function `gst_v4l2_buffer_pool_import_dmabuf':
/usr/src/debug/gstreamer1.0-plugins-nvvideo4linux2/1.14.0-r32.7.3-r0/gst-v4l2/gstv4l2bufferpool.c:564: undefined reference to `ExtractFdFromNvBuffer'
collect2: error: ld returned 1 exit status
make: *** [Makefile:54: libgstnvvideo4linux2.so] Error 1
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
------------------------------------------------------------------------------------------------------

** These 2 packages are causing for crash ** 
gstreamer1.0-plugins-nvarguscamerasrc_1.0.0-r32.7.3.bb
gstreamer1.0-plugins-nvvideo4linux2-1.14.0-r32.7.3
------------------------------------------------------------------------------------------------------
**local.conf**
MACHINE = "jetson-nano-devkit"
DISTRO_FEATURES = "x11 opengl"

IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash tar.gz"
------------------------------------------------------------------------------------------------------

**bblayers.conf**
BBLAYERS ?= " \
  /home/sermed/Desktop/Yocto/poky/meta \
  /home/sermed/Desktop/Yocto/poky/meta-poky \
  /home/sermed/Desktop/Yocto/poky/meta-yocto-bsp \
  /home/sermed/Desktop/Yocto/meta-openembedded/meta-oe \
  /home/sermed/Desktop/Yocto/meta-openembedded/meta-python \
  /home/sermed/Desktop/Yocto/meta-openembedded/meta-multimedia \
  /home/sermed/Desktop/Yocto/meta-tegra \
  "
------------------------------------------------------------------------------------------------------

**Additional context**
i was successfully able to build same image could days ago but after i installed fresh Ubuntu 22.04 (previously machine had kinda fresh ubuntu as well i don't recall that i dowload something specific ) now i can't build "core-image-sato-dev" , if i initally build "core-image-sato" if successfully finishes and then  i try "dev" one i got error and i tried "core-image-sato" again and it's also failed. 
------------------------------------------------------------------------------------------------------

I am following that guilde : https://developer.ridgerun.com/wiki/index.php/Yocto_Support_for_NVIDIA_Jetson_Platforms_-_Setting_up_Yocto#Select_your_Machine 
Has anyone else faced that problem before ?

dwalkes, 2023-05-31:
@hmen5656 

> I am following that guilde : https://developer.ridgerun.com/wiki/index.php/Yocto_Support_for_NVIDIA_Jetson_Platforms_-_Setting_up_Yocto#Select_your_Machine

Those instructions are out of date I believe, and not supported or created by anyone who activiely participates in this discussion board.  I'd recommend using https://github.com/OE4T/tegra-demo-distro instead.  If you can reproduce there let me know the branch you are using and I'll attempt to reproduce.

hmen5656, 2023-05-31:
I have tried on https://github.com/OE4T/tegra-demo-distro for jetson-nano-devkit-cot and mentioned error didn't show up , however when i build  demo-image-base on kirkstone-l4t-r32.7.x branch i got another error which is :

ERROR: demo-image-base-1.0-r0 do_image_tegraflash: ExecutionError('/home/sermed/Desktop/tegra-demo-distro/build/tmp/work/jetson_nano_devkit_cot-oe4t-linux/demo-image-base/1.0-r0/temp/run.do_image_tegraflash.2506914', 1, None, None)

| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function set_image_size
| DEBUG: 379740.400000 = 292108 * 1.300000
| DEBUG: 14680064.000000 = max(379740.400000, 14680064)[14680064.000000] + 0
| DEBUG: 14680064.000000 = int(14680064.000000)
| DEBUG: 14680064 = aligned(14680064)
| DEBUG: returning 14680064
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_tegraflash
| cp: cannot stat '/home/sermed/Desktop/tegra-demo-distro/build/tmp/deploy/images/jetson-nano-devkit-cot/initrd-flash-kernel/Image-initrd-flash.cboot': No such file or directory
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/sermed/Desktop/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-base.bb:do_image_tegraflash) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4819 tasks of which 4783 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/sermed/Desktop/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-base.bb:do_image_tegraflash
Summary: There was 1 ERROR message, returning a non-zero exit code.

dwalkes, 2023-06-01:
> I have tried on https://github.com/OE4T/tegra-demo-distro for jetson-nano-devkit-cot and mentioned error didn't show up , however when i build demo-image-base on kirkstone-l4t-r32.7.x branch i got another error which is :

I can reproduce that error as well with the `jetson-nano-devkit-cot` - I think there's a problem with the logic at https://github.com/OE4T/meta-tegra/blame/8130f4221543ad50230da265d493c9579239fed8/classes/image_types_tegra.bbclass#L79-L86 for the `cot` images which I haven't been able to figure out in a few minutes of staring at it.

Did you actually intend to build [chain of trust FIT images](https://github.com/OE4T/tegra-demo-distro/wiki/Tegra-%26-Uboot-Flattened-Image-Tree-(FIT-image))?  Or did you just use that machine because you found it in the `tegra-demo-distro`?

If you don't need/want to use the `cot` image specifically I've verified the `jetson-nano-devkit` builds successfully on latest tegra-demo-distro.  I don't have a nano devkit handy to try this on but I've used the 2GB devkit recently on this branch so I believe it works.

Can you please change the title of the issue to match this new issue?  Something like "cannot stat Image-initrd-flash.cboot attempting to build jetson-nano-devkit-cot on kirkstone-l4t-r32.7.x" ?

hmen5656, 2023-06-01:
I also managed to compile for `jetson-nano-devkit` with `demo-image-base` ,`demo-image-sato` and with another custom meta layer . so everything is working pretty good so far . Thanks a lot for helping out !!

madisongh, 2023-06-05:
That `Image-initrd-flash.cboot` kernel image is for initrd flashing, which I never tested in conjunction with using FIT images on the u-boot platforms.  A workaround, if initrd flashing isn't needed, would be to disable it.  Setting `TEGRAFLASH_INITRD_FLASH_IMAGE = ""` should do that.

closed 2023-07-15 by madisongh