OE4T/meta-tegra #1280: Q: Using an alternate device tree for initrd flashing?

crdelsey, 2023-06-22:
**Describe the bug**
I want to flash multiple Orin AGXs at the same time, so I was looking at `initrd-flash` since that seems to replace the old `massflash` tool. To get it to work, it seems the USB interface on the Orin needs to be in `otg` mode. However, for my application, I need that interface in `host` mode. 

I managed to get things to work by cloning my device-tree, but with the USB interface in `otg` mode, and then using that alternate dtb in the `prepare_for_rcm_boot` [function](https://github.com/OE4T/meta-tegra/blob/master/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh#L195) and the `run_rcm_boot` [function](https://github.com/OE4T/meta-tegra/blob/master/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh#L217) . This seems it uses the alternate device tree for the flashing step, but replaces the alternate device-tree with my application device-tree, so I'm in `host` mode after the reboot

Before investing more time in making this robust, I just wanted to get some feedback on whether this is the right approach, or if there's a better way to solve the problem

Thanks,
Carl

madisongh, 2023-06-23:
Yeah, the OTG port does need to be able to switch into device mode for the flashing to work from the initrd.  Cloning your whole device tree to make that one change seems a bit burdensome, though.   Maybe something like we do for the bootloader DTB could be done for the kernel DTB - make a separate copy and apply an overlay (or maybe provide a customization function where you could use `fdtput` to directly modify the copy, if the existing customization functions in image_types_tegra aren't sufficient)?

crdelsey, 2023-06-27:
Thanks for the suggestions. I wasn't familiar with fdtput. I'll look into it.

closed 2023-06-27 by crdelsey