OE4T/meta-tegra #1281: libnvdla_compiler.so: cannot open shared object file

dferrarigrowtech, 2023-06-23:
I'm running demo-image-full image on jetson nano with yocto dunfell. I would like to run a nvidia container. When I import tensorrt through python in the container I get the following error:

`ImportError: libnvdla_compiler.so: cannot open shared object file: No such file or directory`

# To reproduce
- Build demo-image-full image on jetson nano with yocto dunfell
- Run `docker run -it --rm --runtime nvidia nvcr.io/nvidia/l4t-tensorrt:r8.2.1-runtime`
- Inside the container open a python terminal with `python3` and type `import tensorrt`

If I do the same thing with the jetson OS I don't get this messages. I think it could be related to the tegra-demo-distro or meta-tegra
Thank you!

dwalkes, 2023-06-23:
Thanks @dferrarigrowtech 

I can reproduce.

If I copy the content of `tegra-libraries-core.csv` into `l4t.csv` the `libnvdla_compiler.so` appears under `/usr/lib/aarch64-linux-gnu/libnvdla_compiler.so`.  So for some reason it seems at least some the .csv files under `/etc/nvidia-container-runtime/host-files-for-container.d` are being ignored.  May be related to https://github.com/OE4T/meta-tegra/pull/962#issuecomment-1100924427 but I thought this was specific to Jetpack 5.  I haven't found the place where l4t.csv is referenced/hard-coded yet.

dwalkes, 2023-06-24:
> I haven't found the place where l4t.csv is referenced/hard-coded yet.

I think it's [here](https://github.com/NVIDIA/libnvidia-container/blob/61f57bcdf7aa6e73d9a348a7e36ec9fd94128fb2/src/nvc_info.c#L130)

From `docker inspect nvidia nvcr.io/nvidia/l4t-tensorrt:r8.2.1-runtime`
I see:
```
...
            "Env": [
...
                "NVIDIA_REQUIRE_JETPACK_HOST_MOUNTS=base-only",
```

And if I run:
```
docker run -it --rm --runtime nvidia -e NVIDIA_REQUIRE_JETPACK_HOST_MOUNTS= nvcr.io/nvidia/l4t-tensorrt:r8.2.1-runtime
```
I'm able to run `python3` and type `import tensorrt` without issue.  So that's at least a workaround.

See related comment at https://github.com/NVIDIA/nvidia-docker/issues/1598#issuecomment-1024859273

I suspect the fix here is to do what we do with Jetpack 5 and try to put everything in l4t.csv.  Curious if anyone else has thoughts about this.

dferrarigrowtech, 2023-06-24:
Thanks for the quick reply and the workaround!

I just read this, according to nvidia:

> for JetPack 4.x all of the .csv files should get used by --runtime nvidia (as long as the container is derived from l4t-base).

[Source](https://forums.developer.nvidia.com/t/host-libraries-for-nvidia-container-runtime/234752/2)

madisongh, 2023-07-02:
A partial backport of #1038 is what's needed here - just the part that consolidates all of the BSP-related passthrough mounts into `l4t.csv`, without removing the parts where we have separate CSV files for other pacakages.  I wasn't aware that NVIDIA had published containers targeted at JetPack 4/L4T R32.x that used the new `base-only` passthrough mechanism, but support was added for them in the more recent releases.  With JetPack 5/L4T R35.x, all containers NVIDIA publishes use the new mechanism.

ichergui, 2023-08-27:
Hey @dferrarigrowtech 
Any update on this ?

dferrarigrowtech, 2023-08-27:
Nope, I'm using the suggested workaround

ichergui, 2023-08-27:
Can I close this ticket ?

dferrarigrowtech, 2023-08-28:
The issue is not completely fixed, but at least there is a workaround available! You can close it

closed 2023-08-28 by ichergui

dferrarigrowtech, 2024-02-24:
Sorry to bother you again but I just encountered a blocking issue with the aforementioned workaround.

Steps to reproduce:
- Start the container with the env. variable as mentioned above: `docker run -it --rm --runtime nvidia -e NVIDIA_REQUIRE_JETPACK_HOST_MOUNTS= nvcr.io/nvidia/l4t-tensorrt:r8.2.1-runtime`
- Update and install some packages in the container: `apt-get update && apt-get upgrade -y && apt-get install -y python3.8-dev build-essential`
- Update python installation tools: `python3.8 -m pip install -U pip setuptools`
- Install pycuda: `python3.8 -m pip install pycuda`

The installation process fails. But if you run the same instructions without `-e NVIDIA_REQUIRE_JETPACK_HOST_MOUNTS=`, then pycuda will be installed without any issue. The problem is that I need to use both `tensorrt` (which works only with the env variable setted) and `pycuda` (which works only without the env variable)