OE4T/meta-tegra #1293: current kirkstone L4T_VERSION leads to non existing branch on linux-tegra-5.10

creich, 2023-07-15:
current version of the kirkstone branch set's the `L4T_VERSION` to `35.3.1`. that version is reused within `linux-tegra_5.10.bb` to determine which branch to checkout from `linux-tegra-5.10`. 
unfortunately during the last few weeks that branch has been deleted from linux-tegra, which renders the kirstone branch of meta-tegra not buildable as is.
`ERROR: linux-tegra-5.10.104+gitAUTOINC+26cfd067b9-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/OE4T/linux-tegra-5.10.git;protocol=https;name=machine;branch=oe4t-patches-l4t-r35.3.1')`

I am not sure which branch should be used atm. At least found the very same commit on branch `oe4t-patches-l4t-r35.1` but since that version seems to be 'lower' than `35.3.1`, I wanted to discuss with you guys, which branch to use instead.

madisongh, 2023-07-15:
Are you sure you don't have a bbappend somewhere that is modifying the branch name from how it's set in the main recipe?

This is what I get for the source branch setting with a build off kirkstone latest:
```
$ bitbake -e virtual/kernel | grep '^SRCBRANCH'
SRCBRANCH="oe4t-patches-l4t-r35.3.ga"
```
which is the correct name.

creich, 2023-07-15:
you're right, there has been a bbappend.
sorry for the confusion

closed 2023-07-15 by creich