OE4T/meta-tegra #1298: Problem occur during SD Card boot

Pranjal-Parmar, 2023-07-17:
My SD Card is not displaying /boot/extlinux/extlinux.conf file after SD Card flash 

i have follow this steps for flashing the sd card 

step 1: Add the WIC profile

		cd $YOCTO_DIR
		mkdir meta-tegra/wic
step 2: create the file meta-tegra/wic/jetson-sdcard.wks with the following content: 	
		part / --source rootfs --ondisk mmcblk2 --fstype=ext4 --label root --align 4

step 3: Then add the following line to my machine configuration file $YOCTO_DIR/meta-tegra/conf/machine/jetson_agx_orin.conf
		WKS_FILE ?= "jetson-sdcard.wks"
step 4: adding IMAGE_FSTYPES = "tegraflash tar.gz wic.gz" to build/conf/local.conf file 

step 5: rebuild my image bitbake core-image-sato-dev
step 6: for flashing SD Card i have used this gunzip -c core-image-sato-dev-jetson-tx1.wic.gz | sudo dd of=/dev/sdd bs=10M status=progress

madisongh, 2023-07-17:
You don't indicate what actual hardware you're trying to use, or which branch you're working off of, but if you follow the instructions on the [wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) for creating and SDcard for booting, does it work?

madisongh, 2023-07-29:
@Pranjal-Parmar any update on this?

ichergui, 2023-08-27:
Closing this ticket since no update.
@Pranjal-Parmar  don't hesitate to re-open new issue.

closed 2023-08-27 by ichergui

RynaSr, 2023-09-13:
I am facing the same issue. Should I open a new ticket?
My hardware is Jetson Xavier NX Development kit with SD card boot having TX2 processor. I tried with poky master and kirkstone branch in line with meta-tegra. I have done all procedures as listed in original post.