OE4T/meta-tegra #1302: libgcc 8.5 do_configure() fails with yesterdays Poky mickledore update

lms-jk, 2023-07-27:
Hi,

our custom Yocto build failed at the do_configure() step for libgcc 8.5:
<details>
  <summary>config.log</summary>
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Runtime Library configure 1.0, which was
generated by GNU Autoconf 2.64.  Invocation command line was

  $ ../../../../../../../work-shared/gcc-8.5.0-r0/gcc-8.5.0/libgcc/configure --build=x86_64-linux --host=aarch64-leica-linux --target=aarch64-leica-linux --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot --enable-version-specific-runtime-libs --with-gxx-include-dir=/usr/include/c++/8.5.0 --with-gnu-ld --enable-shared --enable-languages=c,c++ --enable-threads=posix --disable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --disable-install-libiberty --enable-libstdcxx-pch --program-prefix=aarch64-leica-linux- --program-suffix=-8.5.0 --without-local-prefix --enable-lto --disable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-gxx-include-dir=/not/exist{target_includedir}/c++/8.5.0 --with-sysroot=/not/exist --with-build-sysroot=/home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot --without-long-double-128 libgcc_cv_powerpc_float128=no --disable-static --disable-nls --with-glibc-version=2.28 --enable-initfini-array --enable-__cxa_atexit

## --------- ##
## Platform. ##
## --------- ##

hostname = eab330fc114b
uname -m = x86_64
uname -r = 6.3.6-arch1-1
uname -s = Linux
uname -v = #1 SMP PREEMPT_DYNAMIC Mon, 05 Jun 2023 15:12:57 +0000

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/jk/pyrite/distro/build/tmp/sysroots-uninative/x86_64-linux/usr/bin
PATH: /home/jk/pyrite/distro/layers/poky/scripts
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot-native/usr/bin/aarch64-leica-linux-8.5.0
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot/usr/bin/crossscripts
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot-native/usr/sbin
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot-native/usr/bin
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot-native/sbin
PATH: /home/jk/pyrite/distro/build/tmp/work/armv8a-leica-linux/libgcc-8/8.5.0-r0/recipe-sysroot-native/bin
PATH: /home/jk/pyrite/distro/layers/poky/bitbake/bin
PATH: /home/jk/pyrite/distro/build/tmp/hosttools

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2268: checking build system type
configure:2282: result: x86_64-pc-linux-gnu
configure:2302: checking host system type
configure:2315: result: aarch64-leica-linux-gnu
configure:2418: checking for --enable-version-specific-runtime-libs
configure:2431: result: yes
configure:2479: checking for a BSD-compatible install
configure:2547: result: /home/jk/pyrite/distro/build/tmp/hosttools/install -c
configure:2563: checking for gawk
configure:2579: found /home/jk/pyrite/distro/build/tmp/hosttools/gawk
configure:2590: result: gawk
configure:2690: checking for aarch64-leica-linux-ar
configure:2717: result: aarch64-leica-linux-gcc-ar
configure:2782: checking for aarch64-leica-linux-lipo
configure:2812: result: no
configure:2822: checking for lipo
configure:2852: result: no
configure:2874: checking for aarch64-leica-linux-nm
configure:2901: result: aarch64-leica-linux-gcc-nm
configure:2966: checking for aarch64-leica-linux-ranlib
configure:2993: result: aarch64-leica-linux-gcc-ranlib
[... 225 lines cut ...]

configure: exit 1
</details>

The Yocto team removed the `--enable-standard-branch-protection` flag from their gcc-configure-common.inc in [yesterdays update|https://git.yoctoproject.org/poky/commit/?h=mickledore&id=151fd33bbe25233d17cd95ea29bb72ee50172425] which causes the error.

We re-added the flag to meta-tegras 'recipes-devtools/gcc-for-nvcc/gcc-8.5.inc' as a workaround:
<details>
  <summary>patch</summary>
diff --git a/recipes-devtools/gcc-for-nvcc/gcc-8-configure-common.inc b/recipes-devtools/gcc-for-nvcc/gcc-8-configure-common.inc
index caefa2c8..5e4d3480 100644
--- a/recipes-devtools/gcc-for-nvcc/gcc-8-configure-common.inc
+++ b/recipes-devtools/gcc-for-nvcc/gcc-8-configure-common.inc
@@ -33,6 +33,7 @@ EXTRA_OECONF = "\
     --program-prefix=${TARGET_PREFIX} \
     --program-suffix=-${BINV} \
     --without-local-prefix \
+       --enable-standard-branch-protection \
     ${EXTRA_OECONF_BASE} \
     ${EXTRA_OECONF_GCC_FLOAT} \
     ${EXTRA_OECONF_PATHS} \
</details>

Cheers,
Jan

llongi, 2023-07-27:
Same happens on kirkstone with last update. The flag was moved to TUNE_CCARGS:

https://git.yoctoproject.org/poky/commit/?h=kirkstone&id=a2bf83842a72c3dfe24a824c650a2f64d3fdf1b0

which seems to break gcc-for-nvcc.

madisongh, 2023-07-27:
I fixed this in master by updating gcc-for-nvcc to gcc 10, to make it more compatible with the newer gcc in OE-Core.

I'm not sure that's the right solution for the stable branches, though.  Filtering out the unrecognized added option in the right places (the libgcc-8 recipe and cuda.bbclass, I think) might be better.   Any preferences?

madisongh, 2023-07-27:
> but can't find anywhere CUDA_GCCVERSION is actually used, so is this still relevant?

Probably not, since we now have the separate gcc-for-nvcc toolchain.  I'll have to go back and refresh my memory on where that came from, though.

> Does R32.7.4's CUDA 10 still strictly work only with GCC 8?

I think that's right.  The CUDA documentation describes which version(s) of the different C toolchains you can use with each CUDA version.

So for the `kirkstone-l4t-r32.7.x` branch, we'll have to just filter out the added flag.

lms-jk, 2023-07-28:
For Mickledore, we don't have any strong preferences. We are not sure how disruptive the update to GCC 10 would be, but are open to test a mickledore with GCC 10.

Personally, I would leave GCC at version 8 since Nanbield is only a few weeks away.

kekiefer, 2023-07-28:
I'm in favor of sticking with the status quo for both Mickledore and Kirkstone. Thanks to the analysis above, I've added the following:

```
TUNE_CCARGS:remove = "-mbranch-protection=standard"
```

to `recipes-devtools/gcc-for-nvcc/gcc-8-common.inc`, and it is at least enough to get GCC 8 compiling again.

Are we all on the same page about doing this to recover these branches?

madisongh, 2023-07-29:
Closing this now, as the needed patches have gone into mickledore, kirkstone, and kirkstone-l4-r32.7.x.

closed 2023-07-29 by madisongh

lms-jk, 2023-07-31:
Thanks for the quick fix, guys!

PR #1305 (2023-07-29): gcc-for-nvcc: fix breakage in libgcc-8 with upstream TUNE_CCARGS change
Upstream commit
https://git.yoctoproject.org/poky/commit/?id=a2bf83842a72c3dfe24a824c650a2f64d3fdf1b0 added "-mbranch-protection=standard" to TUNE_CCARGS that breaks the gcc-for-nvcc (GCC 8) build during configure of libgcc-8:

configure:3731: error: in .../libgcc': configure:3734: error: cannot compute suffix of object files: cannot compile

Take this flag out of TUNE_CCARGS

Fixes #1302
commit 4dc2a8d3 (2023-07-29): gcc-for-nvcc: fix breakage in libgcc-8 with upstream TUNE_CCARGS change

Upstream commit
https://git.yoctoproject.org/poky/commit/?id=a2bf83842a72c3dfe24a824c650a2f64d3fdf1b0
added "-mbranch-protection=standard" to TUNE_CCARGS that breaks the
gcc-for-nvcc (GCC 8) build during configure of libgcc-8:

configure:3731: error: in .../libgcc': configure:3734: error: cannot compute suffix of object files: cannot compile

Take this flag out of TUNE_CCARGS

Fixes #1302
PR #1306 (2023-07-29): cuda.bbclass: filter TUNE_CCARGS for CUDA builds
to remove a flag added in OE-Core from compilations for CUDA targets that use an older compiler (through nvcc) which doesn't understand the new flag.

Fixes #1302
commit 5e7f658a (2023-07-29): gcc-for-nvcc: fix breakage in libgcc-8 with upstream TUNE_CCARGS change

Upstream commit
https://git.yoctoproject.org/poky/commit/?id=a2bf83842a72c3dfe24a824c650a2f64d3fdf1b0
added "-mbranch-protection=standard" to TUNE_CCARGS that breaks the
gcc-for-nvcc (GCC 8) build during configure of libgcc-8:

configure:3731: error: in .../libgcc': configure:3734: error: cannot compute suffix of object files: cannot compile

Take this flag out of TUNE_CCARGS

Fixes #1302