OE4T/meta-tegra #1340: imx477 camera / firmware issues with Orin/Forge

gsally-sarcos, 2023-08-22:
Almost have orin/forge setup working!  Having issues getting the camera to work.  The nvidia recommended way to get data from the imx477 camera is like so:

`gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=(int)4056, height=(int)3040, format=(string)NV12, framerate=(fraction)30/1' ! nvv4l2h264enc bitrate=80000000 ! h264parse ! qtmux ! filesink location=test.mp4 -e`

However, nvarguscamerasrc has been blacklisted. To get a better picture, I deleted `~/.cache/gstreamer-1.0/registry.aarch64.bin` and re-ran gst-inspect-1.0, this produces this output in the syslog

```
[  246.719916] nvgpu: 17000000.ga10b               gp10b_priv_ring_isr:217  [ERR]  ringmaster intr status0: 0x00000000, status1: 0x00000003
[  246.719934] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:684  [ERR]  GPC0 write error: ADR 0x004199b0 WRDAT 0x00000000 master 0x0000000b
[  246.719936] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:688  [ERR]  INFO 0x1840820b: (subid 0x00000018 priv_level 0 local_ordering 1)
[  246.719938] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:694  [ERR]  CODE 0xbadf5040
[  246.719942] nvgpu: 17000000.ga10b nvgpu_cic_mon_report_err_safety_services:92   [ERR]  Error reporting is not supported in this platform
[  246.719945] nvgpu: 17000000.ga10b ga10b_priv_ring_decode_error_code:542  [ERR]  [Error Type]: client error
[  246.719947] nvgpu: 17000000.ga10b    decode_pri_client_badf50_error:353  [ERR]  [Extra Info]: no such address, value(0x40)
[  246.719953] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:684  [ERR]  GPC1 write error: ADR 0x004199b0 WRDAT 0x00000000 master 0x0000000b
[  246.719954] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:688  [ERR]  INFO 0x1840820b: (subid 0x00000018 priv_level 0 local_ordering 1)
[  246.719955] nvgpu: 17000000.ga10b      ga10b_priv_ring_isr_handle_1:694  [ERR]  CODE 0xbadf5040
```
I'm not sure how to proceed, there's similar issues over on the nvidia developer's site, but the help stops for non-jetpack configurations.   Not sure if `gp10b_priv_ring_isr` is a cause or a symptom.  I've gone as far as overwriting all of the firmware with the firmware included with the latest JetPack, this didn't produce a change in behavior.  I checked and the md5sums for the for the firmware placed by the yocto build and what's jetpack and I think is being used is the same.

ichergui, 2023-08-27:
Hey @gsally-sarcos 
Did you try with `L4T` stock ? if so, is it working properly ?

AustinSchuh, 2023-09-04:
I've been able to get this to work for our board.  (8GB NX + Seeed j401 + imx477)  https://github.com/frc971/meta-frc971/tree/main is what I needed to do.

Your error sounds suspiciously like you are missing `camera_overrides.isp`

chamois94, 2023-09-05:
> Almost have orin/forge setup working! Having issues getting the camera to work. The nvidia recommended way to get data from the imx477 camera is like so:
>
> `gst-launch-1.0 nvarguscamerasrc ! [...]

Hello, do you find a solution ? Was it the camera_overrides.isp file ?
About the CTI Forge board, do you have specific meta configuration ? Eventually can you share it ?

gsally-sarcos, 2023-09-05:
@AustinSchuh, let me give that a try.  Thanks for the pipelines for moving the data across the network!  I was using netcat to move data from the device.  It seems the the folks at Nvidia just don't understand (or maybe just don't care, who knows) the typical use case for their devices will be w/o a monitor; from their perspective, the jetson is essentially an ARM-based desktop replacement instead of an embedded system.  /soapbox :)

I found that when I copied the firmware from R35.3.1 (literally, everything), the gstreamer pipeline worked, but the captured images approximated white noise.  I will track down what is different between the two and report back.

gsally-sarcos, 2023-09-06:
This is the set of firmware I found required to get the camera in some state of working:

```
71aeb0f904c96faf47e9c5b50d1429bf  ./ga10b/acr-gsp.text.encrypt.bin.prod
f1173654b0b93793deb423926e08a289  ./ga10b/NETA_img.bin
74ee8d5f7c5e46460855edbcfa1c4f34  ./ga10b/NETC_img.bin
a63061b2a611edfc2156da95067be2db  ./ga10b/gpccs.bin
70381c0996cafc5c3a95048e7e80fda4  ./ga10b/acr-gsp.data.encrypt.bin.prod
faacccf6bd3d2406c955a088dbe92637  ./ga10b/gpmu_ucode_next_prod_image.bin
aa3b084a0e58b8be685ed5da70d20b3e  ./ga10b/gpmu_ucode_next_prod_desc.bin
057aae9c8b07ca45ba0f82c273803ac7  ./ga10b/acr-gsp.manifest.encrypt.bin.out.bin.prod
a81e07cbab124cae94a306be84851d77  ./ga10b/fecs.bin
b80198d40ca24294e78997171ab32d1c  ./ga10b/gpccs_pkc_sig.bin
88fe14b4c9858cf76868e880a4837e18  ./ga10b/fecs_pkc_sig.bin
0d7173de548b4336d599c53a4d5f54b5  ./ga10b/NETD_img.bin
d0a6011ae4bb38aa362a04bd17d55fc9  ./ga10b/NETB_img.bin
3073b4593b5448934e02f7365dd174e4  ./ga10b/pmu_pkc_prod_sig.bin
9c8df15b09e8aedf12389be56af71cf6  ./tegra23x/nvhost_nvjpg013.fw
e69a2bb8925060a55771cecb2750e09c  ./tegra23x/nvhost_nvenc080.fw
ea65bb723f52fbd2b636fd8702e44990  ./tegra23x/nvhost_nvdec050_desc_prod.bin
a06f613fd43a149eec0b311e3a5fdb89  ./nvhost_vic042.fw
```

These files are from the R35.3.1 jetpack /lib/firmware directory distributed with the RFS. 

I'm using the camera_overrides.isp  from @AustinSchuh  https://github.com/frc971/meta-frc971/tree/main project.  The result is an out-of-focus image when I use this gstreamer pipeline: 

```
gst-launch-1.0 nvarguscamerasrc sensor-id=0 num-buffers=10 do-timestamp=true ! \
	'video/x-raw(memory:NVMM), width=(int)4056,height=(int)3040, framerate=40/1' ! \
	nvvidconv ! \
	nvjpegenc ! \
	filesink location="./test-file.jpg"
```

gsally-sarcos, 2023-09-06:
I also discovered that ```"nvidia, tegra-camera-platform"```  the space after the comma is very important!  When missing, this driver won't initialize, but if you have a camera depending on this driver, it will still initialize and appear to be running, but the code in ```drivers/video/tegra/camera/tegra_camera_platform.c``` is in some undefined state, leading to a kernel panic with no feedback.  I'll have to see if I can submit a patch to the file to add ```"nvidia,tegra-camera-platform"``` to of_device_id.

closed 2023-12-28 by madisongh