OE4T/meta-tegra #1341: CUDA Legacy CMake Compatibility does not follow `list` command rules

mascojames, 2023-08-23:
**Describe the bug**
`cuda.bbclass` appends a list of space-separated `CUDA_NVCC_FLAGS` to the `EXTRA_OECMAKE` for all recipes inheriting cuda.
The legacy `FindCUDA` that is provided by OE4T/tegra-cmake-overrides notes that for input variable `CUDA_NVCC_FLAGS`: 

 `"NOTE: multiple arguments must be  semi-colon delimited (e.g. --compiler-options;-Wall)"`

If the cmake file to be built uses the `list(APPEND` functionality in cmake to append to the `CUDA_NVCC_FLAGS`, those provided by the bbclass become one "megaflag" that is passed to NVCC in quotes which breaks nvcc flag parsing in very odd ways.

**To Reproduce**
Steps to reproduce the behavior:
1. Create recipe that utilizes both the cuda and cmake classes.
2. In `CMakeLists.txt` for test recipe, add `list(APPEND CUDA_NVCC_FLAGS --compiler-options -Wall)`
3. Build using machine from `meta-tegra` and note that flags passed to `nvcc` will be improperly quoted.

**Additional context**
Originally ran into creating recipe for ArrayFire. Sadly, I cannot share the recipe due to corporate rules.

madisongh, 2023-08-23:
If you change the line in `cuda.bbclass` that sets that CMake variable to be
```
-DCUDA_NVCC_FLAGS="${@';'.join(d.getVar('CUDA_NVCC_FLAGS').split())}"
```
does that fix the problem?  If you can give it a try and let me know, I'll queue up the fix, and you should be able to work around the issue in the meantime by overriding the `EXTRA_OECMAKE` setting in your recipe to use the above construct.

madisongh, 2023-08-23:
I've pushed the fix to mickledore and other branches where it applies.

closed 2023-08-23 by madisongh