OE4T/meta-tegra #1345: debian 12 build fails

himanshueip, 2023-08-23:
**Describe the bug**

ERROR: python3-native-3.11.2-r0 do_compile: oe_runmake failed
ERROR: python3-native-3.11.2-r0 do_compile: ExecutionError('/home/himanshu/tegra-oe/tegra-demo-distro/build/tmp/work/x86_64-linux/python3-native/3.11.2-r0/temp/run.do_compile.8907', 1, None, None)
ERROR: Logfile of failure stored in: /home/himanshu/tegra-oe/tegra-demo-distro/build/tmp/work/x86_64-linux/python3-native/3.11.2-r0/temp/log.do_compile.8907
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1 STAGING_LIBDIR=/home/himanshu/tegra-oe/tegra-demo-distro/build/tmp/work/x86_64-linux/python3-native/3.11.2-r0/recipe-sysroot-native/usr/lib STAGING_INCDIR=/home/himanshu/tegra-oe/tegra-demo-distro/build/tmp/work/x86_64-linux/python3-native/3.11.2-r0/recipe-sysroot-native/usr/include LIB=lib
| make: *** No targets.  Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (virtual:native:/home/himanshu/tegra-oe/tegra-demo-distro/layers/meta/recipes-devtools/python/python3_3.11.2.bb:do_compile) failed with exit code '1'

**To Reproduce**
. ./setup-env --machine jetson-xavier-nx-devkit build
bitbake demo-image-full

**Additional context**
👍 Will be starting contribution

madisongh, 2023-08-23:
That failure is with a pretty core Yocto recipe, rather than being something to do with meta-tegra.

You don't mention which branch your were using, but it looks like mickledore... have you been through a simple trial run using the [Yocto Project Quick Build](https://docs.yoctoproject.org/4.2.2/brief-yoctoprojectqs/index.html) documentation as a guide (the link there is for the Yocto Project mickledore/4.2.2 release)?

ichergui, 2023-08-27:
Hey @himanshueip 
I agree with @madisongh. This issue is unrelated with `tegra` meta layer. 
Any update on this ? Otherwise I will close the ticket.

himanshueip, 2023-08-28:
Hey @ichergui 
The issue seems to be resolved on downgrading to debian 11.
Thanks you and @madisongh  . You can close the ticket.

closed 2023-08-28 by himanshueip