OE4T/meta-tegra #1347: Build errors on Jetson TX2 Yocto- cuda-driver-10.2.300-1-r0 do_fetch: Failed

Arshdsam, 2023-08-27:
Hi,

I followed the below mentioned steps to build yocto image (demo-image-full) for my jetson tx2, but the build fails in between due to fetch error.

1. Gone through the steps mentioned in https://github.com/OE4T/tegra-demo-distro in my ubuntu 18.04 machine.
2. Downloaded all HOST COMPONENTS packages in ~/Download/nvidia/sdkm_downloads. SDK manager version: 1.9.3.10904 Jetpack Version: 4.6.4
3. tegra-demo-distro branch: dunfell (L4T R32.7.4)

Performed the below codes:
*) git clone https://github.com/OE4T/tegra-demo-distro.git	*) cd tegra-demo-distro	*) git submodule update --init	*) source ./setup-env --distro tegrademo --machine jetson-tx2.

Manually set the NVIDIA_DEVNET_MIRROR to ~/Download/nvidia/sdkm_downloads in ~/tegra-demo-distro/build/conf/local.conf file.	
Then, "bitbake demo-image-full"
On running these steps, I get the following error.
$ bitbake demo-image-full
Loading cache: 100% |########################################################################################################################| Time: 0:00:00
Loaded 3624 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################| Time: 0:00:01
Parsing of 2446 .bb files complete (2445 cached, 1 parsed). 3625 targets, 145 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe4t-linux"
MACHINE              = "jetson-tx2-devkit"
DISTRO               = "tegrademo"
DISTRO_VERSION       = "3.1.27+snapshot-20230827"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 = "HEAD:bce2280258497b42e8de94fbbc37cde627e2d1c5"
meta-tegra           
contrib              = "HEAD:133446740965f2ec7c97694d163d47db4009aa83"
meta-oe              
meta-python          
meta-networking      
meta-filesystems     = "HEAD:b8b0b06821d4d4df0cce4f07fa31a8ca1dd38f46"
meta-virtualization  = "HEAD:3f7feca9a1fc9b9a97187f5515ad3cca4bd93cb0"
meta-tegra-community = "HEAD:fc56b4d578187962695f1d58f3174f38427bc94b"
meta-tegra-support   
meta-demo-ci         
meta-tegrademo       = "dunfell:89ba608d8358fc707199cadffe74d96532ddf26b"

Initialising tasks: 100% |###################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 2157 Found 0 Missed 2157 Current 1149 (0% match, 34% complete)
NOTE: Executing Tasks
WARNING: cuda-driver-10.2.300-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb;name=dev;subdir=cuda-driver-10.2.300-1, attempting MIRRORS if available
ERROR: cuda-driver-10.2.300-1-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1825"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/hope/Dunfell/tegra-demo-distro/sources/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/bin/zstd-l4t-workaround-native:/home/hope/Dunfell/tegra-demo-distro/layers/scripts:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot/usr/bin/crossscripts:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/sbin:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/bin:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/sbin:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/bin:/home/hope/Dunfell/tegra-demo-distro/repos/poky/bitbake/bin:/home/hope/Dunfell/tegra-demo-distro/sources/tmp/hosttools"; export HOME="/home/hope"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -O /home/hope/Dunfell/tegra-demo-distro/sources/downloads/cuda-driver-dev-10-0_10.2.300-1_arm64.deb.tmp -P /home/hope/Dunfell/tegra-demo-distro/sources/downloads 'https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb' --progress=dot -v failed with exit code 8, no output
ERROR: cuda-driver-10.2.300-1-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb;name=dev;subdir=cuda-driver-10.2.300-1')
ERROR: Logfile of failure stored in: /home/hope/Dunfell/tegra-demo-distro/sources/tmp/work/armv8a_tegra-oe4t-linux/cuda-driver/10.2.300-1-r0/temp/log.do_fetch.22847
ERROR: Task (/home/hope/Dunfell/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-driver_10.2.300-1.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1555 tasks of which 1440 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/hope/Dunfell/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-driver_10.2.300-1.bb:do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

I tried multiple times (after: "rm -rf tmp" ) and ended up with same errors. 
When i checked my sdkm_downloads folder, these packages were present.
cuda-repo-cross-aarch64-ubuntu1804-10-2-local_10.2.460-1_all.deb
cuda-repo-l4t-10-2-local_10.2.460-1_arm64.deb
cuda-repo-ubuntu1804-10-2-local_10.2.460-450.115-1_amd64.deb
and, i even downloaded another driver package- " cuda-repo-ubuntu1804-10-2-local-10.2.89-440.33.01_1.0-1_amd64.deb" 
and placed in my sdkm_downloads.

Please help I'am stucked since, I'am very new to Yocto Project. So any help from your side is appreciated.
Please let me know if I am missing something here.

Thanks in Advance !!!

ichergui, 2023-08-27:
Hey @Arshdsam 
It should be working fine. `dunfell` branch is one of our active branches.
I'm not sure why you need **step 2** ?

Arshdsam, 2023-08-27:
hey @ichergui 
So don't we need to set NVIDIA_DEVNET_MIRROR = "file:/// $ {HOME}/Downloads/nvidia/sdkm_downloads" and for that i initiated step 2.

Arshdsam, 2023-08-27:
@ichergui I tried to access the link in the error and its showing "site can't be reached". Can you please check.

ichergui, 2023-08-27:
> hey @ichergui So don't we need to set NVIDIA_DEVNET_MIRROR = "file:/// $ {HOME}/Downloads/nvidia/sdkm_downloads" and for that i initiated step 2.

Nope, we don't need that

ichergui, 2023-08-27:
> @ichergui I tried to access the link in the error and its showing "site can't be reached". Can you please check.

I'm working on updating our active branches. I will check `dunfell` and I will keep you posted.

Arshdsam, 2023-08-27:
> > hey @ichergui So don't we need to set NVIDIA_DEVNET_MIRROR = "file:/// $ {HOME}/Downloads/nvidia/sdkm_downloads" and for that i initiated step 2.
> 
> Nope, we don't need that

By deleting the NVIDIA_DEVNET_MIRROR from local.conf also, I'am still getting the same error.

Arshdsam, 2023-08-27:
> > @ichergui I tried to access the link in the error and its showing "site can't be reached". Can you please check.
> 
> I'm working on updating our active branches. I will check `dunfell` and I will keep you posted.

I tried to built the image using kirkstone-l4t-r32.7.x branch, and there also I'am getting the same error.
ERROR: cuda-driver-10.2.300-1-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1825"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/hope/KirkTX2/sources/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/hope/KirkTX2/poky/scripts:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot/usr/bin/crossscripts:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/sbin:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/usr/bin:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/sbin:/home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/recipe-sysroot-native/bin:/home/hope/KirkTX2/poky/bitbake/bin:/home/hope/KirkTX2/sources/tmp/hosttools"; export HOME="/home/hope"; /usr/bin/env wget -t 2 -T 30 --passive-ftp -P /home/hope/KirkTX2/sources/downloads 'https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb' --progress=dot -v failed with exit code 8, no output
ERROR: cuda-driver-10.2.300-1-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb;name=dev;subdir=cuda-driver-10.2.300-1')
ERROR: Logfile of failure stored in: /home/hope/KirkTX2/sources/tmp/work/armv8a_tegra-poky-linux/cuda-driver/10.2.300-1-r0/temp/log.do_fetch.30825
ERROR: Task (/home/hope/KirkTX2/meta-tegra/recipes-devtools/cuda/cuda-driver_10.2.300-1.bb:do_fetch) failed with exit code '1'
so, i think i'am missing something !!! 
or is it from the NVIDIA side.
Waiting for you response after checking.

Arshdsam, 2023-08-27:
> > @ichergui I tried to access the link in the error and its showing "site can't be reached". Can you please check.
> 
> I'm working on updating our active branches. I will check `dunfell` and I will keep you posted.

I tried the kirkstone-l4t-r32.7.x branch (with Jetpack 4.6.4) also. But result is again the same do_fetch error in between the build process.
So, am i missing something here.
Or, is it from the NVIDIA side.
Waiting for you response after checking.

ichergui, 2023-08-28:
I tried both `kirkstone-l4t-r32.7.x` and `dunfell` with success build.

Arshdsam, 2023-08-28:
> I tried both `kirkstone-l4t-r32.7.x` and `dunfell` with success build.

Thankyou for checking out !!! 
So, can you guess what are the options that i could try to resolve this issues ? Or, Can you briefly explain how i could make a full image with tegra-demo-distro for my jetson TX2. Any help from your side is appreciated.

Thanks.

ichergui, 2023-08-28:
Hey @Arshdsam 
Please follow the steps below:
```
$ git clone https://github.com/OE4T/tegra-demo-distro.git -b dunfell 
$ cd tegra-demo-distro
$ git submodule update --init --recursive
$ . ./setup-env --machine jetson-tx2-devkit --distro tegrademo build-testdistro
$ bitbake demo-image-full
```
Let me know how it goes :slightly_smiling_face:

madisongh, 2023-08-28:
@Arshdsam If you  manually do a `wget` or `curl` on the URL mentioned in the log - https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda-cudart/cuda-driver-dev-10-0_10.2.300-1_arm64.deb;name=dev;subdir=cuda-driver-10.2.300-1 - what do you see?  This could be a problem either with NVIDIA's servers (maybe a regional problem) or with your local network's firewall blocking access to the URLs.

ichergui, 2023-08-28:
It looks like an issue related to your machine (CPU and RAM). 
Could you please run the following command; 
```
$ bitbake demo-image-full -c cleanall && bitbake demo-image-full -f
```

madisongh, 2023-08-28:
Since you mention you're using Ubuntu 18.04, I think this could be the  problem [mentioned here](https://github.com/Freescale/meta-freescale/issues/1593#issuecomment-1620614452).  Looks like fixes for it were recently back-ported to OE-Core dunfell, so I've updated the dunfell branch in the demo distro to pick them up.

If you do a `git pull` at the top of the repo, then `git submodule update` to update the submodules, and rebuild, hopefully that should clear up the problem.

quaresmajose, 2023-08-28:
Looks like this issue is related with https://lists.yoctoproject.org/g/yocto/topic/100638466#60801
Can you try this uninative fix https://git.openembedded.org/openembedded-core/commit/?id=1c5c8ff97ba0a7f9adc592d702b865b3d166a24b

Arshdsam, 2023-08-28:
> It looks like an issue related to your machine (`CPU` and `RAM`). Could you please run the following command;
> 
> ```
> $ bitbake demo-image-full -c cleanall && bitbake demo-image-full -f
> ```

@ichergui  I have followed the above steps and still got the same errors.

Arshdsam, 2023-08-28:
> Since you mention you're using Ubuntu 18.04, I think this could be the problem [mentioned here](https://github.com/Freescale/meta-freescale/issues/1593#issuecomment-1620614452). Looks like fixes for [...]

@madisongh  I followed the above methods. Everything seems fine but at 99% i got this error again: 
NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/4.2/x86_64-nativesdk-libc-4.2.tar.xz;sha256sum=6a86d71eeafba4fefec600c9bf8cf4a01324d1eb52788b6e398d3f23c10d19fb (will check PREMIRRORS first)
WARNING: /home/hope/DunfellOS/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-full.bb:do_build is tainted from a forced run:  0:00:03
Initialising tasks: 100% |###################################################################################################################| Time: 0:00:07
Sstate summary: Wanted 1955 Found 0 Missed 1955 Current 1792 (0% match, 47% complete)
NOTE: Executing Tasks
ERROR: demo-image-full-1.0-r0 do_image_ext4: Execution of '/home/hope/DunfellOS/tegra-demo-distro/build/tmp/work/jetson_tx2_devkit-oe4t-linux/demo-image-full/1.0-r0/temp/run.do_image_ext4.5371' failed with exit code 139
ERROR: Logfile of failure stored in: /home/hope/DunfellOS/tegra-demo-distro/build/tmp/work/jetson_tx2_devkit-oe4t-linux/demo-image-full/1.0-r0/temp/log.do_image_ext4.5371
Log data follows:
| DEBUG: Executing python function prepare_excluded_directories
| DEBUG: Python function prepare_excluded_directories finished
| DEBUG: Executing python function set_image_size
| DEBUG: 4716634.000000 = 3628180 * 1.300000
| DEBUG: 12053964.000000 = max(4716634.000000, 12053964)[12053964.000000] + 0
| DEBUG: 12053964.000000 = int(12053964.000000)
| DEBUG: 12053964 = aligned(12053964)
| DEBUG: returning 12053964
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', '/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-core/glibc/cross-localedef-native_2.31.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.2.bb:do_populate_sysroot', '/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/rpm/rpm_4.14.2.1.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/e2fsprogs/e2fsprogs_1.45.7.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', '/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/createrepo-c/createrepo-c_0.15.7.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/perl/perl_5.30.1.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/python/python3_3.8.17.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/opkg/opkg_0.4.2.bb:do_populate_sysroot', '/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb:do_populate_sysroot', 'virtual:native:/home/hope/DunfellOS/tegra-demo-distro/layers/meta/recipes-devtools/dnf/dnf_4.2.2.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['prelink-native', 'cross-localedef-native', 'opkg-utils-native', 'depmodwrapper-cross', 'rpm-native', 'e2fsprogs-native', 'makedevs-native', 'pigz-native', 'qemuwrapper-cross', 'update-rc.d-native', 'createrepo-c-native', 'perl-native', 'pseudo-native', 'python3-native', 'opkg-native', 'ldconfig-native', 'mklibs-native', 'dnf-native', 'elfutils-native', 'autoconf-native', 'gnu-config-native', 'binutils-native', 'libtool-native', 'automake-native', 'quilt-native', 'shadow-native', 'systemd-systemctl-native', 'mkfontscale-native', 'qemu-native', 'gdk-pixbuf-native', 'gettext-minimal-native', 'texinfo-dummy-native', 'attr-native', 'util-linux-native', 'pkgconfig-native', 'zlib-native', 'gconf-native', 'expat-native', 'file-native', 'openssl-native', 'ninja-native', 'sqlite3-native', 'glib-2.0-native', 'xz-native', 'libxml2-native', 'cmake-native', 'curl-native', 'gtk+3-native', 'shared-mime-info-native', 'bzip2-native', 'gdbm-native', 'libffi-native', 'libtirpc-native', 'libnsl2-native', 'readline-native', 'libsolv-native', 'libarchive-native', 'desktop-file-utils-native', 'kmod-native', 'debianutils-native', 'patch-native', 'dbus-native', 'popt-native', 'db-native', 'libdnf-nati [... cut ...]
ERROR: Task (/home/hope/DunfellOS/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-full.bb:do_image_ext4) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9025 tasks of which 8582 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/hope/DunfellOS/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-full.bb:do_image_ext4
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

I'm stuck at out of ideas, so Any help is well appreciated.
Thanks !!!

Arshdsam, 2023-08-29:
@ichergui @madisongh Is there any other way which i can solve this issues. Waiting for your response. :)

Arshdsam, 2023-08-29:
> Looks like this issue is related with https://lists.yoctoproject.org/g/yocto/topic/100638466#60801 Can you try this uninative fix https://git.openembedded.org/openembedded-core/commit/?id=1c5c8ff97ba0a7f9adc592d702b865b3d166a24b

@quaresmajose Thankyou for the reply. I noted your point,  checked the current yocto-uninative.inc and it gas the following data:
UNINATIVE_MAXGLIBCVERSION = "2.38"
UNINATIVE_VERSION = "4.2"

UNINATIVE_URL ?= "http://downloads.yoctoproject.org/releases/uninative/${UNINATIVE_VERSION}/"
UNINATIVE_CHECKSUM[aarch64] ?= "cff40e7bdde50aeda06707af8c001796a71b4cf33c5ae1616e5c47943ff6b94e"
UNINATIVE_CHECKSUM[i686] ?= "a70516447e9a9f1465ffaf1c7f89e79d1692d2356d86fd2a5a63acd908db1ff2"
UNINATIVE_CHECKSUM[x86_64] ?= "6a86d71eeafba4fefec600c9bf8cf4a01324d1eb52788b6e398d3f23c10d19fb"

So, do i need to agian update it or ??

quaresmajose, 2023-08-29:
> @quaresmajose Thankyou for the reply. I noted your point, checked the current yocto-uninative.inc and it gas the following data: UNINATIVE_MAXGLIBCVERSION = "2.38" UNINATIVE_VERSION = "4.2" [...]

Look like not as you have already the 4.2 and the fix lands in the 4.1

```
$ git log --oneline lmp-mirrors/dunfell -- meta/conf/distro/include/yocto-uninative.inc
35baf2ceb0 yocto-uninative: Update to 4.2 for glibc 2.38
600ef23e30 yocto-uninative: Update hashes for uninative 4.1

```

Arshdsam, 2023-08-29:
@madisongh @ichergui I tried to bitbake core-image-minimal, it also ended up with the same error at 99%. I tried using the inputs from @quaresmajose and then also, there was no difference at all. The problem happens @ do image ext4.
ERROR MESSAGE:
ERROR: Execution of '/home/hope/DunfellOS/tegra-demo-distro/build/tmp/work/jetson_tx2_devkit-oe4t-linux/demo-image-full/1.0-r0/temp/run.do_image_ext4.23014' failed with exit code 139
ERROR: Task (/home/hope/DunfellOS/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-full.bb:do_image_ext4) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9025 tasks of which 9018 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/hope/DunfellOS/tegra-demo-distro/layers/meta-tegrademo/recipes-demo/images/demo-image-full.bb:do_image_ext4
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

quaresmajose, 2023-08-29:
Can you please try doing the same build but using another host machine like the ubuntu 20.04? or using a container like crops. It seems that this issue only happens on ubuntu 18.04.

You can also try just with poky (without meta-tegra) on your ubuntu 18.04 but instated of build demo-image-full please use core-image-minimal.

Arshdsam, 2023-08-31:
@madisongh @ichergui @quaresmajose At last I was able to build the demo-image-full without any errors in ubuntu 18.04 LTS itself, using the patch mentioned by @quaresmajose.  I deleted the build directory and started bitbake once again without any additions to the existing local.conf file.  Thankyou guys for your timely supports, means a lot !!!

ichergui, 2023-08-31:
Closing this ticket since it was solved.
Thanks @madisongh and @quaresmajose

closed 2023-08-31 by ichergui