OE4T/meta-tegra #1359: Network issues in UEFI after second boot

lms-jk, 2023-09-07:
Hi,

we are using the latest meta-tegra from the mickledore branch on an AGX Orin Devkit and experiment with the new PXE Boot option of Jetson Linux 35.4.1.

Unfortunately we experience some very weird networking issue, where we have no issues on the first boot after flashing a new image, but can't ping any host on the network on the second boot. Obtaining an IP address from our DHCP server still seems to work on the second boot:

```
Shell> ifconfig -l

-----------------------------------------------------------------

name         : eth0
Media State  : Media present
policy       : dhcp
mac addr     : XX:XX:XX:XX:XX:XX

ipv4 address : 192.168.167.50

subnet mask  : 255.255.255.0

default gateway: 0.0.0.0

  Routes (2 entries):
    Entry[0]
     Subnet : 192.168.167.0
     Netmask: 255.255.255.0
     Gateway: 0.0.0.0
    Entry[1]
     Subnet : 0.0.0.0
     Netmask: 0.0.0.0
     Gateway: 192.168.167.1

DNS server   :
-----------------------------------------------------------------

Shell> ping 192.168.167.1
Ping 192.168.167.1 16 data bytes.

2 packets transmitted, 0 received, 100% packet loss, time 0ms
Shell>
```

Does somebody else experience the same issue?

dwalkes, 2023-09-08:
Hi @lms-jk 

Have you checked to see if this is something specific to meta-tegra? Can you reproduce with Nvidia's source release?  If so you may want to cross post on their developer forum.

lms-jk, 2023-09-08:
Hi @dwalkes 

we will try Nvidia's release next week to check.

In the meantime we found a workaround: When we reset the NVRAM variables in the UEFI with Device Manager -> NVIDIA Configuration -> Reset Settings & reboot we can use the networking again. We have to do this after every reboot, but it is better than flashing a new image.

lms-jk, 2023-09-12:
We experienced the same issue with the prebuilt image provided by Nvidia. I opened a post on their developer forum: https://forums.developer.nvidia.com/t/jetson-linux-35-4-1-network-issues-in-uefi-after-second-boot/266130

closed 2023-12-28 by madisongh