OE4T/meta-tegra #1360: setup-nv-boot-control causing rcu stall on JAOi

tbottomparallel, 2023-09-07:
I'm trying to bring up a Jetson AGX Orin Industrial module, right now just on a devkit carrier.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch `wip-l4t-r35.4.1` or `kirkstone` with `MACHINE` based on 'jetson-agx-orin-industrial'
2. hard-flash to orin industrial module on devkit carrier board
3. Boot logs look completely normal until `setup-nv-boot-control` runs:
```
Running Set up redundant boot configuration:
[   12.859911] FAT-fs (mmcblk0p10): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   15.317673] tegra-hda 3510000.hda: azx_get_response timeout, switching to polling mode: last cmd=0x001f0500
[   33.965663] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[   33.965860] rcu:     0-...0: (1 GPs behind) idle=85a/1/0x4000000000000000 softirq=547/549 fqs=1889
[   33.966114]  (detected by 7, t=5252 jiffies, g=-283, q=1189)
[   33.966282] Task dump for CPU 0:
[   33.966374] task:kworker/u24:0   state:R  running task     stack:    0 pid:    7 ppid:     2 flags:0x0000002a
[   33.966671] Workqueue: efi_rts_wq efi_call_rts
[   33.966796] Call trace:
[   33.966887]  __switch_to+0xc8/0x120
[   33.966992]  0xaff
[   33.967061]  efi_call_rts+0x1b0/0x44c
[   33.967171]  process_one_work+0x1d8/0x4b0
[   33.967297]  worker_thread+0x144/0x470
[   33.967404]  kthread+0x160/0x164
[   33.967495]  ret_from_fork+0x10/0x24
[   33.967591] rcu: ====For debug only: Start Printing Blocked Tasks====<print_other_cpu_stall>
[   33.968050] task:kworker/5:0     state:D stack:    0 pid:   40 ppid:     2 flags:0x00000028
[   33.969268] Workqueue: pm pm_runtime_work
[   33.969885] Call trace:
[   33.970270]  __switch_to+0xc8/0x120
[   33.970814]  __schedule+0x314/0x9a0
[   33.971351]  schedule+0x78/0x104
[   33.973650]  schedule_timeout+0x9c/0x1d4
[   33.977588]  wait_for_completion_timeout+0x88/0x110
[   33.982402]  tegra_bpmp_transfer+0x1c0/0x3c0
[   33.986604]  tegra_bpmp_clk_transfer+0x88/0xd4
[   33.991063]  tegra_bpmp_clk_unprepare+0x50/0xb0
[   33.995616]  clk_core_unprepare+0x94/0x260
[   33.999639]  clk_unprepare+0x38/0x50
[   34.003140]  tegra_sdhci_set_host_clock.part.0+0x94/0x350
[   34.008563]  sdhci_tegra_runtime_suspend+0xdc/0x144
[   34.013378]  pm_generic_runtime_suspend+0x40/0x60
[   34.018099]  __rpm_callback+0xe0/0x1a0
[   34.022037]  rpm_callback+0x74/0xa0
[   34.025712]  rpm_suspend+0x110/0x650
[   34.029037]  pm_runtime_work+0xc0/0xcc
[   34.032800]  process_one_work+0x1d8/0x4b0
[   34.036826]  worker_thread+0x144/0x470
[   34.040587]  kthread+0x160/0x164
[   34.043999]  ret_from_fork+0x10/0x24
[   34.047617] task:khugepaged      state:D stack:    0 pid:   86 ppid:     2 flags:0x00000028
[   34.055813] Call trace:
[   34.058437]  __switch_to+0xc8/0x120
[   34.061939]  __schedule+0x314/0x9a0
[   34.065526]  schedule+0x78/0x104
[   34.068851]  schedule_timeout+0x194/0x1d4
[   34.073049]  wait_for_completion+0x88/0x120
[   34.077164]  __flush_work.isra.0+0x16c/0x2d0
[   34.081538]  flush_work+0x24/0x30
[   34.084954]  lru_add_drain_all+0x190/0x200
[   34.088981]  khugepaged+0x9c/0x1d00
[   34.092475]  kthread+0x160/0x164
[   34.095799]  ret_from_fork+0x10/0x24
[   34.099320] task:kworker/5:2     state:D stack:    0 pid:  148 ppid:     2 flags:0x00000028
[   34.107353] Workqueue: pm pm_runtime_work
[   34.111374] Call trace:
[   34.113913]  __switch_to+0xc8/0x120
[   34.117589]  __schedule+0x314/0x9a0
[   34.121262]  schedule+0x78/0x104
[   34.124500]  schedule_preempt_disabled+0x2c/0x44
[... 183 lines cut ...]
8be4df61-93ca-11d2-aa0d-00e098032b8c-ConIn
8be4df61-93ca-11d2-aa0d-00e098032b8c-ConInDev
8be4df61-93ca-11d2-aa0d-00e098032b8c-ConOut
8be4df61-93ca-11d2-aa0d-00e098032b8c-ConOutDev
8be4df61-93ca-11d2-aa0d-00e098032b8c-ErrOut
8be4df61-93ca-11d2-aa0d-00e098032b8c-ErrOutDev
8be4df61-93ca-11d2-aa0d-00e098032b8c-Key0000
8be4df61-93ca-11d2-aa0d-00e098032b8c-Key0001
8be4df61-93ca-11d2-aa0d-00e098032b8c-Lang
8be4df61-93ca-11d2-aa0d-00e098032b8c-LangCodes
8be4df61-93ca-11d2-aa0d-00e098032b8c-OsIndicationsSupported
8be4df61-93ca-11d2-aa0d-00e098032b8c-PlatformLang
8be4df61-93ca-11d2-aa0d-00e098032b8c-PlatformLangCodes
8be4df61-93ca-11d2-aa0d-00e098032b8c-PlatformRecovery0000
8be4df61-93ca-11d2-aa0d-00e098032b8c-SecureBoot
8be4df61-93ca-11d2-aa0d-00e098032b8c-SetupMode
8be4df61-93ca-11d2-aa0d-00e098032b8c-SignatureSupport
8be4df61-93ca-11d2-aa0d-00e098032b8c-Timeout
8be4df61-93ca-11d2-aa0d-00e098032b8c-VendorKeys
d9bee56e-75dc-49d9-b4d7-b534210f637a-certdb
d9bee56e-75dc-49d9-b4d7-b534210f637a-certdbv
eb704011-1402-11d3-8e77-00a0c969723b-MTC
```

Any debugging advice or insights greatly appreciated, happy to gather more debug info too

ichergui, 2023-09-08:
Hello @tbottomparallel 
Thanks for reporting this issue. I will try to reproduce this issue. 
I will keep you posted.
Did you try with `L4T/Jetpack` stock ?

ichergui, 2023-09-09:
Hey @tbottomparallel 

I tried to reproduce the bug but I can't.

Kindly see below: 

```
root@jetson-agx-orin-devkit-industrial:~# systemctl restart setup-nv-boot-control                                                                                                                            
root@jetson-agx-orin-devkit-industrial:~# 
root@jetson-agx-orin-devkit-industrial:~# journalctl -u setup-nv-boot-control --no-pager 
Jul 28 08:26:33 jetson-agx-orin-devkit-industrial systemd[1]: Starting Set up redundant boot configuration...
Jul 28 08:26:33 jetson-agx-orin-devkit-industrial systemd[1]: setup-nv-boot-control.service: Deactivated successfully.
Jul 28 08:26:33 jetson-agx-orin-devkit-industrial systemd[1]: Finished Set up redundant boot configuration.
Sep 09 12:05:05 jetson-agx-orin-devkit-industrial systemd[1]: Starting Set up redundant boot configuration...
Sep 09 12:05:05 jetson-agx-orin-devkit-industrial systemd[1]: setup-nv-boot-control.service: Deactivated successfully.
Sep 09 12:05:05 jetson-agx-orin-devkit-industrial systemd[1]: Finished Set up redundant boot configuration.
root@jetson-agx-orin-devkit-industrial:~# 
``` 

* Running the command line `setup-nv-boot-control`
```
root@jetson-agx-orin-devkit-industrial:~# /usr/bin/setup-nv-boot-control 
root@jetson-agx-orin-devkit-industrial:~# echo $?
0
root@jetson-agx-orin-devkit-industrial:~#
```

tbottomparallel, 2023-09-11:
@ichergui thanks for testing that. I have a couple other bringup items I need to do first, but I'll test with stock L4T asap and post the results.

ichergui, 2023-09-11:
> @ichergui thanks for testing that. I have a couple other bringup items I need to do first, but I'll test with stock L4T asap and post the results.

Thanks

ichergui, 2023-09-23:
Hey @tbottomparallel 
Any update on this ?

tbottomparallel, 2023-09-26:
@ichergui sorry none yet, it's been a busy couple of weeks. But I should be able to look at it soon.

closed 2024-03-24 by madisongh

tbottomparallel, 2024-05-02:
This took a lot longer to get back to than originally expected. For anyone who stumbles across this issue, it was 100% user error.

I added a fixed regulator to the device tree
```
        fixed-regulators {
                tpa3116_enable: amp-enable {
                        compatible = "regulator-fixed";
                        regulator-name = "tpa3116-enable";
                        regulator-min-microvolt = <5000000>;
                        regulator-max-microvolt = <5000000>;
                        enable-active-high;
                        gpio = <&tegra_main_gpio TEGRA234_AON_GPIO(BB, 2) GPIO_ACTIVE_HIGH>;
                        startup-delay-us = <1000>;
                };
        };
```

which resulted in `gpio-358 (PC.01               |fixed-regulators:amp) out lo`

`&tegra_main_gpio` should have been `&tegra_aon_gpio`:
```
        fixed-regulators {
                tpa3116_enable: amp-enable {
                        compatible = "regulator-fixed";
                        regulator-name = "tpa3116-enable";
                        regulator-min-microvolt = <5000000>;
                        regulator-max-microvolt = <5000000>;
                        enable-active-high;
                        gpio = <&tegra_aon_gpio TEGRA234_AON_GPIO(BB, 2) GPIO_ACTIVE_HIGH>;
                        startup-delay-us = <1000>;
                };
        };
```
for the correct result `gpio-326 (PBB.02              |fixed-regulators:amp) out lo`

It escaped notice because I also accidentally defaulted that gpio to `output-high` instead of `output-low`.