OE4T/meta-tegra #1365: tegra234-flash-helper.py pinmux override

teebr, 2023-09-13:
Hello,

I ran into a problem recently when flashing a custom device which uses an Orin AGX 64GB SoM. With the latest batch of devices we received the SoM is newer and now the pinmux is being overridden by [this part](https://github.com/OE4T/meta-tegra/blob/kirkstone/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra234-flash-helper.sh#L385) of the flash helper script. It seems like SOMs to date have had a FAB of 500, but these new ones have a FAB of TS1, which meets the egrep expression. 

I commented out the block from 383-401 and flashing seems to be successful. Could someone explain context for this block is and if some/ all of it is okay to skip, at least in cases like mine?

madisongh, 2023-09-13:
The logic there is a paraphrase of the logic at the top of the `update_flash_args_common()` function in the `jetson-agx-orin-devkit.conf` file in the L4T kit.  I think, though, that we missed bracketing that if block with a check that board SKU isn't `0005` (which should be the board SKU for the 64GB AGX Orin).

teebr, 2023-09-14:
Thanks @madisongh, I have confirmed that this patch for the R35.3.1 helper script works with the older and newer SoMs:

```
diff --git a/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra234-flash-helper.sh b/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra234-flash-helper.sh
index c12c877e..84156b2d 100644
--- a/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra234-flash-helper.sh
+++ b/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra234-flash-helper.sh
@@ -385,10 +385,12 @@ if [ "$BOARDID" = "3701" ]; then
             exit 1
             ;;
     esac
-    if [ "$chip_sku" = "00" -o "$chip_sku" = "D0" ] &&
-	   echo "$FAB" | egrep -q '^(TS[123]|EB[123]|[012]00)$'; then
-	PINMUX_CONFIG="tegra234-mb1-bct-pinmux-p3701-0000.dtsi"
-	PMC_CONFIG="tegra234-mb1-bct-padvoltage-p3701-0000.dtsi"
+    if [ "${board_sku}" != "0005" ]; then
+        if [ "$chip_sku" = "00" -o "$chip_sku" = "D0" ] &&
+            echo "$FAB" | egrep -q '^(TS[123]|EB[123]|[012]00)$'; then
+        PINMUX_CONFIG="tegra234-mb1-bct-pinmux-p3701-0000.dtsi"
+        PMC_CONFIG="tegra234-mb1-bct-padvoltage-p3701-0000.dtsi"
+        fi
     fi
     if ! [ "$BOARDSKU" = "0000" -o "$BOARDSKU" = "0001" -o "$BOARDSKU" = "0002" ]; then
 	BPFDTB_FILE=$(echo "$BPFDTB_FILE" | sed -e"s,3701-0000,3701-$BOARDSKU,")
```

Unfortunately I can't test with R35.4.1 right now, but I think it's safe to say it should still apply?

madisongh, 2023-09-15:
> Unfortunately I can't test with R35.4.1 right now, but I think it's safe to say it should still apply?

Yes, should be fine.  That bit of the `update_flash_args_common` function hasn't changed between 35.3.1 and 35.4.1.

PR #1377 (2023-09-20): tegra-helper-scripts: fix AGX Orin 64GB handling in t234 helper
Comparing to the logic in the L4T scripts, we were missing a check that the board SKU isn't 0005 before seeing if the pinmux and PMC config files need to be overridden based on the chip SKU and FAB values.

Fixes #1365
closed 2023-09-20 by madisongh