OE4T/meta-tegra #1366: SD card image contains empty boot for jetson xavier NX developer kit

RynaSr, 2023-09-13:
I am working on [NVIDIA Jetson Xavier NX Developer Kit](https://a.co/d/h11X1vL) (812674024318), 16 GB and I followed following steps to generate image to flash SD card for booting- 
1: Add the WIC profile
    `mkdir meta-tegra/wic`
2: create the file meta-tegra/wic/jetson-sdcard.wks with the following content:
    `part / --source rootfs --ondisk mmcblk2 --fstype=ext4 --label root --align 4`
3: Then add the following line to my machine configuration file $YOCTO_DIR/meta-tegra/conf/machine/jetson-xavier-nx-devkit.conf
    `WKS_FILE ?= "jetson-sdcard.wks"`
4: adding `IMAGE_FSTYPES = "tegraflash tar.gz wic.gz"` to build/conf/local.conf file
5: rebuild my image `bitbake core-image-sato-dev`
6: for flashing SD Card `gunzip -c core-image-sato-dev-jetson-xavier-nx-devkit.wic.gz | sudo dd of=/dev/sdd bs=10M status=progress`

**Note : I have made this images using master branch of poky and meta-tegra and also with Kirstone branch for both.**

As per https://developer.ridgerun.com/wiki/index.php/Yocto_Support_for_NVIDIA_Jetson_Platforms_-_Flashing_the_Jetson_Platform, once I have flashed the SD card, I need to access the /boot/extlinux/extlinux.conf file of the SD Card. I am unable to see any content inside boot directory named extlinux. It contains efi directory inside boot, which is empty.

This issue is similar to reported issue #https://github.com/OE4T/meta-tegra/issues/1298

madisongh, 2023-09-13:
If you're using the stock Xavier NX dev kit, please just follow the instructions on our own [wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit).  The page you reference provides special instructions for  the Jetson TX1 and Jetson TX2, which work differently.

If you're just starting out, I would also recommend using our [demo distro](https://github.com/OE4T/tegra-demo-distro), you might find it simpler to get something working.

1ykos, 2023-09-14:
Same here with Orin Nano devkit . I don't just want to boot it somehow, I've managed to flash our Jetson boards with the NVIDIA SDK manager. But I want to be able to modify the kernel to help in driver development for a new camera module.

1ykos, 2023-09-14:
@RynaSr can you try dunfell too? There is no support for Orin in dunfell, but there should  be for Xavier.

RynaSr, 2023-09-14:
> @RynaSr can you try dunfell too? There is no support for Orin in dunfell, but there should be for Xavier.

I had tried that.. It didn't work for me.. wic image doesn't work at all. It creates only rootfs images in wic ext. I don't think that's the whole sd card image to boot the system. I had build multiple images from multiple branches but no luck.
I am trying to go over @madisongh suggestion. I will update the result here.

RynaSr, 2023-09-14:
> If you're using the stock Xavier NX dev kit, please just follow the instructions on our own [wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit). The page you reference provides special instructions for the Jetson TX1 and Jetson TX2, which work differently.
> 

So this worked. Board booted up and it showed NVIDIA logo on screen. Though keyboard and mouse both were connected but I was unable to get control of the prompt. Am I suppose to do any configuration to enable them as HID?

> If you're just starting out, I would also recommend using our [demo distro](https://github.com/OE4T/tegra-demo-distro), you might find it simpler to get something working.

As NVIDIA logo came up, looks like poky/meta-tegra - Kirkstone and master branch are working for Jetson Xavier NX Dev kit. But your recommendation looks much easy and board specific. I have put up a build and would like to try it out. Thank you

dwalkes, 2023-09-14:
>  Though keyboard and mouse both were connected but I was unable to get control of the prompt. Am I suppose to do any configuration to enable them as HID

You shouldn't need to do anything... which MACHINE, branch, and image did you end up using?  I could try to reproduce.

> I would also recommend using our [demo distro](https://github.com/OE4T/tegra-demo-distro),
> Kirkstone and master branch are working for Jetson Xavier NX Dev kit.

Do you mean working other than the keyboard and mouse?  Or did you see something different when you used tegra-demo-distro versus when you built your own distribution?

RynaSr, 2023-09-14:
> > Though keyboard and mouse both were connected but I was unable to get control of the prompt. Am I suppose to do any configuration to enable them as HID
> 
> You shouldn't need to do anything... which MACHINE, branch, and image did you end up using? I could try to reproduce.

I tried MACHINE ?= "jetson-xavier-nx-devkit"
branch and image name is listed below - 
1. [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) (kirkstone) - demo-image-full-jetson-xavier-nx-devkit.tegraflash.tar.gz
2. poky (kirkstone) / meta-tegra (kirkstone) - core-image-sato-dev-jetson-xavier-nx-devkit.tegraflash.tar.gz
3. poky (master) / meta-tegra (master) - core-image-sato-dev-jetson-xavier-nx-devkit.tegraflash.tar.gz

> Do you mean working other than the keyboard and mouse? Or did you see something different when you used tegra-demo-distro versus when you built your own distribution?

With working, I meant to say only NVIDIA logo is coming up with all these images. Keyboard and mouse is not working with any of the above listed image. Following are the steps I am following to flash and bring up the board -

1.  tar -x -f demo-image-full-jetson-xavier-nx-devkit.tegraflash.tar.gz
2. sudo ./dosdcard.sh /dev/sdb
Last step creates 22 partitions on SD card. I put SD card into the board  after finishing second step. I hope for SPI "./doflash.sh" steps are not needed while I am trying to boot from the SD card. Please correct me if I am wrong.

dwalkes, 2023-09-14:
> only NVIDIA logo is coming up with all these images
> I hope for SPI "./doflash.sh" steps are not needed while I am trying to boot from the SD card. Please correct me if I am wrong.

I would definitely try a full `./doflash.sh` if you aren't booting up after a `./dosdcard.sh`.

1ykos, 2023-09-15:
Thank you, the wiki instructions did work, I can boot and USB devices work too!

RynaSr, 2023-09-15:
> ```
> CORE_IMAGE_EXTRA_INSTALL = "kernel-modules"
> ```

This makes a lot more sense. I have included it, thank you for mentioning.. I will check and post the results..

Can you please confirm if I am booting from SD card in Jetson Xavier NX board, I need not to perform [Using an SDcard with the Jetson AGX Xavier  ](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#using-an-sdcard-with-the-jetson-agx-xavier) step?
After doing [Unpacking the tegraflash package](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#unpacking-the-tegraflash-package), I am performing steps mentioned in [Writing an SDcard](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#writing-an-sdcard) and putting SD card back into the board. I am assuming I need not to perform [Flashing the Jetson](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#flashing-the-jetson) too.

RynaSr, 2023-09-15:
> Thank you, the wiki instructions did work, I can boot and USB devices work too!

Have you tried that on Jetson Xavier NX or orin board? Did you boot from SD card? If yes, had you performed [Flashing the Jetson](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#flashing-the-jetson) ?

madisongh, 2023-09-17:
> Can you please confirm if I am booting from SD card in Jetson Xavier NX board, I need not to perform [Using an SDcard with the Jetson AGX Xavier ](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit#using-an-sdcard-with-the-jetson-agx-xavier)step?

That's correct.  The AGX Xavier dev kit normally boots from its internal eMMC, so there's special extra stuff in place to allow for booting from its SDcard slot instead.  With the Xavier NX dev kit, no extra steps are needed.

RynaSr, 2023-09-18:
> That's correct. The AGX Xavier dev kit normally boots from its internal eMMC, so there's special extra stuff in place to allow for booting from its SDcard slot instead. With the Xavier NX dev kit, [...]

This is strange that steps being so simple, it is not working for me. I am not sure, where I am doing wrong. I am sharing the conf files along with the steps I am following. Appreciate if you can guide -

```
1. checked out tegra-demo-distro (kirkstone)
2. abc@xyz2:~/tegra-demo-distro$ . ./setup-env --machine jetson-xavier-nx-devkit
3. renamed local.conf -> local.txt to be able to attach here for reference. 
[local.txt](https://github.com/OE4T/meta-tegra/files/12650750/local.txt)

4. abc@xyz2:~/tegra-demo-distro/build$ bitbake  demo-image-full
5. abc@xyz2:~/tegra-demo-distro/build/tmp/deploy/images/jetson-xavier-nx-devkit$ tar -xjf demo-image-full-jetson-xavier-nx-devkit-20230915155120.tegraflash.tar.gz
6. abc@xyz2:~/tegra-demo-distro/build/tmp/deploy/images/jetson-xavier-nx-devkit$ ./dosdcard.sh /dev/sdb
```

and I insert the SD card back on the board to boot up. NVIDIA logo comes up and keyboard + mouse doesn't work.

RynaSr, 2023-09-18:
@madisongh I am unable to get serial connection too. I had connected micro USB but can't see any COM connection getting open. It is working with NVIDIA provided image. Please check the instruction I mentioned above on how I am creating and flashing the image and suggest me what could be wrong. Thank you

dwalkes, 2023-09-19:
@RynaSr have you tried `./doflash.sh` instead of `./dosdcard.sh`?

> I had connected micro USB but can't see any COM connection getting open

I think you need to use the UART pins on the debug header for the UART terminal by default.

RynaSr, 2023-09-19:
https://github.com/OE4T/meta-tegra/issues/1366#issuecomment-1718321610 worked

closed 2023-09-19 by RynaSr

RynaSr, 2023-09-25:
> @RynaSr have you tried `./doflash.sh` instead of `./dosdcard.sh`?
> 
> > I had connected micro USB but can't see any COM connection getting open
> 
> I think you need to use the UART pins on the debug header for the UART terminal by default.

Yes.. And that was the main issue.. I flashed SPI and it worked.. It appears there was compatibility issue with 4.x with 5.x images..