OE4T/meta-tegra #1368: gstreamer crashes on low resolution

housefm666, 2023-09-15:
**Describe the bug**
gstreamer crashes on low resolution (320x240, 640x480)

The pipeline:
```
$ gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=BGRA,width=640,height=480 ! videoconvert ! video/x-raw,format=NV12 ! nvvidconv ! nvoverlaysink sync=false
$ gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=BGRA,width=320,height=240 ! videoconvert ! video/x-raw,format=NV12 ! nvvidconv ! nvoverlaysink sync=false
```

But no problem with higher resolutions:
```
$ gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=BGRA,width=1280,height=1024 ! videoconvert ! video/x-raw,format=NV12 ! nvvidconv ! nvoverlaysink sync=false
$ gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=BGRA,width=1024,height=768 ! videoconvert ! video/x-raw,format=NV12 ! nvvidconv ! nvoverlaysink sync=false
```
The solution to get to work gstreamer again was to remove this patch for nvvidconv:

`0003-Update-allocator-to-use-actual-frame-sizes.patch`

What is the purpose of this patch, surely we need it?

Board: Jetson Nano 2GB
Branches: Where this patch exists and the board is supported

- kirkstone-l4t-r32.7.x
- dunfell
- dunfell-l4t-r32.6.1

Output:
```
root@jetson-nano-2gb-devkit:~# gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=BGRA,width=320,height=240 ! videoconvert ! video/x-raw,format=NV12 ! nvvidconv ! nvoverlaysink sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 2323' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:03.526775571
Setting pipeline to NULL ...
```

dwalkes, 2023-09-15:
> 0003-Update-allocator-to-use-actual-frame-sizes.patch
> What is the purpose of this patch, surely we need it?

Looks like it was originally a part of https://github.com/OE4T/meta-tegra/pull/872 for the PR at https://github.com/OE4T/meta-tegra/pull/864

The comments at https://github.com/OE4T/meta-tegra/pull/864#issuecomment-998080464 might be related.

kekiefer, 2023-09-15:
It looks like this will take some care to unwind because there's a few ways these can get used. Be sure to check that the nvcompositor works with current gstreamer versions if you're proposing to drop the patch. I'll keep an eye on this thread for updates.

housefm666, 2023-09-16:
Thank you for the info, I can confirm that GStreamer compiles and works with the above-mentioned commands without the patch on these branches:

[dunfell-l4t-r32.6.1](https://github.com/OE4T/meta-tegra/tree/dunfell-l4t-r32.6.1)
[dunfell](https://github.com/OE4T/meta-tegra/tree/dunfell)
[kirkstone-l4t-r32.7.x](https://github.com/OE4T/meta-tegra/tree/kirkstone-l4t-r32.7.x)

madisongh, 2023-09-16:
So the script mentioned [here](https://github.com/OE4T/meta-tegra/pull/864#issuecomment-988162362) works OK without the patch?

madisongh, 2023-09-17:
Ran some tests today, and we **do** need to retain the patch for the `nvcompositor` plugin.

The SIGSEGV is occurring in the `nvoverlaysink` plugin, one of the OpenMAX plugins.  It has some questionable logic in it around whether it should be treating incoming buffers as `NvBufSurface` structures or actual frames, deciding based on the size of the buffer.  So this looks like another manifestation of  #907, which is unlikely to get fixed any time soon.

If you really need to use that particular plugin, you could downgrade your gstreamer recipes to the 1.14 version by adding the [meta-tegra/contrib](https://github.com/OE4T/meta-tegra/tree/kirkstone-l4t-r32.7.x/contrib) layer and following the instructions in the README.md file there.

housefm666, 2023-09-18:
> So the script mentioned [here](https://github.com/OE4T/meta-tegra/pull/864#issuecomment-988162362) works OK without the patch?

I lost my focus on this thread, sorry.

So my test seems to be OK:
```
root@jetson-nano-2gb-devkit:~# WIDTH=320
Troot@jetson-nano-2gb-devkit:~# HEIGHT=240
Croot@jetson-nano-2gb-devkit:~# CAPS="video/x-raw, width=$WIDTH, height=$HEIGHT"
root@jetson-nano-2gb-devkit:~# POSITIONING="\
_> sink_0::xpos=0 sink_0::ypos=0 sink_0::width=$WIDTH sink_0::height=$HEIGHT \
_> sink_1::xpos=$WIDTH sink_1::ypos=0 sink_1::width=$WIDTH sink_1::height=$HEIGHT \
2> sink_2::xpos=0 sink_2::ypos=$HEIGHT sink_2::width=$WIDTH sink_2::height=$HEIGHT \
3> sink_3::xpos=$WIDTH sink_3::ypos=$HEIGHT sink_3::width=$WIDTH sink_3::height=$HEIGHT \
> "
root@jetson-nano-2gb-devkit:~# gst-launch-1.0 -v nvcompositor name=comp $POSITIONING ! fakesink \
v> videotestsrc is-live=true ! $CAPS ! nvvidconv ! comp. \
> videotestsrc is-live=true ! $CAPS ! nvvidconv ! comp. \
> videotestsrc is-live=true ! $CAPS ! nvvidconv ! comp. \
> videotestsrc is-live=true ! $CAPS ! nvvidconv ! comp.
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc1.GstPad:src: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter1.GstPad:src: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc2.GstPad:src: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter2.GstPad:src: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/Gstnvvconv:nvvconv1.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/Gstnvvconv:nvvconv0.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/GstNvCompositor:comp.GstNvCompositorPad:sink_1: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/GstNvCompositor:comp.GstNvCompositorPad:sink_1: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/Gstnvvconv:nvvconv0.GstPad:sink: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/Gstnvvconv:nvvconv1.GstPad:sink: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/Gstnvvconv:nvvconv1.GstPad:sink: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter1.GstPad:sink: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/Gstnvvconv:nvvconv2.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/GstNvCompositor:comp.GstNvCompositorPad:sink_2: caps = video/x-raw(memory:NVMM), width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, format=(string)RGBA
/GstPipeline:pipeline0/Gstnvvconv:nvvconv2.GstPad:sink: caps = video/x-raw, format=(string)RGBA, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
[... cut ...]
/GstPipeline:pipeline0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw(memory:NVMM), format=(string)RGBA, width=(int)640, height=(int)480, framerate=(fraction)30/1, colorimetry=(string)sRGB
Redistribute latency...
Redistribute latency...
[   35.977228] vdd-fan: disabling
[   35.980566] vdd-usb-vbus: disabling
[   35.984142] vdd-usb-vbus2: disabling
[   35.987777] vddio-sdmmc3-ap: disabling
[   35.991631] avdd-io-edp-1v05: disabling
[   35.995486] vdd-usb-hub-en: disabling
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:17.184144629
Setting pipeline to NULL ...
Freeing pipeline ..
```

housefm666, 2023-09-18:
> If you really need to use that particular plugin, you could downgrade your gstreamer recipes to the 1.14 version by adding the [meta-tegra/contrib](https://github.com/OE4T/meta-tegra/tree/kirkstone-l4t-r32.7.x/contrib) layer and following the instructions in the README.md file there.

Thank you for the comment, I used PREFERRED_VERSION variable in my local conf previously, but it is better.

closed 2023-09-24 by madisongh