OE4T/meta-tegra #1372: kernel backtrace and reboot loop on kirkstone-l4t-r32.7.x branch, because of weston user isn't the root

housefm666, 2023-09-16:
**Describe the bug**

I finally figured out the problem, but I don't know how to fix it on meta-tegra layer level and what is in the background.

I had to rollback weston user in the init script as it is in dunfell:
`sed 's/WESTON_USER=weston/WESTON_USER=root/1' -i poky-kirkstone/meta/recipes-graphics/wayland/weston-init/init`

Environment: Nano 2GB and sysvinit

See the [kirkstone_crash.txt](https://github.com/OE4T/meta-tegra/files/12641667/kirkstone_crash.txt) and this:

```
[   10.762451] podgov: can't create debugfs directory
[   10.767304] Kernel panic - not syncing: nvhost_scale_emc_debug_init
[   10.773559] CPU: 1 PID: 2119 Comm: weston Not tainted 4.9.299-l4t-r32.7.3+gff65a92ec55e #1
[   10.781804] Hardware name: NVIDIA Jetson Nano 2GB Developer Kit (DT)
[   10.788142] Call trace:
[   10.790589] [<ffffff800808c184>] dump_backtrace+0x0/0x16c
[   10.795979] [<ffffff800808c318>] show_stack+0x28/0x34
[   10.801022] [<ffffff8008fbca10>] dump_stack+0x98/0xbc
[   10.806063] [<ffffff8008fb9a84>] panic+0x120/0x284
[   10.810850] [<ffffff8008cf9648>] nvhost_scale_emc_debug_init.isra.0+0x1c4/0x1dc
[   10.818145] [<ffffff8008cf9844>] nvhost_pod_init+0x1e4/0x274
[   10.823796] [<ffffff8008cf99cc>] nvhost_pod_event_handler+0xf8/0x17c
[   10.830146] [<ffffff8008cf5c58>] devfreq_add_device.part.0+0x324/0x410
[   10.836667] [<ffffff8008cf6b98>] devm_devfreq_add_device+0x78/0xe0
[   10.843821] [<ffffff8001035d8c>] gk20a_scale_init+0xf8/0x17c [nvgpu]
[   10.851149] [<ffffff800102f38c>] gk20a_pm_finalize_poweron+0x36c/0x400 [nvgpu]
[   10.859601] [<ffffff800102f5e0>] gk20a_pm_runtime_resume+0x40/0x74 [nvgpu]
[   10.866527] [<ffffff80087b5cc0>] pm_generic_runtime_resume+0x40/0x60
[   10.872914] [<ffffff80087b80c8>] __rpm_callback+0x78/0xa0
[   10.878334] [<ffffff80087b8128>] rpm_callback+0x38/0xa0
[   10.883584] [<ffffff80087b8b2c>] rpm_resume+0x4b8/0x710
[   10.888836] [<ffffff80087b9e44>] __pm_runtime_resume+0x50/0x7c
[   10.896516] [<ffffff800102f530>] gk20a_busy+0x110/0x180 [nvgpu]
[   10.904255] [<ffffff800100f92c>] gk20a_ctrl_dev_open+0x8c/0x160 [nvgpu]
[   10.910896] [<ffffff800826bc78>] chrdev_open+0xb8/0x224
[   10.916174] [<ffffff8008261e60>] do_dentry_open+0x190/0x360
[   10.921814] [<ffffff800826349c>] vfs_open+0x58/0x80
[   10.926758] [<ffffff800827751c>] do_last+0x28c/0x810
[   10.931782] [<ffffff8008278790>] path_openat+0x90/0x1c0
[   10.937070] [<ffffff8008279be4>] do_filp_open+0x74/0x100
[   10.942449] [<ffffff8008263908>] do_sys_open+0x154/0x23c
[   10.947814] [<ffffff8008263a80>] SyS_openat+0x40/0x50
[   10.952928] [<ffffff8008083900>] el0_svc_naked+0x34/0x38
[   10.958299] SMP: stopping secondary CPUs
[   10.962662] Kernel Offset: disabled
[   10.966217] Memory Limit: none
[   11.053248] Rebooting in 5 seconds..
```

madisongh, 2023-09-17:
It's highly unlikely that the weston user being non-root is the problem here.  If you're mixing the 32.7.3 kernel with R32.7.4 userland BSP support, that's a possible reason.  Another is that weston is starting before nvpmodel and nvphsd have had a chance to run.

housefm666, 2023-09-18:
Thank you for the hint, the problem is that the weston starts too early, if I move S09weston -> S23weston, it can also solve the problem. Probably the weston can not (auto) load the GPU driver as weston user?

housefm666, 2023-09-18:
With your last commits on kirkstone-l4t-r32.7.x branch as you rearranged the init scripts, this issue was also solved, it can be closed. Many thanks!

closed 2023-09-18 by madisongh