OE4T/meta-tegra #1373: Unable to recover from blankscreen/powersave on kirkstone-l4t-r32.7.x

housefm666, 2023-09-17:
**Describe the bug**

On the kirkstone-l4t-r32.7.x, after idle-time (default 300 sec) the system goes into power-saving mode, and we do not get the screen back by pressing the keyboard or by clicking or moving the mouse.

Kernel message:
```
[   21.260121] tegradc tegradc.0: blank - powerdown
[   21.325419] extcon-disp-state extcon:disp-state: cable 47 state 0 already set.
[   21.325421] Extcon AUX1(HDMI) disable
[   33.932101] vdd-fan: disabling
[   33.935566] vdd-usb-vbus: disabling
[   33.939260] vdd-usb-vbus2: disabling
[   33.942981] vddio-sdmmc3-ap: disabling
[   33.946928] avdd-io-edp-1v05: disabling
[   33.950864] vdd-usb-hub-en: disabling
```

For testing purposes, you can shorten the idle-time, by adding this parameter:
```
/etc/xdg/weston/weston.ini
[core]
idle-time=10
```

Under Dunfell, this problem does not exist, but there the system does not go into power-saving mode, only we have a blank screen.

housefm666, 2023-09-18:
On dunfell:

```
[ 1102.957007] tegradc tegradc.0: blank - normal
```

then it can wakeup:
```
[ 1606.315842] tegradc tegradc.0: unblank
[ 1606.349425] tegradc tegradc.0: sync windows ret = 242
```

dwalkes, 2023-09-19:
This looks similar to something only I've been able to reproduce on Jetpack 5 so far in https://github.com/OE4T/meta-tegra/issues/1069

housefm666, 2023-09-19:
I tested it on [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) with the same result. I tried to run `xset --display display:0 -q` but the command was missing, then I added it, and told me no display was found.