OE4T/meta-tegra #1376: Kernel panic when reading the fuses through the nvmem sysfs interface

shrimpyoda, 2023-09-19:
Hi all

As mentioned in the title, I'm getting kernel panics when reading the fuse values through the nvmem sysfs interface. My build is based on the kirkstone branch and latest commit d92737fee5282a2604fde64f211f94ada6d35302, and I'm running it on an agx orin module.

The kernel panic doesn't cause the system to reboot and the messages stops after about ~30 seconds, which is around the same time the `cat` command is taking to return.

To reproduce one can just issue `cat /sys/bus/nvmem/devices/fuse/nvmem | hexdump` and check the kernel logs.

```shell
[  489.188239] CPU:0, Error: cbb-fabric@0x13a00000, irq=34
[  489.188365] **************************************
[  489.188483] CPU:0, Error:cbb-fabric, Errmon:2
[  489.190167] 	  Error Code		: SLAVE_ERR

[  489.195677] 	  Error Code		: SLAVE_ERR
[  489.199615] 	  MASTER_ID		: CCPLEX
[  489.202939] 	  Address		: 0x381031c
[  489.206528] 	  Cache			: 0x1 -- Bufferable 
[  489.210551] 	  Protection		: 0x2 -- Unprivileged, Non-Secure, Data Access
[  489.217378] 	  Access_Type		: Read
[  489.220702] 	  Access_ID		: 0x12
[  489.220704] 	  Fabric		: cbb-fabric
[  489.227614] 	  Slave_Id		: 0x1e
[  489.230588] 	  Burst_length		: 0x0
[  489.233913] 	  Burst_type		: 0x1
[  489.237326] 	  Beat_size		: 0x2
[  489.240477] 	  VQC			: 0x0
[  489.243276] 	  GRPSEC		: 0x7e
[  489.246427] 	  FALCONSEC		: 0x0
[  489.249578] 	**************************************
[  489.254402] ------------[ cut here ]------------
[  489.259035] WARNING: CPU: 0 PID: 2507 at drivers/soc/tegra/cbb/tegra234-cbb.c:577 tegra234_cbb_isr+0x130/0x170
[  489.361929] CPU: 0 PID: 2507 Comm: cat Tainted: G        W  O      5.10.120-l4t-r35.4.ga+g76678311c10b #1
[  489.371639] Hardware name: Unknown Jetson AGX Orin Developer Kit/Jetson AGX Orin Developer Kit, BIOS v35.4.1 08/04/2023
[  489.382665] pstate: 60400089 (nZCv daIf +PAN -UAO -TCO BTYPE=--)
[  489.388702] pc : tegra234_cbb_isr+0x130/0x170
[  489.393165] lr : tegra234_cbb_isr+0x10c/0x170
[  489.397627] sp : ffff800010003e60
[  489.401040] x29: ffff800010003e60 x28: ffff0919448ebb00 
[  489.406550] x27: 0000000000000001 x26: 0000000000000080 
[  489.412064] x25: ffffb474a6141d28 x24: ffffb474a6a52e30 
[  489.417575] x23: ffffb474a63f6008 x22: 0000000000000022 
[  489.422913] x21: ffffb474a6878368 x20: ffffb474a6878358 
[  489.428426] x19: 0000000000000002 x18: ffffffffffffffff 
[  489.433852] x17: 000000000033faa9 x16: ffffb474a5c18590 
[  489.439277] x15: ffff800090003977 x14: 0000000000000006 
[  489.444788] x13: ffff80001000397f x12: 0000000000000010 
[  489.450126] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[  489.455638] x9 : ffffb474a6707a00 x8 : 0000000100008ab3 
[  489.461064] x7 : ffffb474a675fa00 x6 : 000000000003ad88 
[  489.466489] x5 : ffff09206cef5958 x4 : 0000000000000ab3 
[  489.471914] x3 : 0000000000000000 x2 : ffffb474a4c312e0 
[  489.477250] x1 : ffff0919448ebb00 x0 : 0000000100010000 
[  489.482589] Call trace:
[  489.485041]  tegra234_cbb_isr+0x130/0x170
[  489.489067]  __handle_irq_event_percpu+0x68/0x2a0
[  489.493790]  handle_irq_event+0x6c/0x150
[  489.497551]  handle_fasteoi_irq+0xac/0x1f4
[  489.501577]  __handle_domain_irq+0x88/0xf0
[  489.505603]  gic_handle_irq+0xd0/0x150
[  489.509538]  el1_irq+0xd0/0x180
[  489.512517]  __memcpy+0x98/0x180
[  489.515752]  tegra_bpmp_clk_unprepare+0x50/0xb0
[  489.520479]  clk_core_unprepare+0x94/0x260
[  489.524502]  clk_unprepare+0x38/0x50
[  489.528002]  tegra30_fuse_read+0x60/0xac
[  489.531765]  tegra_fuse_read+0x60/0x84
[  489.535707]  bin_attr_nvmem_read+0xa8/0xe0
[  489.539730]  sysfs_kf_bin_read+0x70/0xb0
[  489.543488]  kernfs_fop_read_iter+0xa8/0x1d0
[  489.547780]  new_sync_read+0xf8/0x18c
[  489.551277]  vfs_read+0x134/0x1c0
[  489.554515]  ksys_read+0x78/0x100
[  489.557927]  __arm64_sys_read+0x28/0x34
[  489.561866]  el0_svc_common.constprop.0+0x80/0x1c4
[  489.566764]  do_el0_svc+0x74/0x8c
[  489.570178]  el0_svc+0x1c/0x2c
[  489.573240]  el0_sync_handler+0x9c/0x120
[  489.577001]  el0_sync+0x16c/0x180
[  489.580413] ---[ end trace b43c41e80a9a361d ]---
```

madisongh, 2023-09-20:
Does this also happen with stock JetPack/L4T?  I would guess so, given the nature of the error - it's the CBB telling you that there was some error trying to access the fuse peripheral on the internal bus.  I don't know much about how that works, or even if the nvmem-based fuse driver is supported in the L4T kernel.  You might get more help posting a question about it over on the NVIDIA developer forums.

If you don't see this traceback with stock L4T but you do with your Yocto builds, we'll need to figure out what's different between the two that might be causing it to occur.  I can't think of anything that would, though, off the top of my head.

shrimpyoda, 2023-09-21:
Yes, I just checked and got the same behaviour with the stock JetPack. I will reach out to NVIDIA. Thanks @madisongh!

closed 2023-11-01 by madisongh