OE4T/meta-tegra #1389: Support enabling rootfs A/B redundancy

Lexmark-chad, 2023-10-04:
Many will want to enable rootfs A/B redundancy for their builds.  It makes sense that `meta-tegra` supports this directly.  Per a comment from @madisongh in the OE4T gitter chat:

"adding a variable to control that, and maybe also having it automatically default to using the redundant flash layout (assuming the NVIDIA-provided ones are usable for Yocto builds) should simplify things."

Tracking the work so that hopefully someone can eventually get around to this.

eg9152, 2023-11-27:
I plan to soon jump into A/B partitions in support of field upgrades of our devices.  On TX2 we used our own solution with u-boot scripts and an extra root partition on TX2.  Now with Orin and Xavier not having u-boot as an option, I will be looking at NVIDIA's "built-in" A/B booting and associated cboot support.

Posting this in case anyone has already started work on this and I'm duplicating effort.  I'll update here when I make progress on it and maybe discuss what would make sense in meta-tegra.

dwalkes, 2023-11-28:
I have a swupdate implementation for Jetpack 5 A/B with rootfs and capsule updates working on a private repo.  I'm hoping to separate out the parts which could go to meta-tegra-community this week and put together a version I could share publicly.

The swupdate implementation was mostly just putting together notes from several sources, most of them on https://matrix.to/#/#OE4T_jetpack-5-device-update:gitter.im or linked from there.

dwalkes, 2023-12-02:
I've got a PR started at https://github.com/OE4T/meta-tegra/pull/1428 for this, would appreciate any help testing on different platforms/configs or replacing any custom MACHINE implementations for rootfs a/b support already out there.