OE4T/meta-tegra #1401: populate_sdk with deepstream-6.3 and tegra-mmapi on kirkstone branch

aurelien-enchanted-tools, 2023-10-16:
Hello,

branch: kirkstone

When we include both deepstream-6.3 and tegra-mmapi in the image, and we generate a sdk, we obtain:
```
Downloading file:/home/yoctouser/build/tmp/work/custom_board-custom-linux * check_data_file_clashes: Package deepstream-6.3-dev wants to install file /home/yoctouser/build/tmp/work/custom_board-custom-linux/custom-image/1.0-r0/sdk/image/usr/local/oe-sdk-hardcoded-buildpath/sysroots/armv8a-custom-linux/usr/include/nvbufsurface.h
	But that file is already provided by package  * tegra-mmapi-dev
 * check_data_file_clashes: Package deepstream-6.3-dev wants to install file /home/yoctouser/build/tmp/work/custom_board-custom-linux/custom-image/1.0-r0/sdk/image/usr/local/oe-sdk-hardcoded-buildpath/sysroots/armv8a-custom-linux/usr/include/nvbufsurftransform.h
	But that file is already provided by package  * tegra-mmapi-dev
/custom-image-x11/1.0-r0/oe-sdk-repo/armv8a_tegra/gstreamer1.0-plugins-nvcompositor-dev_1.20.5-r35.4.1-r0_armv8a_tegra.ipk.
```

bchoineubility, 2023-10-17:
Hi,

I got a same issue when I do populate_ext_sdk also.

`ERROR: ros-image-world-1.0-r0 do_sdk_depends: The file /usr/include/nvbufsurftransform.h is installed by both tegra-mmapi and deepstream-6.2, aborting
ERROR: Logfile of failure stored in: /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/jetson_agx_xavier_devkit-poky-linux/ros-image-world/1.0-r0/temp/log.do_sdk_depends.813214
ERROR: Task (/home/bchoi/nvidia-yocto-bsp/sources/meta-ros/meta-ros-common/recipes-core/images/ros-image-world.bb:do_sdk_depends) failed with exit code '1'`

madisongh, 2023-10-17:
I've fixed this in the deepstream recipe meta-tegra-community, on master, mickledore, and kirkstone branches.

closed 2023-10-17 by madisongh

bchoineubility, 2023-10-17:
> I've fixed this in the deepstream recipe meta-tegra-community, on master, mickledore, and kirkstone branches.

Many thanks !!!

bchoineubility, 2023-10-18:
Hi ,

This issue would remain in the deepstream-6.2 also.

Can you  apply to deepstream-6.2 or do I backport it to deepstream-6.2 ?