OE4T/meta-tegra #1410: File Exists error on populate_sdk_ext

bchoineubility, 2023-11-08:
Hi, 
I am trying to extract eSDK with core-image-sato, but, FileExistsError is occurred.

```
ERROR: core-image-sato-1.0-r0 do_sdk_depends: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:extend_recipe_sysroot(d)
     0003:
File: '/home/bchoi/work/tegra-only_build/sources/poky/meta/classes/staging.bbclass', lineno: 612, function: extend_recipe_sysroot
     0608:                    if "/bin/" in l or "/sbin/" in l:
     0609:                        # defer /*bin/* files until last in case they need libs
     0610:                        binfiles[l] = (targetdir, dest)
     0611:                    else:
 *** 0612:                        staging_copyfile(l, targetdir, dest, postinsts, seendirs)
     0613:
     0614:    # Handle deferred binfiles
     0615:    for l in binfiles:
     0616:        (targetdir, dest) = binfiles[l]
File: '/home/bchoi/work/tegra-only_build/sources/poky/meta/classes/staging.bbclass', lineno: 158, function: staging_copyfile
     0154:        os.symlink(linkto, dest)
     0155:        #bb.warn(c)
     0156:    else:
     0157:        try:
 *** 0158:            os.link(c, dest)
     0159:        except OSError as err:
     0160:            if err.errno == errno.EXDEV:
     0161:                bb.utils.copyfile(c, dest)
     0162:            else:
Exception: FileExistsError: [Errno 17] File exists: '/home/bchoi/work/tegra-only_build/bchoi-build/tmp/sysroots-components/armv8a_tegra/tegra-mmapi/usr/include/nvbufsurftransform.h' -> '/home/bchoi/work/tegra-only_build/bchoi-build/tmp/work/jetson_agx_xavier_devkit-poky-linux/core-image-sato/1.0-r0/recipe-sysroot/usr/include/nvbufsurftransform.h'

ERROR: Logfile of failure stored in: /home/bchoi/work/tegra-only_build/bchoi-build/tmp/work/jetson_agx_xavier_devkit-poky-linux/core-image-sato/1.0-r0/temp/log.do_sdk_depends.1209635
ERROR: Task (/home/bchoi/work/tegra-only_build/sources/poky/meta/recipes-sato/images/core-image-sato.bb:do_sdk_depends) failed with exit code '1'
```

Then I removed below file manually.
`
/home/bchoi/work/tegra-only_build/bchoi-build/tmp/work/jetson_agx_xavier_devkit-poky-linux/core-image-sato/1.0-r0/recipe-sysroot/usr/include/nvbufsurftransform.h
`

I tried to do populate_sdk_ext again, but, following error is occurred.

```
Initialising tasks: 100% |####################################################################################################################################################################################################################################| Time: 0:00:05
Sstate summary: Wanted 4 Local 0 Mirrors 0 Missed 4 Current 3565 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: core-image-sato-1.0-r0 do_populate_sdk_ext: Failed to generate filtered task list for extensible SDK:

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-full-cmdline
    core-image-sato
    core-image-weston
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86-64'.

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
ERROR: bitbake failed:
WARNING: You have included the meta-virtualization layer, but 'virtualization' has not been enabled in your DISTRO_FEATURES. Some bbappend files may not take effect. See the meta-virtualization README for details on enabling virtualization support.
WARNING: You are using a local hash equivalence server but have configured an sstate mirror. This will likely mean no sstate will match from the mirror. You may wish to disable the hash equivalence use (BB_HASHSERVE), or use a hash equivalence server alongside the sstate mirror.
Loading cache...done.
Loaded 0 entries from dependency cache.
Parsing recipes...done.
ERROR: Nothing PROVIDES 'tegra-minimal-initramfs'
tegra-minimal-initramfs was skipped: incompatible with machine qemux86-64 (not in COMPATIBLE_MACHINE)
Parsing of 2958 .bb files complete (0 cached, 2958 parsed). 4500 targets, 288 skipped, 0 masked, 0 errors.

Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
ERROR: Logfile of failure stored in: /home/bchoi/work/tegra-only_build/bchoi-build/tmp/work/jetson_agx_xavier_devkit-poky-linux/core-image-sato/1.0-r0/temp/log.do_populate_sdk_ext.1249350
ERROR: Task (/home/bchoi/work/tegra-only_build/sources/poky/meta/recipes-sato/images/core-image-sato.bb:do_populate_sdk_ext) failed with exit code '1'
NOTE: Tasks Summary: Attempted 8395 tasks of which 8388 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bchoi/work/tegra-only_build/sources/poky/meta/recipes-sato/images/core-image-sato.bb:do_populate_sdk_ext
Summary: There was 1 ERROR message, returning a non-zero exit code.
bchoi@nvidia-build:~/work/tegra-only_build/bchoi-build$
```

My local.conf is default configuration plus as following:

```
MACHINE = "jetson-agx-xavier-devkit"

BB_NUMBER_THREADS = "8"
PARALLEL_MAKE  = "-j 8"

IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
DISTRO_FEATURES:append = " virtualization"
DISTRO_FEATURES:append = " security"
DISTRO_FEATURES:append = " seccomp"
DISTRO_FEATURES:append = " nvidia-docker"

PACKAGE_CLASSES = "package_deb"
```

**Besides above**, tegra-espimage and tegra-initrd-flash-initramfs would have the same issue.

BR,
Mark

dwalkes, 2023-11-08:
Hi Mark,
Which branch are you using?

I can't reproduce with [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) nanbield branch, the configuration below and `bitbake core-image-sato -c populate_sdk_ext`

```
BB_VERSION           = "2.6.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe4t-linux"
MACHINE              = "jetson-agx-xavier-devkit"
DISTRO               = "tegrademo"
DISTRO_VERSION       = "4.3+snapshot-bc00caadc9bd5eab854d4b7e6574a2eb9747ac79"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 = "HEAD:bc00caadc9bd5eab854d4b7e6574a2eb9747ac79"
meta-tegra           = "HEAD:bdacbb486159f3814dbe72e3264f2d1d1841bbc1"
meta-oe
meta-python
meta-networking
meta-filesystems     = "HEAD:278c3f75e32f38f71bb52d161fe06bcb6f6bdd2f"
meta-virtualization  = "HEAD:4c8860080355dddaec2af7eec46e1c059c718851"
meta-tegra-community = "HEAD:d6777fb011a1fad9c146c7b3fc573e4827262292"
meta-tegra-support
meta-demo-ci
meta-tegrademo       = "nanbield:eaff838cee805824d5529300856abb5af3efe52e"
```

Please share the branch you are using and whether you can reproduce without any of your changes and with https://github.com/OE4T/tegra-demo-distro

bchoineubility, 2023-11-08:
Hi Dan,

I am using poky distro. Should I add meta-tegra-demo-distro layer also?

```
Loading cache: 100% |################################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 4500 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-agx-xavier-devkit"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.10"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "HEAD:1e6f147c686a91b9a5996f529e32ee1852f946b7"
meta-oe
meta-python
meta-filesystems
meta-networking
meta-multimedia      = "HEAD:f95484417e3d3e65ca15b460ba71dfd35773f0e4"
meta-clang           = "HEAD:71321ddf78ea522b87a6b4bffefb14c988a6d921"
meta-tegra           = "HEAD:2a808b6afb221974196d2eeab3537140ebb63cc7"
meta-qt5             = "HEAD:bff5bd937f0776166e81a63f3dd39ede348ef758"
meta-chromium        = "HEAD:b279a4dc078ba6957815e2496820f1eadbafd58a"
meta-virtualization  = "HEAD:b3b3dbc67504e8cd498d6db202ddcf5a9dd26a9d"
meta-tegra-community = "HEAD:56ce73e5f5a36cea832894946787bd024b71493e"
```

dwalkes, 2023-11-09:
This looks like a duplicate of https://github.com/OE4T/meta-tegra/issues/1401, your meta-tegra-community is out of date with respect to upstream at https://github.com/OE4T/meta-tegra-community/commits/kirkstone and you are missing the backported fixes for deepstream header file collisions in https://github.com/OE4T/meta-tegra-community/pull/101

bchoineubility, 2023-11-09:
Oh  thanks. 

I look into it.

closed 2023-11-09 by bchoineubility