OE4T/meta-tegra #1415: SRCREV is ignored in linux-tegra_4.9.bb

MichaelPres, 2023-11-14:
Hi,

In branch `kirkstone-l4t-r32.7.x`, the SRCREV is ignored:
```
rhc@ubuntu:/mnt/repos/kirkstone-offline/build$ buildhistory-collect-srcrevs | grep linux-tegra
SRCREV_machine:pn-linux-tegra = "a1cf17bc0aa0dbafcd8c43514f57d90dc69442a0"
```

Here is what I propose:
```
diff --git a/recipes-kernel/linux/linux-tegra_4.9.bb b/recipes-kernel/linux/linux-tegra_4.9.bb
index 6528f29..5c2471c 100644
--- a/recipes-kernel/linux/linux-tegra_4.9.bb
+++ b/recipes-kernel/linux/linux-tegra_4.9.bb
@@ -22,7 +22,7 @@ LINUX_VERSION_EXTENSION ?= "-l4t-r${L4T_VERSION}"
 SCMVERSION ??= "y"
 
 SRCBRANCH = "oe4t-patches${LINUX_VERSION_EXTENSION}"
-SRCREV = "a1cf17bc0aa0dbafcd8c43514f57d90dc69442a0"
+SRCREV_machine = "a1cf17bc0aa0dbafcd8c43514f57d90dc69442a0"
 KBRANCH = "${SRCBRANCH}"
 SRC_REPO = "github.com/OE4T/linux-tegra-4.9;protocol=https"
 KERNEL_REPO = "${SRC_REPO}"
```

madisongh, 2023-11-23:
`SRCREV` is not ignored; the value its set to gets propagated to the `SRCREV_machine` variable (since our source repo has `name=machine`).  Since we don't actually use a `meta` repository, we could probably update the recipe to set `SRCREV_FORMAT` to the null string and remove the `name=machine` parameter from the SRC_URI setting.  That would clean up the buildhistory output... I'm just not sure if there any assumptions in the `kernel-yocto` bbclass about there being a source repo tagged with `name=machine` that would break builds if we did that.

MichaelPres, 2024-05-22:
@ichergui I am using the patch locally since @madisongh  say it is not required in previous message. I don't have the time look for SRCREV_FORMAT.

ichergui, 2024-07-22:
closing this issue since no updates. Don't hesitate to reopen new one if the issue still existing

closed 2024-07-22 by ichergui