OE4T/meta-tegra #1417: UEFI crashes for a headless boot (no display connected)

picanteAvi, 2023-11-15:
If there is no HDMI cable connected, UEFI will crash in `T194GraphicsOutputDxe` driver init. Full log is attached but the relevant pieces are here:

NV-DISP snippet
```
[0003.840] I> regulator 'vdd-hdmi-5v0' already enabled
[0003.847] I> regulator 'vdd-hdmi-5v0' already enabled
[0003.847] E> tegrabl_display_init_regulator: hdmi cable is not connected
[0003.848] E> tegrabl_display_get_pdata, failed to parse dtb settings
[0003.849] E> cannot find any other nvdisp nodes
[0003.849] E> no valid display unit config found in dtb
[0003.856] W> display init failed
[0003.857] initializing target
[0003.859] calling apps_init()
[0003.862] starting app kernel_boot_app
[0003.866] I> Kernel type = Normal
```

UEFI snippet:
```
ERROR:   MPIDR 0x80000000: exception reason=0 syndrome=0xbe000000
ERROR:   **************************************
ERROR:   RAS Error in L2, ERRSELR_EL1=0x200:
ERROR:   	Status = 0xfc006612
ERROR:   	IERR = SCF to L2 Slave Error Read: 0x66
ERROR:   SERR = Error response from slave: 0x12
ERROR:   	Overflow (there may be more errors) - Uncorrectable
ERROR:   	Uncorrectable (this is fatal)
ERROR:   	MISC0 = 0x80000000400000
ERROR:   	MISC1 = 0x20240000000
ERROR:   	ADDR = 0x800000001520101c
ERROR:   **************************************
ERROR:   **************************************

ERROR:   RAS Error in SCF_IOB, ERRSELR_EL1=0x401:
ERROR:   	Status = 0xf4009604
ERROR:   	IERR = CBB Interface Error: 0x96
ERROR:   SERR = Assertion failure: 0x4
ERROR:   	Uncorrectable (this is fatal)
ERROR:   	MISC0 = 0x40
ERROR:   	MISC1 = 0x264a444461
ERROR:   	ADDR = 0x800000001520101c
ERROR:   **************************************
ERROR:   RAS error handled!
ERROR:   sdei_dispatch_event returned -1
Head index=0 DC_DISP_SYNC_WIDTH  = 0x00000000
```

It seems like UEFI is trying to access some display control registers that are not inited after nvdisp hands over control? My workaround was to add a patch to the nvdisp binary. Basically, it will return a no-error when displays are not found, let the initialization complete, then fail when trying to access the controller.
This gives the following log output for NV-DISP
```
[0003.763] I> regulator 'vdd-hdmi-5v0' already enabled
[0003.769] I> regulator 'vdd-hdmi-5v0' already enabled
[0003.770] E> tegrabl_display_init_regulator: hdmi cable is not connected
[0003.770] I> retrieved tmds range from prod_list_hdmi_soc
[0003.771] E> cannot find any other nvdisp nodes
[0003.777] E> I2C: slave not found in slaves.
[0003.778] E> I2C: Could not write 0 bytes to slave: 0x00a0 with repeat start true.
[0003.786] E> I2C_DEV: Failed to send register address 0x00000000.
[0003.791] E> I2C_DEV: Could not read 128 registers of size 1 from slave 0xa0 at 0x00000000 via instance 3.
[0003.801] E> could not read edid
[0003.809] I> hdmi_enable, starting HDMI initialisation
[0003.813] I> hdmi_enable, HDMI initialisation complete
[0003.822] initializing target
[0003.823] calling apps_init()
[0003.824] starting app kernel_boot_app
[0003.825] I> Kernel type = Normal
```
This seems hacky and there is probably a better way to fix it, whatever that might be.

[0006-allow-headless-boot.patch](https://github.com/OE4T/meta-tegra/files/13369699/0006-allow-headless-boot.patch)
[uefi-crash-when-display-not-inited.log](https://github.com/OE4T/meta-tegra/files/13369700/uefi-crash-when-display-not-inited.log)

dwalkes, 2023-11-17:
Can you reproduce with https://github.com/OE4T/tegra-demo-distro ?  If so, which `MACHINE`, branch, and image?

ichergui, 2024-05-22:
Hi @picanteAvi 
Any updates on this issue ? 
Otherwise I will close it.