OE4T/meta-tegra #1437: Using earlycon in the cmdline.

bchoineubility, 2023-12-12:
Hi,

I have a question.

If I add earlycon  to extlinux.conf, can I see logs before boot logs on the fb console?

I hope to know exact usage of earlyconf for tegra 194.

BR,
Mark

madisongh, 2023-12-14:
I don't think the kernel's earlycon support works for framebuffer consoles - it's too early in the boot for that, and you usually specify a UART for the earlycon output.

bchoineubility, 2023-12-14:
@madisongh 

Thanks.  can I assign earlycon to other uart, for example uart 1?

madisongh, 2023-12-16:
I've never tried it, but in theory it should be possible.  You might want to check the NVIDIA dev forums... there have been many threads there on the topic of using alternate UARTs for console output.

bchoineubility, 2023-12-17:
@madisongh 

OK thanks. 

BR,
Mark

closed 2023-12-17 by bchoineubility