OE4T/meta-tegra #1440: libgl-mesa meta-tegra and meta-intel-realsense

aurelien-enchanted-tools, 2023-12-12:
Hello,

Branch mickledore

I obtain an error when I use together meta-tegra and meta-intel-realsense to build librealsense2-graphical-examples, I obtain:
```
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - conflicting requests
 *   - nothing provides libgl-mesa needed by librealsense2-graphical-examples-2.53.1-r0.armv8a_tegra194
 * 
 * Solution 1:
 *   - do not ask to install a package providing librealsense2-graphical-examples
```
I found a workaround by revert 4cc19319d2ff0d215203c2a599a7832cd8d7f800 in meta-tegra. 

Thank you in advance.

madisongh, 2023-12-14:
Try just dropping that `libgl-mesa` reference from the RDEPENDS for the examples package.  Not sure why they put it there in the first place, looks like it might be just an historical artifact.  These days, there is better support in OE-Core for GL providers other than mesa, and the build should automatically create the needed runtime dependencies based on what the examples are linking against.

aurelien-enchanted-tools, 2024-02-29:
This works also, thank you.