OE4T/meta-tegra #1444: Allow PR modification without effect on other recipes

aurelien-enchanted-tools, 2023-12-17:
Hello World, 

branch: mickledore

Could you remove the PR number from `${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}/Linux_for_Tegra` to allow to PR increment in bbappends without have to create other empty bbappends just to increment their PR to be compatible. 
For example, if I bbappend the `tegra-binaries_35.4.1.bb` and I increment the PR of this recipe via the bbappend, I have to create also bbappends for `tegra-bootfiles_35.4.1.bb`, `tegra-redundant-boot-rollback_35.4.1.bb`, and so on, to allow a correct build. If we remove `-${PR}` in `${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}/Linux_for_Tegra`, the bbappend's PR in `tegra-binaries_35.4.1.bbappend` has not effect on other recipes.

```diff
diff --git a/recipes-bsp/tegra-binaries/tegra-binaries-35.4.1.inc b/recipes-bsp/tegra-binaries/tegra-binaries-35.4.1.inc
index 7b5f904c..b061dd7f 100644
--- a/recipes-bsp/tegra-binaries/tegra-binaries-35.4.1.inc
+++ b/recipes-bsp/tegra-binaries/tegra-binaries-35.4.1.inc
@@ -15,7 +15,7 @@ do_convert_crlf_to_lf () {
 	find ${S} -type f "-(" -name "*.cfg" -o -name "*.dts*" "-)" -exec dos2unix {} \;
 }
 
-S = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}/Linux_for_Tegra"
+S = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}/Linux_for_Tegra"
 B = "${WORKDIR}/build"
 
 INHIBIT_PACKAGE_STRIP = "1"
diff --git a/recipes-bsp/tegra-binaries/tegra-binaries_35.4.1.bb b/recipes-bsp/tegra-binaries/tegra-binaries_35.4.1.bb
index 6980816e..55b545a2 100644
--- a/recipes-bsp/tegra-binaries/tegra-binaries_35.4.1.bb
+++ b/recipes-bsp/tegra-binaries/tegra-binaries_35.4.1.bb
@@ -4,9 +4,9 @@ SECTION = "base"
 
 require tegra-binaries-${PV}.inc
 
-WORKDIR = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}"
+WORKDIR = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}"
 SSTATE_SWSPEC = "sstate:tegra-binaries::${PV}:${PR}::${SSTATE_VERSION}:"
-STAMP = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}"
+STAMP = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}"
 STAMPCLEAN = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-*"
 
 INHIBIT_DEFAULT_DEPS = "1"
diff --git a/recipes-bsp/tegra-binaries/tegra-pinmux-dts2cfg_35.4.1.bb b/recipes-bsp/tegra-binaries/tegra-pinmux-dts2cfg_35.4.1.bb
index 660e420c..cdd19392 100644
--- a/recipes-bsp/tegra-binaries/tegra-pinmux-dts2cfg_35.4.1.bb
+++ b/recipes-bsp/tegra-binaries/tegra-pinmux-dts2cfg_35.4.1.bb
@@ -16,12 +16,12 @@ BBCLASSEXTEND = "native nativesdk"
 
 # Need to setup workdir like set in include file after extending to native
 # Also see tegra-flashtools-native for similar setup.
-WORKDIR = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}"
+WORKDIR = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}"
 SSTATE_SWSPEC = "sstate:tegra-binaries-${L4T_BSP_ARCH}::${PV}:${PR}::${SSTATE_VERSION}:"
-STAMP = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}"
+STAMP = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}"
 STAMPCLEAN = "${STAMPS_DIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-*"
 
-S = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}/Linux_for_Tegra"
+S = "${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}/Linux_for_Tegra"
 
 do_configure[noexec] = "1"
 do_compile[noexec] = "1"
```

madisongh, 2023-12-17:
Why are you updating PR with your bbappends?  Task hashes should change, triggering rebuilds of the dependent recipes automatically, if your bbappend updates the downloaded BSP in a significant way.  If that isn't happening, it might be better to investigate from that end of things first.

aurelien-enchanted-tools, 2023-12-18:
Maybe I’m telling a story, but it seems that hashes do not take into account the files accompanying the recipes. So, in my workflow I prefer increment PR when I do an MR.

madisongh, 2023-12-18:
> task hashes do not take into account the files accompanying the recipes

That used to be the case, but things changed a few years back to fix that (search for `file-checksums` in OE-Core).

I'll admit that the relationship between the tegra-binaries recipe and the others that consume its files is a bit weird, which is why I 
 worked on restructuring the recipes to reduce the need for adding a bbappend to the tegra-binaries recipe.   If there are specific recipes in `recipes-bsp/tegra-binaries` that I haven't done that for yet, which you need to modify, I'd prefer making that sort of change.

In the short run, though, you could have all your bbappend files do something like `require my-tegra-binaries-pr-bump.inc`, instead of modifying PR directly themselves.  That way, you'd only have to do the PR change in one place each time you make a change.  I've done something similar for other interdependent groups of recipes in the past.  You'd still have a bunch of bbappends, of course.

closed 2024-03-24 by madisongh