OE4T/meta-tegra #1461: [Xavier NX DEVKIT emmc] Can't install deepstream-6.3 without x11 Kirkstone Branch

Islam-Hussein-11, 2024-01-19:
**Describe the bug**
I can't build deepstream-6.3 without x11. Is there a way to install it without x11  ?

**To Reproduce**
DISTRO_FEATURES:remove = " x11"
bitbake deepstream-6.3

**Additional context**

ERROR: deepstream-6.3-6.3.0-1-r0 do_package_qa: QA Issue: /opt/nvidia/deepstream/deepstream-6.3/lib/libnvds_3d_glprocess.so contained in package deepstream-6.3 requires libX11.so.6()(64bit), but no providers found in RDEPENDS:deepstream-6.3? [file-rdeps]
ERROR: deepstream-6.3-6.3.0-1-r0 do_package_qa: QA Issue: /usr/bin/deepstream-transfer-learning-app contained in package deepstream-6.3-samples requires libX11.so.6()(64bit), but no providers found in RDEPENDS:deepstream-6.3-samples? [file-rdeps]
ERROR: deepstream-6.3-6.3.0-1-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /yocto/build/tmp/work/armv8a_tegra-poky-linux/deepstream-6.3/6.3.0-1-r0/temp/log.do_package_qa.3132
NOTE: recipe deepstream-6.3-6.3.0-1-r0: task do_package_qa: Failed
ERROR: Task (/yocto/sources/poky/../meta-tegra-community/recipes-devtools/deepstream/deepstream-6.3_6.3.0-1.bb:do_package_qa) failed with exit code '1'
NOTE: recipe deepstream-6.3-6.3.0-1-r0: task do_package_write_ipk: Succeeded
NOTE: Tasks Summary: Attempted 2154 tasks of which 2136 didn't need to be rerun and 1 failed.

madisongh, 2024-01-20:
NVIDIA provides most of their stuff in pre-built binary form, and it's built on their Jetson Linux flavor of Ubuntu, which has X11 support.  So dependencies like this are pretty typical, and we don't have any control over it.

It might be technically possible to modify the deepstream recipe to package the OpenGL/X11-dependent binaries separately (or just not install them at all). However, that might remove some features, and the specifics of that separation might change from version to version, so maintaining it could become a problem.  And then you might run into the same issue with the next NVIDIA package you want to include in your images, and have to go through the same analysis for that one.

I long ago gave up on trying to remove `x11` from DISTRO_FEATURES in my Jetson-based projects, which are also headless. It was just too difficult to maintain.  I don't recommend it.

Islam-Hussein-11, 2024-01-20:
Thank you so much for your quick response. I wannt to ask you for hack tip or a cookbook to deal with recipes. I've gone through yocto documentation and nothing provides that deep details which I found in nvidia packages (I'm New to yocto). So any recommendations ?

madisongh, 2024-02-02:
The Yocto Project site has [other sources](https://www.yoctoproject.org/community/learn/) for learning materials that might help, and our [wiki](https://github.com/OE4T/meta-tegra/wiki) has some information.  You can also ask questions here or in our gitter/Matrix channel, the link for which is on the wiki home page.

closed 2024-03-24 by madisongh