OE4T/meta-tegra #1462: 0001-Make-plugin-directory-relative-to-ORIGIN.patch broken due to meta-oe upgrade of v4l-utils to 1.26.1. autotools replaced with meson

kk6ho, 2024-01-20:
**Description**

1. v4l-utils was upgraded in master branch of meta-oe from 1.24 to 1.26.1. It's build system changed from Autotools to Meson. @kraj 's mucl related update to  `0001-Make-plugin-directory-relative-to-ORIGIN.patch` 45bf0e8e ("v4l-utils: Use RTLD_DI_ORIGIN when its available", 2023-12-13)  fails to apply hunk 4.  It tries to modify `configure.ac` which no longer exists.

2. meta-tegra has 2 copies of the patch in these locations:

`external/openembedded-layer/recipes-multimedia/v4l2apps/v4l-utils/0001-Make-plugin-directory-relative-to-ORIGIN.patch`
`recipes-multimedia/libv4l2/libv4l2-minimal/0001-Make-plugin-directory-relative-to-ORIGIN.patch`

They were identical in nanbield, but now different in master.  Should they stay identical?

**To Reproduce**
`bitbake v4l-utils` using the master branches of the meta layers.

```
ERROR: v4l-utils-1.26.1+git-r0 do_patch: Applying patch '0001-Make-plugin-directory-relative-to-ORIGIN.patch' on target directory '/home/geoff/yocto-master/build/tmp/work/armv8a_tegra-oe-linux/v4l-utils/1.26.1+git/git'
CmdError('quilt --quiltrc /home/geoff/yocto-master/build/tmp/work/armv8a_tegra-oe-linux/v4l-utils/1.26.1+git/recipe-sysroot-native/etc/quiltrc push', 0, "stdout: Applying patch 0001-Make-plugin-directory-relative-to-ORIGIN.patch
patching file lib/libv4l2/v4l2-plugin.c
Hunk #1 succeeded at 20 (offset -1 lines).
Hunk #2 succeeded at 51 (offset -1 lines).
Hunk #3 succeeded at 122 (offset -1 lines).
can't find file to patch at input line 103
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/configure.ac
|+++ b/configure.ac
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch 0001-Make-plugin-directory-relative-to-ORIGIN.patch does not apply (enforce with -f)

stderr: ")
ERROR: Logfile of failure stored in: /home/geoff/yocto-master/build/tmp/work/armv8a_tegra-oe-linux/v4l-utils/1.26.1+git/temp/log.do_patch.1598950
ERROR: Task (/home/geoff/yocto-master/build/../meta-openembedded/meta-oe/recipes-multimedia/v4l2apps/v4l-utils_1.26.1.bb:do_patch) failed with exit code '1'

```

kraj, 2024-01-20:
we need to convert HAVE_RTLD_DI_ORIGIN detection logic to a meson option. Or second way would be delete the hunk for configure.ac and  pass` -DRTLD_DI_ORIGIN` via `CFLAGS` in bbappend
```
CFLAGS:append:libc-glibc = " -DRTLD_DI_ORIGIN"
```

kk6ho, 2024-01-20:
I was going to attempt a PR first, but I'm weak on Autotools and less familiar with Meson. Sounds like a good reason for me to read the Meson tutorial. :-)

PR #1463 (2024-01-20): Update v4l-utils bbappend and libv4l2-minimal recipe
to align with the new version in meta-oe.

Fixes #1462.
closed 2024-01-20 by madisongh