OE4T/meta-tegra #1465: tegra-bl.cap & tegra-kernel.cap not generated in "jetson-orin-nano-devkit-nvme"

haizzus, 2024-01-31:
**Describe the bug**
tegra-bl.cap & tegra-kernel.cap not generated in "jetson-orin-nano-devkit-nvme", but they are generated in ""jetson-orin-nano-devkit" target. This applies to all the targets which are built for nvme I believe.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch '...' with `MACHINE` based on 'jetson-orin-nano-devkit-nvme'
2. Build with bitbake argument '...'
3. The capsules are missing in deploy/images folder.

madisongh, 2024-02-02:
Yep, this would be the case for any machine config where the kernel resides on an external device, rather than internal storage.  The BUP payload generator uses the flash layout file for the internal devices only, so no kernel BUP gets created in that case.  Since the BUPs are the input for generating the capsules, no kernel capsule gets created, either.

TBH, I don't think using a UEFI capsule to update your kernel is a good idea.  But I'll look into figuring out a way to solve this, if there are folks that really need this functionality.

dwalkes, 2024-02-02:
> TBH, I don't think using a UEFI capsule to update your kernel is a good idea. 

See the strategy [here](https://github.com/Trellis-Logic/meta-swupdate-boards/pull/1/files#diff-d59cbb1d431e89f757bc07a9240ebca976e9accfb770eb4a871d4bc39b7b2c9cR30-R37) which you can use as an alternative, and which I think matches how a few others are handling kernel updates without capsule update based on discussions in [the device update channel on gitter/matrix](https://matrix.to/#/#OE4T_jetpack-5-device-update:gitter.im)

haizzus, 2024-02-02:
> See the strategy [here](https://github.com/Trellis-Logic/meta-swupdate-boards/pull/1/files#diff-d59cbb1d431e89f757bc07a9240ebca976e9accfb770eb4a871d4bc39b7b2c9cR30-R37) which you can use as an [...]

Thanks, this approach matches mine also.

closed 2024-02-02 by haizzus