OE4T/meta-tegra #1468: wip-l4t-r36.2.0-dp: tegra-container-passthrough fails to fetch

quaresmajose, 2024-02-07:
**Describe the bug**
A clear and concise description of what the bug is.

Bitbake fails to fetch the `tegra-container-passthrough`
```
ERROR: tegra-container-passthrough-36.2.0-20231218214829-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/srv/oe/build/tmp-lmp/sysroots-uninative/x86_64-linux/usr/bin:/srv/oe/layers/openembedded-core/scripts:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot-native/usr/bin/aarch64-lmp-linux:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot/usr/bin/crossscripts:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot-native/usr/sbin:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot-native/usr/bin:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot-native/sbin:/srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/recipe-sysroot-native/bin:/srv/oe/bitbake/bin:/srv/oe/build/tmp-lmp/hosttools"; export HOME="/home/builder"; /usr/bin/env wget -t 2 -T 30 --passive-ftp -O /srv/oe/downloads/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb.tmp -P /srv/oe/downloads 'https://repo.download.nvidia.com/jetson/common/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb' --progress=dot -v failed with exit code 8, no output
ERROR: tegra-container-passthrough-36.2.0-20231218214829-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://repo.download.nvidia.com/jetson/common/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb;subdir=tegra-container-passthrough-36.2.0-20231218214829/full;name=gstreamer')
ERROR: Logfile of failure stored in: /srv/oe/build/tmp-lmp/work/armv8a_tegra-lmp-linux/tegra-container-passthrough/36.2.0-20231218214829/temp/log.do_fetch.131962
```
**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'wip-l4t-r36.2.0-dp' with `MACHINE` based on 'agx-orin-devkit'
2. Build with bitbake argument '...'
3. Deploy to hardware with method '...'
4. See error '...' with link to any relevant logs at '...'

It also fail outside of bitbake
```
$ wget https://repo.download.nvidia.com/jetson/common/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb
--2024-02-07 15:39:55--  https://repo.download.nvidia.com/jetson/common/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving repo.download.nvidia.com (repo.download.nvidia.com)... 2.17.211.11, 2.17.211.144
Connecting to repo.download.nvidia.com (repo.download.nvidia.com)|2.17.211.11|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-07 15:39:55 ERROR 404: Not Found.
```

madisongh, 2024-02-07:
It's working for me:
```
$ wget https://repo.download.nvidia.com/jetson/t234/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb
--2024-02-07 07:54:25--  https://repo.download.nvidia.com/jetson/t234/pool/main/n/nvidia-l4t-gstreamer/nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb
Resolving repo.download.nvidia.com (repo.download.nvidia.com)... 23.203.166.13, 23.203.166.41
Connecting to repo.download.nvidia.com (repo.download.nvidia.com)|23.203.166.13|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1205790 (1.1M) [application/x-debian-package]
Saving to: ‘nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb’

nvidia-l4t-gstreamer_36.2.0-2023 100%[==========================================================>]   1.15M  --.-KB/s    in 0.08s   

2024-02-07 07:54:25 (13.7 MB/s) - ‘nvidia-l4t-gstreamer_36.2.0-20231218214829_arm64.deb’ saved [1205790/1205790]
```
Not much we can do about failures in NVIDIA's CDN.

madisongh, 2024-02-07:
Oh wait, yours used the `common` pool, whereas I used the `t234` pool.  Yep, that would be a bug in the recipe then.

joekale, 2024-02-07:
Right now a colleague is having a similar problem on the kirkstone-32.7.x branch for the tx2-nx. His is failing to download cudnn and tensorrt from the common pool.

madisongh, 2024-02-07:
> Right now a colleague is having a similar problem on the kirkstone-32.7.x branch for the tx2-nx. His is failing to download cudnn and tensorrt from the common pool.

That's more likely a CDN problem, since that's the right pool for those.  See [this site](https://repo.download.nvidia.com/jetson/) for the locations of the deb packages.

madisongh, 2024-02-07:
I've pushed an update to the 36.2.0 branch with a fix.  This probably didn't get caught on the autobuilder because another recipe that fetches the gstreamer deb (from the right place) had already run, so the download was already cached by the time it tried to fetch for the container-passthrough build.

quaresmajose, 2024-02-07:
Checked and the https://github.com/OE4T/meta-tegra/commit/12f097bdf516def444e31f0e89188baf3f41e821 fix the issue. Many thanks for the quick fix.

closed 2024-02-07 by quaresmajose