OE4T/meta-tegra #1485: gstnvarguscamerasrc.cpp, execute:751 No cameras available,

wico-frad, 2024-02-20:
Hi,

I am using kirkstone branch of meta-tegra to generate images for "jetson-xavier-nx-devkit-emmc" to boot from Nvme ( I'm using the initrd flashing scripts.

I'm using the MIPI-CSI2 camera from Leopard Imaging and got following error :

`$ root@jetson-xavier-nx-devkit-emmc:~# gst-launch-1.0 nvarguscamerasrc sensor-id=1 num-buffers=1 ! 'video/x-raw(memory:NVMM), width=(int)3864, height=(int)2180, format=(string)NV12, framerate=(fraction)10/1' ! nvj
pegenc ! filesink location=cam_1.jpg
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. gstnvarguscamerasrc.cpp, execute:751 No cameras available
Redistribute latency...
Got EOS from element "pipeline0".
Execution ended after 0:00:00.005776038
Setting pipeline to NULL ...
Freeing pipeline ...
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 277)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 350)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 366)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 379)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)`

Even-though the camera is available as /dev/video1.

I successfully built demo-image-full, boots from eMMC and have been able to capture the raw frames.
But right now I'm stuck when I moved from EMMC to NVMe.

Although the device boots successfully from the nvme, I'm not able to capture the raw frames:

The following error appears in dmesg after running the above command:

`[ 2257.573029] imx334 31-0037: imx334_power_on: power on`
`[2257.573480] tegra-camrtc-capture-vi tegra-capture-vi: failed to find rtcpu device node`
 `[2257.573731] video4linux video1: Error opening VI capture channel 					node /dev/capture-vi-channel0 with err: -19` 
` [ 2283.811613] imx334 30-0037: imx334_open:
 [ 2283.811903] imx334 31-0037: imx334_open:
 [ 2283.812224] imx334 32-0037: imx334_open:
 [ 2295.006094] imx334 30-0037: imx334_open:
 [ 2295.006399] imx334 31-0037: imx334_open:
 [ 2295.010095] imx334 32-0037: imx334_open:
 [ 2304.187638] imx334 30-0037: imx334_open:
 [ 2304.187957] imx334 31-0037: imx334_open:
 [ 2304.188263] imx334 32-0037: imx334_open:`
` [2375.562024] tegra-camrtc-capture-vi tegra-capture-vi: failed to find rtcpu device node
 [ 2375.562257] video4linux video0: Error opening VI capture channel `
                                                                                                       `node /dev/capture-vi-channel0 with err: -19`

Could you please help me to point out the problem please?

Thanks in advance,
M'hamed

fware, 2024-03-20:
Unfortunately, I have to same issue with my CSI2-based camera on yocto on Nano board.  Something might be wrong demo-image-full installed version of gst-launch-1.0.

madisongh, 2024-03-21:
The `-19` error code in the kernel log is ENODEV, which points to a likely kernel/device tree problem.  I don't see any imx334 support in the stock Jetson Linux kernel, so presumably you got a driver and a device tree/overlay from Leopard for this camera?

wico-frad, 2024-03-21:
Hi, 
Thank you for the helpful insight! I double-checked, adding these two line to my lcoal.
UBOOT_EXTLINUX = "1"
UBOOT_EXTLINUX_FDT = "/boot/${DTBFILE}" 

the device boots successfully from external nvme ssd and the camera works fine.

closed 2024-03-21 by madisongh