OE4T/meta-tegra #1518: Cannot access camera via gstreamer nvarguscamerasrc in docker container.

mo-pyy, 2024-04-03:
I am trying to access an imx477 camera connected via csi to a jetson xavier nx using the devkit board. I built the meta-tegra-demo on branch kirkstone using the base image and the following packages:
```
IMAGE_INSTALL:append = " \
    nvidia-docker \
    cuda-libraries \
    balena-engine \
    balena-engine-config \
    python3-jetson-stats \
    tegra-libraries-camera \
    gstreamer1.0 \
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-tegra \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    tegra-libraries-multimedia-v4l \
    tegra-container-passthrough \
    vim \
"
```
I am able to record a test video on the host system using:
```bash
gst-launch-1.0 -e nvarguscamerasrc sensor-id=0 ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=40/1" ! nvv4l2h264enc ! h264parse ! mp4mux ! filesink location=video.mp4
```
Then I launched the container using:
```bash
docker run -it --ipc=host -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE --device /dev/video0:/dev/video0 --gpus all --rm  nvcr.io/nvidia/l4t-base:35.4.1 bash
```
I tried the test video command and this simplified test:  `gst-launch-1.0 -e nvarguscamerasrc sensor-id=0 ! fakesink`.   
Both result in the following error:
```
Setting pipeline to PAUSED ...
Opening in BLOCKING MODE
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
NvMMLiteOpen : Block : BlockType = 4
===== NvVideo: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
GST_ARGUS: Creating output stream
(Argus) Error InvalidState: Failed to load EGL library (in src/eglutils/EGLUtils.cpp, function exports(), line 213)
(Argus) Error InvalidState: Failed to get EGL API access function (in src/eglutils/EGLUtils.cpp, function exports(), line 224)
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 9' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.

^Chandling interrupt.
Interrupt: Stopping pipeline ...
EOS on shutdown enabled -- Forcing EOS on the pipeline
^C
```

I looked at the nvargus-daemon logs in the host system, but could not find any errors besides those also shown when running the command in the host system successfully.

Am I missing something?

mo-pyy, 2024-04-03:
Also I am running into an error when adding the `--privileged` flag to the docker run command:
```
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: mount error: file creation failed: /var/data/balena-engine-data/overlay2/9be0cd6e640bc611cd4eaeddae49c7fd5a4ea94770a013e249421cff28b95259/merged/dev/nvgpu/igpu0/as: invalid argument: unknown.
```
This seemed similiar to [this thread](https://forums.developer.nvidia.com/t/error-running-privileged-l4t-containers-with-the-nvidia-runtime-on-jetpack-5-0-developer-preview/211108/12), but should be fixed in our version of nvidia-container-toolkit. Is it possible that this is related to `mode = "legacy"` in `/etc/nvidia-container-runtime/config.toml`?

madisongh, 2024-06-08:
Have you tried this with the latest changes on kirkstone?  There were some fixes for the container toolkit support that went in at the end of April that eliminated the need to use 'legacy' mode.

mo-pyy, 2024-09-03:
Now I tried it again using the latest commit `34bb59c85d356478e6868550822818ca268018a1` which did not work either.

juliermeSilva, 2024-11-13:
I'm having the same problem when trying to access the Raspberry PI v2.1 camera inside a docker container.
On the Jetson Nano host the camera works perfectly. But inside the container it doesn't. 
OpenCV with access to GSTREAMER successfully installed in the container. GSTREAMER dependencies successfully installed in the container.

**My container:**
docker run -it -d --gpus all --name test4 --privileged --network host --ipc=host \
  -e DISPLAY=:0.0 \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -v /dev:/dev \
  -v /home:/home \
  -v /tmp/argus_socket:/tmp/argus_socket \
  -v /usr/lib/aarch64-linux-gnu/tegra:/usr/lib/aarch64-linux-gnu/tegra \
  -v /usr/lib/aarch64-linux-gnu/tegra-egl:/usr/lib/aarch64-linux-gnu/tegra-egl \
  -p 2225:22 \
  ultralytics/ultralytics:latest-jetson-jetpack4
  

**Error seen in the terminal:**
root@jetson:/ultralytics# gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=640, height=480, framerate=30/1' ! nvvidconv ! 'video/x-raw, format=(string)BGRx' ! videoconvert ! xvimagesink
(Argus) Error FileOperationFailed: Connecting to nvargus-daemon failed: Connection refused (in src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 205)
(Argus) Error FileOperationFailed: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:735 Failed to create CameraProvider
Got EOS from element "pipeline0".
Execution ended after 0:00:00.003569630
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

dwalkes, 2024-11-13:
Do you see anything in nvargus-daemon logs?

Based on https://forums.developer.nvidia.com/t/resolving-video-capture-error-in-docker-container-after-restarting-nvargus-daemon/271196/3 you may want to try volume mounting `/tmp` instead of `/tmp/argus_socket`.

If that doesn't work I'd probably install the equivalent Jetpack image and prove docker works there with the same hardware, use this to compare with your setup.

juliermeSilva, 2024-11-13:
I solved the issue of accessing the Raspberry PI v2.1 Camera, inside a container.
The container creation was correct. I had already installed all the dependencies related to GStreamer.
The error was occurring on my system for two reasons:

- **Reason 1 -** Granting permission to the X server for connections from a specific user. I was not granting permissions. Before starting the docker container I ran the command:
`xhost +si:localuser:root`

- **Reason 2 -** As I mentioned I had already granted all access privileges when creating the container. I created the correct container but tested it incorrectly. The error visible in the previous post is consistent with an incompatible graphics sink in the GStreamer pipeline. The error occurs specifically because _xvimagesink_ is not available or functional in the container environment. Before attempting any communication with GStreamer inside the container, check if the communication plugin is present inside the container using the command:
`gst-inspect-1.0 | grep sink`
When I ran this command inside my container, two sinks appeared in the list: _appsink_ and _nveglglessink_.
I used the _nveglglessink_ sink (developed by Nvidia) to do a new test, rendering the stream directly to a screen using EGL/GLES (taking advantage of NVIDIA hardware graphics acceleration).
The new test:
```
gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=640, height=480, framerate=30/1' ! nvvidconv ! 'video/x-raw, format=(string)BGRx' ! videoconvert ! nveglglessink
```
This time a window showing the camera appeared, proving the connectivity of the container with the Raspberry PI V2.1 camera, via GStreamer.
[image]

Next, I wrote a simple Python code to test the GStreamer reading within an application. This time, the sink used was _appsink_, since it is designed to allow the GStreamer pipeline to send video or audio data directly to an application, such as a Python script, where it will be possible to process the frames, perform AI inference, save to files or perform other custom operations.
The python code:

```
import cv2

# Define the GStreamer pipeline to access the Raspberry Pi v2.1 camera
gst_pipeline = (
    "nvarguscamerasrc ! "
    "video/x-raw(memory:NVMM), width=(int)640, height=(int)480, framerate=(fraction)30/1 ! "
    "nvvidconv ! video/x-raw, format=(string)BGRx ! "
    "videoconvert ! video/x-raw, format=(string)BGR ! appsink"
)

# Initialize video capture with the GStreamer pipeline
cap = cv2.VideoCapture(gst_pipeline, cv2.CAP_GSTREAMER)

if not cap.isOpened():
    print("Error: Unable to access the camera!")
    exit()

print("Camera accessed successfully!!")

while True:
    ret, frame = cap.read()
    if not ret:
        print("Error: Unable to capture the frame!")
        break

    # Display the captured frame
    cv2.imshow("Raspberry Pi cam", frame)

    # Press 'q' to exit (quit)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

# Release the camera and close the windows
cap.release()
cv2.destroyAllWindows()

```

I hope I have contributed something.

paroque28, 2025-01-16:
@mo-krauti 
I have the same issue:
```
(Argus) Error InvalidState: Failed to load EGL library (in src/eglutils/EGLUtils.cpp, function exports(), line 213)
(Argus) Error InvalidState: Failed to get EGL API access function (in src/eglutils/EGLUtils.cpp, function exports(), line 224)
```

Were you able to solve it?

paroque28, 2025-01-16:
Fixed using notes from here:
https://forums.balena.io/t/jetson-support-nvidia-docker-images/340483/44?u=cyanpablo

PR: https://github.com/OE4T/meta-tegra/pull/1811

limeng-linux, 2025-04-16:
@juliermeSilva 
Hi juliermeSilva 

I also encounter a camera issue,  I built the meta-tegra-demo on branch master for Orin nx board, after board boots up, I found out there is no /dev/video0 node.
Did your board have /dev/video0 node after booting up?
Or can you recall some configurations needing to be added so that /dev/video0 is created and camera can works fine.

thanks,
Limeng