OE4T/meta-tegra #1522: Orin TX fails compilation

roby2014, 2024-04-04:
Trying to build `p3768-0000-p3767-0001`, branch `kirkstone`. I get the following:  

```ERROR: linux-tegra-5.10.192+gitAUTOINC+db857ecdbe-r0 do_patch: Could not apply patches for p3768-0000-p3767-0001.
ERROR: linux-tegra-5.10.192+gitAUTOINC+db857ecdbe-r0 do_patch: Patch failures can be resolved in the linux source directory /home/ubuntu/roby/builder/build/tmp/work-shared/p3768-0000-p3767-0001/kernel-source)
ERROR: linux-tegra-5.10.192+gitAUTOINC+db857ecdbe-r0 do_patch: ExecutionError('/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/temp/run.do_patch.1266886', 1, None, None)
ERROR: Logfile of failure stored in: /home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/temp/log.do_patch.1266886
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/ubuntu/roby/builder/build/../layers/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/ubuntu/roby/builder/build/../layers/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/roby/builder/build/../layers/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: ['quilt-native', 'patch-native', 'attr-native']
| NOTE: Skipping as already exists in sysroot: ['kern-tools-native', 'libtool-native', 'gettext-minimal-native', 'texinfo-dummy-native']
| DEBUG: sed -e 's:^[^/]*/:/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/recipe-sysroot-native/:g' /home/ubuntu/roby/builder/build/tmp/sysroots-components/x86_64/quilt-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/home/ubuntu/roby/builder/build/tmp/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/ubuntu/roby/builder/build/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/ubuntu/roby/builder/build/tmp/pkgdata/p3768-0000-p3767-0001:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/ubuntu/roby/builder/build/tmp/work/p3768_0000_p3767_0001-poky-linux/linux-tegra/5.10.192+gitAUTOINC+db857ecdbe-r0/temp/fifo.1266886:g'
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_patch
| (1/18) 0001-dt-add-tegra186-tek-base.patch
| (2/18) 0002-dt-add-tegra186-tek.patch
| (3/18) 0003-dt-build-tegra186-tek.patch
| [INFO]: check of .kernel-meta//patches//./0003-dt-build-tegra186-tek.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0003-dt-build-tegra186-tek.patch with "git am" did not work, trying "apply".
| error: nvidia/platform/t18x/quill/kernel-dts/Makefile: does not exist in index
| [ERROR]: Application of .kernel-meta//patches//./0003-dt-build-tegra186-tek.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR: Could not apply patches for p3768-0000-p3767-0001.
| ERROR: Patch failures can be resolved in the linux source directory /home/ubuntu/roby/builder/build/tmp/work-shared/p3768-0000-p3767-0001/kernel-source)
| WARNING: exit code 1 from a shell command.
NOTE: recipe linux-tegra-5.10.192+gitAUTOINC+db857ecdbe-r0: task do_patch: Failed
ERROR: Task (/home/ubuntu/roby/builder/build/../layers/meta-tegra-orin/recipes-kernel/linux/linux-tegra_5.10.bb:do_patch) failed with exit code '1'
```

madisongh, 2024-04-04:
It looks like you've added patches to the kernel recipe for your TX2-based builds.  Those patches won't apply to the 5.10 kernel in L4T R35 and later.  You really can't mix TX2 and Orin in the same builds - the NVIDIA BSPs are completely different.  That's why we have [separate kirkstone branches](https://github.com/OE4T/meta-tegra/wiki/Which-branch-should-I-use%3F) for the two.

roby2014, 2024-04-04:
Thanks for the information. Whats the proper solution for this? How can I remove/add that package for my specific board?
I build with `kas` so I could add something like `IMAGE_INSTALL:append = "gstreamer1.0-omx-tegra"` to `tx2.yml` and `IMAGE_INSTALL:remove += "gstreamer1.0-omx-tegra"?

madisongh, 2024-04-04:
That sounds like you're asking about #1521 that you raised... I don't really know much about kas, but you need to track down where that package is getting pulled in.  If you've added it directly in your image recipe, then what you propose might work.  It's also possible, though, that it's getting pulled in as a runtime dependency for some other package you're building, in which case you need to fix that package so it no longer has that dependency.  That could be more than just a recipe modification - code may have to be changed.

Similarly for the patch problem you mention here - track down where those patches are being added, and fix your recipes/bbappends/kas settings/etc. so that they apply only to the builds for your TX2-based machine.  The specifics of how to do that really depend on how you've got your build environment set up.

roby2014, 2024-04-04:
I putted `COMPATIBLE_MACHINE="jetson-tx2-devkit"` on `linux-tegra_%.bbappend` which does kernel mods for tx2. However if I compile for orin, I get:
```
ERROR: Nothing PROVIDES 'virtual/kernel'
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-dev PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-dummy PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-tegra, not linux-dummy
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-custom PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto-upstream PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
linux-tegra PROVIDES virtual/kernel but was skipped: incompatible with machine p3768-0000-p3767-0001 (not in COMPATIBLE_MACHINE)
ERROR: Required build target 'myos-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['my-os', 'virtual/kernel']
``` any idea?

madisongh, 2024-04-04:
Try renaming the bbappend to include the version number, e.g. `linux-tegra_4.9%.bbappend`

roby2014, 2024-04-05:
Thank you, fixed by adding:
```
    BBMASK += "/meta-tegra-tek/recipes-kernel/linux/linux-tegra_4.9.bbappend"
    PREFERRED_PROVIDER_virtual/kernel = "linux-tegra"
    PREFERRED_VERSION_virtual/kernel = "5.10"
```

closed 2024-04-05 by roby2014