OE4T/meta-tegra #1530: ./initrd-flash: line 292: flash.idx: No such file or directory

rcptk, 2024-04-15:
Hi. I was trying to flash my ORIN NX 16 GB + custom carrier but now I get these errors:
```
1 > sudo ./initrd-flash 
Starting at 2024-04-15T12:35:37
Machine:      custom-carrier
Rootfs device: nvme0n1p1
Found Jetson device in recovery mode at USB 1-6
== Step 1: Signing binaries at 2024-04-15T12:35:37+01:00 ==
cp: cannot stat 'signed/*': No such file or directory
cp: cannot stat 'signed/flash.xml.tmp': No such file or directory
sed: can't read flash.xml.tmp: No such file or directory
cp: cannot stat 'signed/*': No such file or directory
cp: cannot stat 'signed/flash.xml.tmp': No such file or directory
sed: can't read flash.xml.tmp: No such file or directory
./initrd-flash: line 292: flash.idx: No such file or directory
ERR: no SPI or eMMC boot partition entries found
ERR: signing failed at 2024-04-15T12:35:42+01:00
```

This started happening randomly, I could flash this same image yesterday with no problems.
Any idea or how I can debug this?

madisongh, 2024-04-15:
Have you looked in the log file to see where the signing failed?

And have you followed the guidelines [on the wiki](https://github.com/OE4T/meta-tegra/wiki/Creating-a-custom-MACHINE#jetson-orin) for creating a custom MACHINE definition?

rcptk, 2024-04-15:
1.
```
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0270 ] tegrasign_v3.py --key None --getmode mode.txt
[   0.0272 ] Assuming zero filled SBK key
[   0.0245 ] tegra234-p3767-0000-p3509-a02.dtb has no overlay_dtb files.
[   0.0245 ] Concatenating dtbo files to tegra234-p3767-0000-p3509-a02.dtb
[   0.0249 ] Concatenating L4TConfiguration-rcmboot.dtbo,tegra234-p3767-overlay.dtbo,tegra234-p3767-0000-p3509-a02-hdr40.dtbo to tegra234-p3767-0000-p3509-a02_overlay.dtb
[   0.0251 ] Concatenating bl dtb:(tegra234-p3767-0000-p3509-a02_overlay.dtb), to cpubl binary: rcmboot_uefi_jetson.bin
[   0.0277 ] Get magic id
[   0.0281 ] tegraparser_v2 --get_magic bootloader_stage2
[   0.0285 ] CPBL
[   0.0287 ] partition type bootloader_stage2, magic id = CPBL
[   0.0313 ] tegrahost_v2 --chip 0x23 0 --align rcmboot_uefi_jetson_with_dtb_aligned.bin
[   0.0320 ] tegrahost_v2 --chip 0x23 0 --magicid CPBL --appendsigheader rcmboot_uefi_jetson_with_dtb_aligned.bin zerosbk
[   0.0325 ] adding BCH for rcmboot_uefi_jetson_with_dtb_aligned.bin
[   0.1251 ] tegrasign_v3.py --key None --list rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin_list.xml --pubkeyhash pub_key.key --sha sha512
[   0.1252 ] Assuming zero filled SBK key
[   0.1334 ] Warning: pub_key.key is not found
[   0.1311 ] tegrahost_v2 --chip 0x23 0 --updatesigheader rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.encrypt rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.hash zerosbk
[   0.1391 ] Copying rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt to /media/user/builds/myos.tegraflash
[   0.1391 ] Signed file: /media/user/builds/myos.tegraflash/rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt
ERR: signing failed at 2024-04-15T11:58:59+01:00
```

2. Yes. I was able to flash the same image days ago (no modifications).

rcptk, 2024-04-15:
I think this issue started after I started messing around with `i2cset`

madisongh, 2024-04-15:
That log extract doesn't look complete - there's likely an error being reported by the NVIDIA tools somewhere earlier in the log.

Adding some tools like i2c-utils to your build is unlikely to cause a flashing failure.  More likely it's something to do with your storage layout file.

redeexpressos, 2024-07-18:
Hi. Sorry for delaying, the workaround and problem are described here: https://forums.developer.nvidia.com/t/re-flashing-orin-fails-after-using-i2cset/289658

ichergui, 2024-07-18:
I'm gonna close this issue then. Any objection or concern ?

closed 2024-07-18 by redeexpressos

Islam-Hussein-11, 2024-08-28:
I'm sorry to open that again. I'm facing the exact same problem when I'm trying to make a demo for disk encryption. So after making a small data-partition.img and encrypting it with luks passphrase. I've modified also the sizes in external-flash.xml.in. But at the end I get that response. If you know what is root cause of such problem would be helpful. 
`Starting at 2024-08-28T05:26:43+02:00
Machine:       p3509-a02-p3767-0001
Rootfs device: nvme0n1p1
Waiting for Jetson to appear on USB.........[found: 3-2.4]
== Step 1: Signing binaries at 2024-08-28T05:26:51+02:00 ==
cp: cannot stat 'signed/*': No such file or directory
cp: cannot stat 'signed/flash.xml.tmp': No such file or directory
sed: can't read flash.xml.tmp: No such file or directory
== Step 2: Boot Jetson via RCM at 2024-08-28T05:27:18+02:00 ==
Found Jetson device in recovery mode at USB 3-2.4
== Step 3: Sending flash sequence commands at 2024-08-28T05:27:23+02:00 ==
`