OE4T/meta-tegra #1549: Writing partitions takes a lot of time

redeexpressos, 2024-05-08:
Hello, I'm trying to build a custom image for `p3768-0000-p3767-0001` machine. Everything goes OK, until "Writing partitions" stage:

```
 ✘ sudo ./initrd-flash
Starting at 2024-05-08T10:13:01+01:00
Machine:       p3768-0000-p3767-0001
Rootfs device: nvme0n1p1
Found Jetson device in recovery mode at USB 1-5
== Step 1: Signing binaries at 2024-05-08T10:13:02+01:00 ==
== Step 2: Boot Jetson via RCM at 2024-05-08T10:13:21+01:00 ==
Found Jetson device in recovery mode at USB 1-5
== Step 3: Sending flash sequence commands at 2024-05-08T10:13:24+01:00 ==
Waiting for USB storage device flashpkg from 613c0807.......[/dev/sde]
Device size in blocks: 262144
Unmounted /dev/sde.
== Step 4: Writing partitions on external storage device at 2024-05-08T10:13:48+01:00 ==
Waiting for USB storage device nvme0n1 from 613c0807...[/dev/sde]
Creating partitions
  [02] name=A_kernel start=0 size=262144 sectors
  [03] name=A_kernel-dtb start=0 size=1536 sectors
  [04] name=A_reserved_on_user start=0 size=64768 sectors
  [05] name=B_kernel start=0 size=262144 sectors
  [06] name=B_kernel-dtb start=0 size=1536 sectors
  [07] name=B_reserved_on_user start=0 size=64768 sectors
  [08] name=recovery start=0 size=163840 sectors
  [09] name=recovery-dtb start=0 size=1024 sectors
  [10] name=esp start=0 size=131072 sectors
  [11] name=recovery_alt start=0 size=163840 sectors
  [12] name=recovery-dtb_alt start=0 size=1024 sectors
  [13] name=esp_alt start=0 size=131072 sectors
  [14] name=UDA start=0 size=819200 sectors
  [15] name=reserved start=0 size=982016 sectors
  [01] name=APP (fills to end)
Writing partitions
  Writing boot.img (size=35817472) to /dev/sde2 (size=134217728)...
  Writing kernel_tegra234-p3767-0001-p3768-0000-a0.dtb (size=347347) to /dev/sde3 (size=786432)...
  Writing boot.img (size=35817472) to /dev/sde5 (size=134217728)...
  Writing kernel_tegra234-p3767-0001-p3768-0000-a0.dtb (size=347347) to /dev/sde6 (size=786432)...
  Writing esp.img (size=67108864) to /dev/sde10 (size=67108864)...
  Writing myos-image.ext4 (size=59055800320) to /dev/sde1 (size=248493809152)...
```

Serial logs (from ORIN):
```
Jetson UEFI firmware (version v35.5.0 built on 2024-02-26T13:44:31+00:00)
ESC   to enter Setup.
F11   to enter Boot Manager Menu.
Enter to continue boot.
......EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services and installing virtual address map...
��I/TC: Secondary CPU 1 initializing
I/TC: Secondary CPU 1 switching to normal world boot
I/TC: Secondary CPU 2 initializing
I/TC: Secondary CPU 2 switching to normal world boot
I/TC: Secondary CPU 3 initializing
I/TC: Secondary CPU 3 switching to normal world boot
I/TC: Secondary CPU 6 initializing
I/TC: Secondary CPU 6 switching to normal world boot
I/TC: Secondary CPU 7 initializing
I/TC: Secondary CPU 7 switching to normal world boot
��[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd421]
[    0.000000] Linux version 5.10.192-l4t-35.5.0+gdb857ecdbe10 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.4.0, GNU ld (GNU Binutils) 2.38.20220708) #1 SMP PREEMPT Wed Feb 28 11:28:56 UTC 2024
[    0.000000] OF: fdt: memory scan node memory@80000000, reg size 16,
[    0.000000] OF: fdt:  - 80000000 ,  c0000000
[    0.000000] Machine model: NVIDIA Orin NX Developer Kit
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0xffff0000 SMBIOS 3.0=0x23ba70000 MEMATTR=0x23635e018 RNG=0x22a480018 MEMRESERVE=0x22a8e7f18 
[    0.000000] random: crng init done
[... 719 lines cut ...]
mke2fs 1.46.5 (30-Dec-2021)
Found /tmp/flashpkg.ext4
[    7.154429] Mass Storage Function, version: 2009/09/11
[    7.154583] LUN: removable file: (no medium)
[    7.156058] tegra-xudc 3550000.xudc: EP 0 (type: ctrl, dir: out) enabled
Exported /tmp/flashpkg.ext4 as flashpkg
Waiting for host to connect...[connected]
Waiting for host to disconnect...[    7.676488] tegra-xudc 3550000.xudc: EP 3 (type: bulk, dir: in) enabled
[    7.676712] tegra-xudc 3550000.xudc: EP 2 (type: bulk, dir: out) enabled
[   10.410064] tegra-xudc 3550000.xudc: ep 3 disabled
[   10.410230] tegra-xudc 3550000.xudc: ep 2 disabled
[disconnected]
[   11.173699] tegra-xudc 3550000.xudc: ep 0 disabled
[   11.179897] loop: module loaded
[   11.207688] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
Processing: bootloader 
Processing: export-devices nvme0n1
[   11.213206] Mass Storage Function, version: 2009/09/11
[   11.213355] LUN: removable file: (no medium)
[   11.213894] tegra-xudc 3550000.xudc: EP 0 (type: ctrl, dir: out) enabled
[   11.668444] tegra-xudc 3550000.xudc: EP 3 (type: bulk, dir: in) enabled
[   11.668661] tegra-xudc 3550000.xudc: EP 2 (type: bulk, dir: out) enabled
```

The `Writing myos-image.ext4` stage is taking a lot of time (I've been waiting for 20 minutes). Is this normal? I remember flashing Orin's in the past didn't take this much time..

madisongh, 2024-05-08:
Sounds like you haven't installed `bmaptool` on your development host. See #1499, #1524.

redeexpressos, 2024-05-08:
That solved the issue, thank you.

closed 2024-05-08 by redeexpressos