OE4T/meta-tegra #1555: `p3509-a02-p3767-0000` fails compiling due to tegra-minimal-initramfs' do_image_cpio (kirkstone)

redeexpressos, 2024-05-13:
Building with machine set to `p3509-a02-p3767-0000` leads to the following errors:

```
ERROR: tegra-minimal-initramfs-1.0-r0 do_image_cpio: ExecutionError('/home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/run.do_image_cpio.3605935', 1, None, None)
ERROR: Logfile of failure stored in: /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_image_cpio.3605935
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/ubuntu/z/builder/build/../layers/meta-tegra-orin/recipes-bsp/tegra-binaries/tegra-bootfiles_35.5.0.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/meta-tegra-orin/recipes-bsp/tegra-binaries/tegra-flashtools-native_35.5.0.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/meta-tegra-orin/recipes-bsp/tegra-binaries/tegra-redundant-boot-rollback_35.5.0.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-core/glibc/cross-localedef-native_2.35.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', '/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-core/coreutils/coreutils_9.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/roby/builder/build/../layers/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.5.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/opkg/opkg_0.5.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/perl/perl_5.34.3.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/python/python3-pyyaml_6.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-devtools/python/python3_3.10.13.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-extended/cpio/cpio_2.14.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-extended/pigz/pigz_2.7.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/z/builder/build/../layers/poky/meta/recipes-kernel/dtc/dtc_1.6.1.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['tegra-bootfiles', 'tegra-flashtools-native', 'tegra-redundant-boot-rollback', 'cross-localedef-native', 'ldconfig-native', 'qemuwrapper-cross', 'depmodwrapper-cross', 'coreutils-native', 'update-rc.d-native', 'makedevs-native', 'opkg-utils-native', 'opkg-native', 'perl-native', 'pseudo-native', 'python3-pyyaml-native', 'python3-native', 'cpio-native', 'pigz-native', 'dtc-native', 'zlib-native', 'xz-native', 'gettext-minimal-native', 'texinfo-dummy-native', 'libtool-native', 'flex-native', 'gcc-runtime', 'glibc', 'openssl-native', 'debianutils-native', 'kmod-native', 'systemd-systemctl-native', 'tegra-helper-scripts-native', 'dos2unix-native', 'shared-mime-info-native', 'util-linux-libuuid-native', 'ncurses-native', 'libffi-native', 'bzip2-native', 'libnsl2-native', 'gdbm-native', 'readline-native', 'libtirpc-native', 'sqlite3-native', 'mtools-native', 'dosfstools-native', 'python3-cython-native', 'python3-installer-native', 'libyaml-native', 'python3-wheel-native', 'python3-setuptools-native', 'shadow-native', 'qemu-native', 'attr-native', 'tegra-flashvars', 'make-native', 'libsolv-native', 'libarchive-native', 'm4-native', 'libgcc', 'linux-libc-headers', 'itstool-native', 'libxml2-native', 'glib-2.0-native', 'python3-flit-core-native', 'unzip-native', 'expat-native', 'cmake-native', 'lzo-native', 'zstd-native', 'gnu-config-native', 'libpcre-native', 'gettext-native', 'util-linux-native', 'curl-native', 'libcap-ng-native', 'libpcre2-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function set_image_size
| DEBUG: 7170.800000 = 5516 * 1.300000
| DEBUG: 32768.000000 = max(7170.800000, 32768)[32768.000000] + 0
| DEBUG: 32768.000000 = int(32768.000000)
| DEBUG: 32768 = aligned(32768)
| DEBUG: returning 32768
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_cpio
| 9223 blocks
| NOTE: Copying KERNEL_DEVICETREE entry tegra234-p3767-0000-p3509-a02.dtb to /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload
| NOTE: Copying overlay L4TConfiguration.dtbo to /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload
| NOTE: Copying overlay tegra234-p3767-overlay.dtbo to /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload
| NOTE: Copying overlay tegra234-p3767-0000-p3509-a02-hdr40.dtbo to /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload
| NOTE: Copying overlay L4TConfiguration-rcmboot.dtbo to /home/ubuntu/z/builder/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload
| Board ID(3767) version(000) sku(0000) revision()
| Signing with zerosbk ...
| rollback info is parsed and saved in rollback.bin
| WARNING: failed to find rollback config for " t186ref "
| pad ZEROs to the end of rollback.bin
| 
| Generating signed binaries for 3767-000-0000--1-1-p3509-a02-p3767-0000- ...
| python3 tegraflash.py --chip 0x23 --bl uefi_jetson_with_dtb.bin --sdram_config tegra234-p3767-0000-sdram-l4t.dts --odmdata gbe-uphy-config-8,hsstp-lane-map-3,hsio-uphy-config-0 --applet mb1_t234_prod.bin --cmd "sign" --skipuid --cfg flash.xml --bct_backup --uphy tegra234-mb1-bct-uphylane-si.dtsi --minratchet_config tegra234-mb1-bct-ratchet-p3767-0000.dts --device_config tegra234-mb1-bct-device-p3767-0000.dts --misc_config tegra234-mb1-bct-misc-p3767-0000.dts --scr_config tegra234-mb2-bct-scr-p3767-0000.dts --pinmux_config tegra234-mb1-bct-pinmux-p3767-hdmi-a03.dtsi --gpioint_config tegra234-mb1-bct-gpioint-p3767-0000.dts --pmic_config tegra234-mb1-bct-pmic-p3767-0000-a02.dts --pmc_config tegra234-mb1-bct-padvoltage-p3767-hdmi-a03.dtsi --prod_config tegra234-mb1-bct-prod-p3767-0000.dts --br_cmd_config tegra234-mb1-bct-reset-p3767-0000.dts --dev_params tegra234-br-bct-p3767-0000-l4t.dts,tegra234-br-bct_b-p3767-0000-l4t.dts --deviceprod_config tegra234-mb1-bct-cprod-p3767-0000.dts --wb0sdram_config tegra234-p3767-0000-wb0sdram-l4t.dts --mb2bct_cfg tegra234-mb2-bct-misc-p3767-0000.dts --bldtb tegra234-p3767-0000-p3509-a02.dtb --concat_cpubl_bldtb --cpubl uefi_jetson.bin --overlay_dtb L4TConfiguration.dtbo,tegra234-p3767-overlay.dtbo,tegra234-p3767-0000-p3509-a02-hdr40.dtbo --bins "psc_fw pscfw_t234_prod.bin; mts_mce mce_flash_o10_cr_prod.bin; mb2_applet applet_t234.bin; mb2_bootloader mb2_t234.bin; xusb_fw xusb_t234_prod.bin; dce_fw display-t234-dce.bin; nvdec nvdec_t234_prod.fw; bpmp_fw bpmp_t234-TE980M-A1_prod.bin; bpmp_fw_dtb tegra234-bpmp-3767-0000-a02-3509-a02.dtb; sce_fw camera-rtcpu-sce.img; rce_fw camera-rtcpu-t234-rce.img; ape_fw adsp-fw.bin; spe_fw spe_t234.bin; tos tos-optee_t234.img; eks eks.img"
| Welcome to Tegra Flash
| version 1.0.0
| Type ? or help for help and q or quit to exit
| Use ! to execute system commands
| 
| [   0.5910 ] tegrasign_v3.py --key None --getmode mode.txt
| [   0.5925 ] Assuming zero filled SBK key
| [   0.5689 ] Parsing partition layout
| [   0.5735 ] tegraparser_v2 --pt flash.xml.tmp
| [   0.5779 ] Invalid partition type backup_secondary_gpt
| Error: Return value 4
| Command tegraparser_v2 --pt flash.xml.tmp
| Board ID(3767) version(000) sku(0000) revision()
| Signing with zerosbk ...
| rollback info is parsed and saved in rollback.bin
[... cut ...]
| [   0.5349 ] tegrasign_v3.py --key None --getmode mode.txt
| [   0.5364 ] Assuming zero filled SBK key
| [   0.5285 ] Parsing partition layout
| [   0.5323 ] tegraparser_v2 --pt flash.xml.tmp
| [   0.5359 ] Invalid partition type backup_secondary_gpt
| Error: Return value 4
| Command tegraparser_v2 --pt flash.xml.tmp
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/ubuntu/z/builder/build/../layers/meta-tegra-orin/recipes-core/images/tegra-minimal-initramfs.bb:do_image_cpio) failed with exit code '1'
```

madisongh, 2024-05-13:
It built OK for me, using kirkstone latest in our demo distro.  As you can see from the log, the complaint is from the `tegraparser_v2` tool, trying to parse the storage layout XML file.  I'm not sure how that would happen, since the tool and the XML file both come from the L4T BSP, and as long as they're both from the same version (R35.5.0 in this case), they should be compatible.

You might try cleaning your build workspace and rebuilding, particularly if you've been switching around branches or revisions between builds.

redeexpressos, 2024-05-13:
Thank you, clean building (removing `build` and `layers`) fixed it.

closed 2024-05-13 by redeexpressos