OE4T/meta-tegra #1571: Orin NX won't boot scarthgap build

antoinefaure, 2024-06-01:
Hi all,

I'm starting a project on an Orin NX 16GB ([reComputer j4012](https://www.seeedstudio.com/reComputer-J4012-p-5586.html)).
I have a basic image built with the nanbield branches of meta-tegra, meta-openembedded and poky, and this works fine. But when I update all the metas to scarthgap, the target won't boot at all.
When I connect to a monitor I will see the NVIDIA logo and UEFI menu, and then it will reboot and fails to go further than that.
I have a very basic distro based on poky, with just a few distro features enabled. Image is very basic as well.

I am using the tegraflash image:
```
IMAGE_FSTYPES = "tegraflash"
IMAGES_CLASSES += "image_types_tegra"
```

In both cases I am using the `p3768-0000-p3767-0000` machine that I flash with the `./doflash.sh` script.

You will find uart logs attached.

Any help would be greatly appreciated.
Thanks

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` scarthgap '...' with `MACHINE` based on 'p3768-0000-p3767-0000'
2. Build with bitbake argument 'image-name'
3. Deploy to hardware with method './doflash.sh'

[logs.txt](https://github.com/user-attachments/files/15521191/jetson.txt)

dwalkes, 2024-06-01:
> I have a very basic distro based on poky, with just a few distro features enabled. Image is very basic as well.

If you are just attempting to see a working implementation distribution as opposed to setting up your own you may want to look at https://github.com/OE4T/tegra-demo-distro.  Or if you reproduce a problem there it will be easier for others to reproduce as well.

antoinefaure, 2024-06-01:
Duh, very sorry about that. I have 2 different jetsons and mixed up the install scripts.
Thanks for pointing out my mistake!

closed 2024-06-01 by antoinefaure