OE4T/meta-tegra #1580: Cannot compile kernel on kirkstone

intera91, 2024-06-12:
bitbake virtual/kernel results in 1977 errors, mostly 
pinctrl-tegra194.c with the structure tegra_pingroup no member named <...>

a little extract of log.do_compile:

```
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:430:18: error: 'const struct tegra_pingroup' has no member named 'parked_bit'; did you mean 'pad_bit'?
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:439:18: error: 'const struct tegra_pingroup' has no member named 'e_io_hv_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:441:18: error: 'const struct tegra_pingroup' has no member named 'gpio_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:446:18: error: 'const struct tegra_pingroup' has no member named 'pbias_buf_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:447:18: error: 'const struct tegra_pingroup' has no member named 'preemp_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:448:18: error: 'const struct tegra_pingroup' has no member named 'rfu_in_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:453:18: error: 'const struct tegra_pingroup' has no member named 'pwr_domain'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:916:8: error: initialization of 'signed char:6' from 'char *' makes integer from pointer without a cast [-Werror=int-conversion]
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:916:8: error: initializer element is not computable at load time
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:430:18: error: 'const struct tegra_pingroup' has no member named 'parked_bit'; did you mean 'pad_bit'?
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:439:18: error: 'const struct tegra_pingroup' has no member named 'e_io_hv_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:441:18: error: 'const struct tegra_pingroup' has no member named 'gpio_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:446:18: error: 'const struct tegra_pingroup' has no member named 'pbias_buf_bit'
/data/workspace/kirkstone/build/tmp/work-shared/guidance-jetson-xavier-nx-specs/kernel-source/nvidia/drivers/pinctrl/pinctrl-tegra194.c:447:18: error: 'const struct tegra_pingroup' has no member named 'preemp_bit'
```

madisongh, 2024-06-12:
Really hard to say based on the little data provided.  The machine name isn't one of ours, so I suspect you have other layers in your build that could be breaking things.

dwalkes, 2024-06-12:
@intera91 a suggestion is to attempt to reproduce at https://github.com/OE4T/tegra-demo-distro and if you can't use this to bisect differences in your custom setup.

acostach, 2024-06-14:
Slighthly off-topic but @madisongh , I see the kirkstone branch has 35.5.0 support and is being maintained. Any plans to have 36.3 on kirkstone too? Just the kernel, oot drivers and the uefi firmware would be fantastic to have

Thank you

madisongh, 2024-06-14:
@acostach No plans for an R36-based kirkstone branch. We do have an R35-based scarthgap branch we'll be maintaining to help stage migration between LTS branches and between R35 and R36 BSP releases.

intera91, 2024-06-14:
Solved in kernel 5.10.192 (JP 5.1.3)

closed 2024-06-14 by intera91