OE4T/meta-tegra #1590: Xavier AGX fails to switch partitions after swupdate

KelAmo, 2024-06-18:
I'm setting up swupdate with A/B partitions on a Xavier AGX with a custom carrier board and a recent Kirkstone build. I am able to generate a .swu package and perform a swupdate (swupdate -i *.swu). The update completes successfully, however the xavier fails to switch partitions after a reboot. After the first reboot, I can use nvbootctrl (nvbootctrl set-active-boot-slot [slot-#]), successfully switch partitions and confirm that the swupdate was successful. The behavior is the same no matter which direction I update (A->B, B->A).

Here is serial output after completing a swupdate successfully on partition B (APP_b, slot 1) and power cycling the xavier:
[power-cycle-after-swupdate.txt](https://github.com/user-attachments/files/15879773/power-cycle-after-swupdate.txt)

MB1 correctly reports "Boot chain mechanism: A/B", but all references to slot after that point are to slot 1 (B). I'm not sure why it's failing to switch to slot 0 (A).

joekale-pp, 2024-06-18:
Historically this seems to be related to the TNSPEC value that gets put into the 
UEFI capsule. UEFI rejects the capsule update for a mismatch. The best way to determine why UEFI is rejecting the capsule is to enable the debug logging of UEFI by adding `EDK2_BUILD_MODE:pn-edk2-firmware-tegra = "DEBUG"` to your local.conf. If that doesn't immediately show the issue then you may need to enable more debug logging by using devtool to manually edit the constant linked in [this issue comment](https://github.com/OE4T/meta-tegra/pull/1285#issuecomment-1659362121).

KelAmo, 2024-06-20:
Yeah, it was the TNSPEC. I thought I had ruled that out because I had verified that the custom machine name was getting set correctly, but I didn't catch that the FAB and the BOARDSKU were both incorrect. I had to set them in the local.conf. The debug information helped me catch it, thank you.

closed 2024-06-20 by KelAmo