OE4T/meta-tegra #1597: tegra-uefi-capsules-36.3.0/bup-payload/signed/flash.idx is not found when building SWUpdate image

Bmkool, 2024-06-26:
Hello,

I am attempting to build the scarthgap branch for L4T 36.3.0, and followed the [README to enable SWUpdate](https://github.com/OE4T/tegra-demo-distro/blob/master/layers/meta-tegrademo/dynamic-layers/meta-swupdate/README.md) on an AGX Orin devkit. 

I have run into this error when building `swupdate-image-tegra` for the tegra-demo-distro:
```
The index file /yocto/tegra-demo-distro/build/tmp/work/jetson_agx_orin_devkit-oe4t-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload/signed/flash.idx is not found.
WARNING: /yocto/tegra-demo-distro/build/tmp/work/jetson_agx_orin_devkit-oe4t-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.1332239:255 exit 1 from './doflash.sh'
WARNING: Backtrace (BB generated script): 
	OE4T/tegra-demo-distro#1: tegraflash_custom_sign_bup, /yocto/tegra-demo-distro/build/tmp/work/jetson_agx_orin_devkit-oe4t-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.1332239, line 255
	OE4T/tegra-demo-distro#2: do_compile, /yocto/tegra-demo-distro/build/tmp/work/jetson_agx_orin_devkit-oe4t-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.1332239, line 161
	OE4T/tegra-demo-distro#3: main, /yocto/tegra-demo-distro/build/tmp/work/jetson_agx_orin_devkit-oe4t-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.1332239, line 427
```
Full error log:
[tegra-uefi-capsules.log.do_compile.txt](https://github.com/user-attachments/files/15975599/tegra-uefi-capsules.log.do_compile.txt)

Any recommendations?

Thanks.

madisongh, 2024-06-26:
I've fixed this in master and scarthgap, and have updated the scarthgap branch in the demo distro with this change.

closed 2024-06-26 by madisongh

aurelien-enchanted-tools, 2024-07-08:
Hello, I obtain the same error after this merge.

branch scarthgap 24503b40b6671cffcb5806ec68972ef77e5a798c

```
| [   0.1050 ] partition type bootloader_stage2, magic id = CPBL
| [   0.1286 ] tegrahost_v2 --chip 0x23 0 --align rcmboot_uefi_jetson_with_dtb_aligned.bin
| [   0.1768 ] tegrahost_v2 --chip 0x23 0 --magicid CPBL --appendsigheader rcmboot_uefi_jetson_with_dtb_aligned.bin zerosbk
| [   0.1966 ] adding BCH for rcmboot_uefi_jetson_with_dtb_aligned.bin
| [   0.4843 ] tegrasign_v3.py --key None --list rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin_list.xml --pubkeyhash pub_key.key --sha sha512
| [   0.4846 ] Assuming zero filled SBK key
| [   0.5088 ] Warning: pub_key.key is not found
| [   0.5326 ] tegrahost_v2 --chip 0x23 0 --updatesigheader rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.encrypt rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.hash zerosbk
| [   0.5687 ] Copying rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt to /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload
| [   0.5688 ] Signed file: /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload/rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt
| Signing with zerosbk ...
| rollback info is parsed and saved in rollback.bin
| WARNING: failed to find rollback config for " generic "
| pad ZEROs to the end of rollback.bin
|
| signing images succeeded
|
| The index file /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload/signed/flash.idx is not found.
| WARNING: /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.313285:254 exit 1 from './doflash.sh'
| WARNING: Backtrace (BB generated script):
|         #1: tegraflash_custom_sign_bup, /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.313285, line 254
|         #2: do_compile, /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.313285, line 160
|         #3: main, /home/yoctouser/build_scarthgap/tmp/work/custom_orin-poky-linux/tegra-uefi-capsules/36.3.0/temp/run.do_compile.313285, line 426
NOTE: recipe tegra-uefi-capsules-36.3.0-r0: task do_compile: Failed
ERROR: Task (/home/yoctouser/src/poky/../meta-tegra/recipes-bsp/uefi/tegra-uefi-capsules_36.3.0.bb:do_compile) failed with exit code '1'
```

PR #1598 (2024-06-26): tegra-uefi-capsules: add missing dependencies and setup
* Add missing dependency on virtual/bootloader:do_deploy, needed for all platforms
* Add missing dependency, and staging, for the ESP image, needed for the AGX Orin platform using internal storage, to satisfy the NVIDIA flashing tools while generating the BUP/capsule contents.

Fixes #1597