OE4T/meta-tegra #1601: Question: Are there any plans for updating kirkstone-l4t-r32.7.x to the latest L4T32.7.5?

graugans, 2024-06-28:
Nvidia just have released the [Jetson Linux R32.7.5](https://developer.nvidia.com/embedded/linux-tegra-r3275) I have checked the [Nvidia git](https://nv-tegra.nvidia.com/r/gitweb?p=linux-nvidia.git;a=summary) repo to see if there was a kernel update. I was not able to see if there is a new branch `l4t-r32.7.5` 

I just wanted to ask if there is maybe already some progress in this regards, if not how would one start to perform the transition?

madisongh, 2024-06-28:
I just saw the announcement myself a couple of hours ago, although I knew it would be coming soon due to it being mentioned in a recent PCN. So yes, there are plans, but no, there's been no progress yet, because there's been nothing there to make progress with. :smiley:

madisongh, 2024-06-30:
Done

closed 2024-06-30 by madisongh