OE4T/meta-tegra #1606: Wifi hiccups with tegra-kernel oe4t-patches-l4t-r32.7.4

metalMajor, 2024-07-05:
**Describe the bug**
I am using the jetson nano and for the kernel we refer to the linux tegra kernel branch oe4t-patches-l4t-r32.7.3 and for the wifi we use the cypress 43455 wifi driver and it works great. But we did the update to oe4t-patches-l4t-r32.7.4 and then we get hiccups in the wifi connection. It works, but there are some short hangs now and then, it's not fluent anymore. It really is a huge difference. Is there any commit that could be the culprit for this? Also the latest kernel branch oe4t-patches-l4t-r32.7.5 has the issue. I also tried the latest cypress driver and it does not make a difference, so the kernel branch makes the difference.

Thanks for any pointers,
Best regards

**To Reproduce**
Steps to reproduce the behavior:
1. Use tegra-kernel branch oe4t-patches-l4t-r32.7.4
2. Check wifi is having hiccups.

dwalkes, 2024-07-07:
@metalMajor have you attempted to reproduce with the L4T release as delivered by NVIDIA an a Nano devboard to see if it's related to OE4T patches or some differences in the kernel as delivered by NIVIDA between these releases?

Looking at https://github.com/OE4T/linux-tegra-4.9/commits/oe4t-patches-l4t-r32.7.4 and https://github.com/OE4T/linux-tegra-4.9/commits/oe4t-patches-l4t-r32.7.3 it looks like there are only two different patches applied, both are related to https://github.com/OE4T/linux-tegra-4.9/pull/43 so if you remove those two I think the only remaining difference would be a change from the nvidia provided kernel or something about the way previous patches from 32.7.3 applied to 32.7.4.

ichergui, 2025-01-26:
Hi @metalMajor 
Any update on this issue ?

metalMajor, 2025-02-04:
Sorry, I do not have the time to test this with a devboard.

ichergui, 2025-02-04:
I will close this ticket since no updates. Don't hesitate to open new issue.

closed 2025-02-04 by ichergui