OE4T/meta-tegra #1628: NVS compilation error, kirkstone

intera91, 2024-07-18:
/kernel-source/nvidia/drivers/misc/nvs/nvs_timestamp.c:25:16: error: implicit d$
claration of function 'timespec_to_ns'; did you mean 'timespec64_to_ns'? [-Werror=implicit-function-declaration]
|    25 |         return timespec_to_ns(&ts);
|       |                ^~~~~~~~~~~~~~
|       |                timespec64_to_ns
| cc1: all warnings being treated as errors

ichergui, 2024-07-18:
Hi @intera91 
Could you please share the steps you followed until having the issue ?

intera91, 2024-07-18:
Just a simple bitbake virtual/kernel

ichergui, 2024-07-18:
I will take a look and will get back to you

madisongh, 2024-07-18:
It would help to know which branch you're using, but I'm guessing it's one of the L4T R35.x-based branches with the 5.10 kernel...

The old 32-bit timespec structure, and related functions, were removed in (I think) the 5.6 kernel - part of fixing the Y2038 problem in the kernel. I suspect that NVIDIA just hasn't updated that driver for 5.10, since it's not part of the stock kernel configuration for L4T. Try patching the driver to use `timespec64_to_ns` instead; since we're a 64-bit architecture, timespecs even pre-5.6 should have been mapped into timespec64s anyway.

closed 2024-10-21 by madisongh