OE4T/meta-tegra #1639: Issues Flashing to Jetson Orin nano devkit with secure boot

imoyakin, 2024-07-25:
**Describe the bug**
we are trying to build a custom image for some scenes. After enabling secure boot according to the instructions in the wiki, flashing the device started to cause problems that I couldn't understand.

what did i do:

1.  modify the  `local.conf`,add the config with   `TEGRA_SIGNING_ARGS`, `TEGRA_UEFI_DB_KEY` and `TEGRA_UEFI_DB_CERT`
2. create custom layer, add `recipies-bsp/uefi` and ours app recipe
3. after setup and bitbake , i tar the file `core-image-base-jetson-orin-nano-devkit-custom.tegraflash.tar.gz` and try to flash (use `doflash.sh`)

there is log:
```
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0210 ] Parsing partition layout
[   0.0217 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0236 ] Parsing partition layout
[   0.0241 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0250 ] Boot Rom communication
[   0.0255 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0260 ] BR_CID: ==== has been hiden ====
[   7.9000 ] Sending bct_br
[  16.8955 ] Sending mb1
[  16.8960 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'kirkstone' latest with `jetson orin nano` based on 'tegra234-p3767-0003-p3768-0000'
3. Build with bitbake argument 'core-image-base'
5. Deploy to hardware with method './doflash'
6. See error '...' with link to any relevant logs at '...'

**Additional context**

after use buildtime encrypt, i noticed doflash.sh use ./tegraflash.py with many arguments, not normal condition use ./tegra234-flash-helper.sh

ichergui, 2024-07-26:
Hi @imoyakin 
I assume that you are using `kirkstone-l4t-r32.7.x` branch, right ? 
Is this error happens systematically ? Did you check your USB cable connection ? 
Could you please be more specific about `TEGRA_SIGNING_ARGS` ? Are you using `PKC` key only or both `PKC` + `SBK` keys ?

imoyakin, 2024-07-29:
> I assume that you are using `kirkstone-l4t-r32.7.x` branch, right ?
i'm using the branch 'kirkstone', which current commit is `ce93df05a69792f7971d8e09722cc51c2d9f92b9 `
> Is this error happens systematically ? Did you check your USB cable connection ?
Yes, this error occurs consistently. i belive the USB connection is reliable, because i can flash the system with signing with `TEGRA_SIGNING_ARGS`.
> Could you please be more specific about `TEGRA_SIGNING_ARGS` ? Are you using `PKC` key only or both `PKC` + `SBK` keys ?
my local.conf is 
```
TEGRA_SIGNING_ARGS = "-u /home/ubuntu/Documents/encrypt/PKC.pem -v /home/ubuntu/Documents/encrypt/SBK.key"
```

ichergui, 2024-07-31:
HI @imoyakin 
Could you please share the full logs when flashing your device ?

ichergui, 2024-07-31:
> > I assume that you are using `kirkstone-l4t-r32.7.x` branch, right ?
> 
> i'm using the branch 'kirkstone', which current commit is `ce93df05a69792f7971d8e09722cc51c2d9f92b9 `
> 
The commit hash doesn't exist. Are you sure about it ?
> > Is this error happens systematically ? Did you check your USB cable connection ?
> 
> Yes, this error occurs consistently. i belive the USB connection is reliable, because i can flash the system with signing with `TEGRA_SIGNING_ARGS`.
> 
Are you able to flash other Orin Nano devkit ? 
> > Could you please be more specific about `TEGRA_SIGNING_ARGS` ? Are you using `PKC` key only or both `PKC` + `SBK` keys ?
> > my local.conf add config is
> 
> ```
> TEGRA_SIGNING_ARGS = "-u /home/ubuntu/Documents/encrypt/PKC.pem -v /home/ubuntu/Documents/encrypt/SBK.key"
> ```
> 
> i use both `PKC` + `SBK` keys

imoyakin, 2024-07-31:
These are the full logs when flashing the device. The UART has nothing.

Sorry, I posted the issue to the wrong repo. It should be [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) branch. Other dev boards still do not work. I'm trying to use the NVIDIA official tool to flash an encrypted system. I think I need to understand what has been signed and encrypted. Should I use odmflash.sh first?

Islam-Hussein-11, 2024-08-07:
I have the almost the same problem. It flashes then UART has nothing.

Islam-Hussein-11, 2024-08-08:
The problem supposed to be in carrier board from custom manfacturer. I've tried to flash using nvidia carrier and it works with me now. If you're using custom carrier I recommend you to swap it with nvidia carrier for fusing and then you can get back to you custom for image burning

ichergui, 2024-08-08:
Thanks @Islam-Hussein-11 for the update

imoyakin, 2024-08-13:
I fixed this issue. 
Finally, I guess there might be an error in internal storage,such as cpu_boot or l4t.
After use nvidia official script `./tools/kernel_flash/l4t_initrd_flash.sh` to flash the board internal, this problem has been fixed.

closed 2024-08-13 by imoyakin

Ray9090, 2026-02-27:
@imoyakin Did you flash with the key using only initrd-flash.sh?