OE4T/meta-tegra #1656: nvluks-srv-app does not work as expected (36.3.0 scarthgap)

liuming50, 2024-08-14:
1 Build based on commit:

```
commit 5c23a93f2f5c7eeb9df643842a8b638b39e20718 (HEAD -> scarthgap, oe4t/scarthgap)
Author: Sid-Ali Teir <sidali.teir@cysec.com>
Date:   Thu Aug 8 12:58:10 2024 +0200

    nvidia-l4t-optee: install missing tee-supplicant script
    
    Signed-off-by: Sid-Ali Teir <sidali.teir@wavelit.io>
```

2 test image installed with the following packages:

```
IMAGE_INSTALL:append = " \
+    optee-nvsamples-luks-srv \
+    optee-nvsamples-hwkey-agent \
+    optee-nvsamples \
+"
```

3 tegraflash tarball flashed correctly, after image boot, run the following command:

```
$ sudo nvluks-srv-app --context-string 6533b3ab-d082-4f5f-889d-66fae834a198 --get-unique-pass
```

outputted error:

```
TEEC_InvokeCommand failed 0xffff0008 origin 0x4
```

the same command worked on 35.3.0 kirkstone

ichergui, 2024-08-14:
Hi @liuming50 
Are you using the right EKB image ?

liuming50, 2024-08-21:
@ichergui Hi, good afternoon! May I ask any update on this issue? Could you reproduce this issue, or it's just me seeing this error?

liuming50, 2024-08-22:
@ichergui thanks for the reply!

May I ask is the eks.img somehow related to serial number read from eeprom? Our board does not have eeprom, and since serial number is mandatory now in ./initrd-flash of scarthgap, so we did change that part with the old brcid-to-uid script:

```
$ git diff recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh
diff --git a/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh b/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh
index 3f1fb5d7..9baed501 100644
--- a/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh
+++ b/recipes-bsp/tegra-binaries/tegra-helper-scripts/initrd-flash.sh
@@ -246,7 +246,7 @@ wait_for_usb_storage() {
        for candidate in /dev/sd[a-z]; do
            [ -b "$candidate" ] || continue
            cand_model=$(udevadm info --query=property $candidate | grep '^ID_MODEL=' | cut -d= -f2)
-           if [ "$cand_model" = "$sessid" ]; then
+           if [ -n "$cand_model" ]; then
                cand_vendor=$(udevadm info --query=property $candidate | grep '^ID_VENDOR=' | cut -d= -f2)
                if [ "$cand_vendor" = "$name" ]; then
                    echo "[$candidate]" >&2
@@ -473,8 +473,12 @@ fi
 [ ! -f ./boardvars.sh ] || . ./boardvars.sh
 
 if [ -z "$serial_number" ]; then
-    echo "ERR: did not get device serial number at $(date -Is)" | tee -a "$logfile"
-    exit 1
+    # Generate session id in the old way of JP5, since it may not read serial number
+    # on some boards provided that they dont have eeprom.
+    session_id=$("$here/brcid-to-uid" $BR_CID)
+    session_id=$(echo -n "$session_id" | tail -c8)
+else
+    session_id=$(printf "%x" "$serial_number" | tail -c8)
 fi
 
 session_id=$(printf "%x" "$serial_number" | tail -c8)
```

so even when "$cand_model" = "$sessid", we still do the installation, would that impact the optee?

liuming50, 2024-08-22:
Thanks for the information about https://matrix.to/#/#nv-jetson-secureboot:gitter.im