OE4T/meta-tegra #1662: Building `cuda-toolkit` for the native SDK fails with an QA issue

faro-dr, 2024-08-15:
**Describe the bug**
Building `cuda-toolkit` for the native SDK fails with an QA issue: `nativesdk-cuda-sanitizer: Files/directories were installed but not shipped in any package`

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch `scarthgap`, SHA `caf164e4` with `MACHINE` based on ''p3768-0000-p3767-0001". Added `nativesdk-cuda-toolkit` to `TOOLCHAIN_HOST_TASK` of the image.
2. Build with bitbake argument `-c populate_sdk`
3. Not deployed to hardware (trying to build SDK)
4. Error:

```
ERROR: nativesdk-cuda-sanitizer-12.2.140-1-r0 do_package: QA Issue: nativesdk-cuda-sanitizer: Files/directories were installed but not shipped in any package:
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/TreeLauncherTargetLdPreloadHelper
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/libTreeLauncherPlaceholder.so
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/libInterceptorInjectionTarget.so
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/TreeLauncherSubreaper
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/libTreeLauncherTargetInjection.so
  /opt/scanix-magpie/nodistro.0/sysroots/x86_64-mysdk-linux/usr/local/cuda-12.2/compute-sanitizer/x86/libTreeLauncherTargetUpdatePreloadInjection.so
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-cuda-sanitizer: 7 installed and not shipped files. [installed-vs-shipped]
ERROR: nativesdk-cuda-sanitizer-12.2.140-1-r0 do_package: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /data/ELS/els-workspace/yocto-els/work/build/tmp-glibc/work/x86_64-nativesdk-mysdk-linux/nativesdk-cuda-sanitizer/12.2.140-1/temp/log.do_package.701143
ERROR: Task (virtual:nativesdk:/data/ELS/els-workspace/yocto-els/work/sources/poky/../meta-tegra/recipes-devtools/cuda/cuda-sanitizer_12.2.140-1.bb:do_package) failed with exit code '1'
``` 

**Additional context**
I could fix the problem with this diff:
```
diff --git a/recipes-devtools/cuda/cuda-shared-binaries.inc b/recipes-devtools/cuda/cuda-shared-binaries.inc
index 4ea64dbe..4ef0ae38 100644
--- a/recipes-devtools/cuda/cuda-shared-binaries.inc
+++ b/recipes-devtools/cuda/cuda-shared-binaries.inc
@@ -81,6 +81,7 @@ do_install() {
     rm -f ${D}${prefix}/local/cuda-${CUDA_VERSION}/LICENSE ${D}${prefix}/local/cuda-${CUDA_VERSION}/README ${D}${prefix}/local/cuda-${CUDA_VERSION}/version.txt
     rm -f ${D}${prefix}/local/cuda-${CUDA_VERSION}/EULA.txt
     rm -f ${D}${prefix}/local/cuda-${CUDA_VERSION}/DOCS
+    rm -rf ${D}${prefix}/local/cuda-${CUDA_VERSION}/compute-sanitizer/x86
 }
``` 
I delete this dir because it contains files that are also present in the parent dir. These are not exact copies but all are x84-64 files:

```
$ md5sum lib* x86/lib*
41c5f8941d12be34e0498f9c56c6a8a7  libInterceptorInjectionTarget.so
da803c0240684e8bdfeda714dba25ffe  libTreeLauncherPlaceholder.so
28f2ad51fe4a4411eb2b2de4987b47a1  libTreeLauncherTargetInjection.so
329517545df6124987d54595f46de1bd  libTreeLauncherTargetUpdatePreloadInjection.so
b1a58a4e897931a6268c71c8c4d5ec41  libsanitizer-collection.so
225d5e04abdb458a2fc87ad1d9b13e66  libsanitizer-public.so
bf722d4ad7c1c51903894e452891abb4  x86/libInterceptorInjectionTarget.so
a4237c885d1a323d6a4de796a07cdab1  x86/libTreeLauncherPlaceholder.so
bf476b8ca138e739e0b967620e797635  x86/libTreeLauncherTargetInjection.so
6fddb81d7058219768cf9322ec525fd2  x86/libTreeLauncherTargetUpdatePreloadInjection.so
```

madisongh, 2024-08-18:
The added files are 32-bit x86 binaries, so removing them did look like the better approach.

faro-dr, 2024-08-21:
@madisongh I didn't notice, thank you

PR #1665 (2024-08-18): cuda-sanitizer: remove 32-bit x86 binaries
CUDA 12 added 32-bit x86 tools and libraries to the cuda-sanitizer package. They're not likely needed in our builds or SDKs, so just remove them to avoid packaging issues.

Fixes #1662