OE4T/meta-tegra #1672: [question][kirkstone-] Are the recipes for Gstreamer 1.14.x still required when using tegra-container-passthrough

graugans, 2024-08-23:
A quick question, when using the commit https://github.com/OE4T/meta-tegra/commit/cffdf0bf6ef9e2cddb26547608627a8329e1a9cf with the package tegra-container-passthrough is it still neccessary to use the gstreamer version 1.14 from the contrib layer?

BTW, I stumbled across the ` tegra-container-passthrough` while checking [container-runtime-csv.bbclass](https://github.com/OE4T/meta-tegra/blob/kirkstone-l4t-r32.7.x/classes/container-runtime-csv.bbclass#L53C34-L53C79) where would be a good place to document this? I guess [contrib/README.md](https://github.com/OE4T/meta-tegra/blob/kirkstone-l4t-r32.7.x/contrib/README.md) is a good place for this?

madisongh, 2024-08-25:
Right, with the container passthrough stuff in place, you should be able to drop the gstreamer 1.14 recipes. That readme file is probably as good a place as any to mention it - feel free to send a PR.

madisongh, 2025-10-17:
I've updated the note about the gstreamer 1.14 recipes in that README file.

closed 2025-10-17 by madisongh