OE4T/meta-tegra #1674: Issues flashing Jetson Orin nano devkit with TEGRA_SIGNING_ARGS

wavesid, 2024-08-23:
Hello

**Describe the bug**
I have the exact same setup as this issue: https://github.com/OE4T/meta-tegra/issues/1639
I am using Jetson Orin Nano 8GB devkit, with these options: `TEGRA_SIGNING_ARGS`, `TEGRA_UEFI_DB_KEY` and `TEGRA_UEFI_DB_CERT` set to the following

```
TEGRA_SIGNING_ARGS = "-u PKC.pem -v SBK.key"
TEGRA_UEFI_DB_KEY = "db.key"
TEGRA_UEFI_DB_CERT = "db.crt"
```

This is the logs:

using `sudo ./doflash.sh`

```
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0100 ] Parsing partition layout
[   0.0105 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0119 ] Parsing partition layout
[   0.0122 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0134 ] mb1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --mb1_bin
[   0.0134 ] psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --psc_bl1_bin
[   0.0134 ] Boot Rom communication
[   0.0137 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0141 ] BR_CID: 0x80012344705DF11F2400000013028100
[   0.0405 ] Sending bct_br
[   0.0800 ] Sending mb1
[   0.0807 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
```

using `sudo ./initrd-flash`

```
Starting at 2024-08-23T18:03:47+02:00
Machine:       jetson-orin-nano-devkit-nvme
Rootfs device: nvme0n1p1
Found Jetson device in recovery mode at USB 1-1
== Step 1: Signing binaries at 2024-08-23T18:03:47+02:00 ==
== Step 2: Boot Jetson via RCM at 2024-08-23T18:03:48+02:00 ==
Found Jetson device in recovery mode at USB 1-1
./initrd-flash: line 191: ./rcm-boot.sh: No such file or directory
ERR: RCM boot failed at 2024-08-23T18:03:48+02:00
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'scarthgap' (latest commit) with `MACHINE` based on 'jetson-orin-nano-devkit-nvme'
2. Build with bitbake image
3. Deploy to hardware with method tegraflash using `sudo ./doflash.sh` (or using `sudo ./initrd-flash`)
4. See logs above

**Additional context**
If I set ONLY `TEGRA_UEFI_DB_KEY` and `TEGRA_UEFI_DB_CERT`, it works without issues.
I checked USB connection, using `PKC + SBK` keys, I do not have logs in UART

Let me know if there is any way to debug

madisongh, 2024-08-24:
Did you actually burn the fuses on the target device with the SBK/PKC keys you're using here? It sounds like you did not, since you say it works if you omit them.

madisongh, 2024-08-26:
Unfortunately, the way the Jetsons work, you cannot flash binaries signed with an actual SBK/PKC to a module that has not had its fuses programmed with the same keys. The RCM protocol it uses during flashing is secured using those keys, so communication won't be established if there's a mismatch.

The output from initrd-flash is intentionally brief; see the log file it creates for more detail. You'll likely find that the rcm-boot.sh script could not be generated due to the communication issue.

wavesid, 2024-08-26:
Yes I will retry and keep you updated, thanks for your help!

closed 2024-08-26 by wavesid

wavesid, 2024-09-17:
@ichergui I have the same issue even with burnt fuse. I verified the keys and fusing that happens successfully.

using `sudo ./doflash.sh`
```
$ sudo ./doflash.sh 
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0258 ] Parsing partition layout
[   0.0263 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0280 ] Parsing partition layout
[   0.0285 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0295 ] mb1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --mb1_bin
[   0.0295 ] psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --psc_bl1_bin
[   0.0295 ] Boot Rom communication
[   0.0300 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0303 ] BR_CID: 0xEA012344705DF11F2400000013028100
[   0.2528 ] Sending bct_br
[   0.2529 ] Sending mb1
[   0.2545 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
```

using `sudo ./initrd-flash -u pkc.key -v sbk.key`
```
$ sudo ./initrd-flash -u pkc.key -v sbk.key
WARN: binaries already signed; ignoring signing options
Starting at 2024-09-17T17:51:54+02:00
Machine:       jetson-orin-nano-devkit-nvme
Rootfs device: nvme0n1p1
Found Jetson device in recovery mode at USB 1-1
== Step 1: Signing binaries at 2024-09-17T17:51:54+02:00 ==
== Step 2: Boot Jetson via RCM at 2024-09-17T17:51:55+02:00 ==
Found Jetson device in recovery mode at USB 1-1
./initrd-flash: line 191: ./rcm-boot.sh: No such file or directory
ERR: RCM boot failed at 2024-09-17T17:51:55+02:00
```