OE4T/meta-tegra #1682: scarthgap-l4t-r35.x kas build errors

tjhgit, 2024-08-30:
**Describe the bug**
BUG Nr1
l4t_version.bbclass not found

BUG Nr2
| install: cannot create regular file '/home/yocto/git/ext/meta-mender-community/build/tmp/work/jetson_agx_xavier_devkit-oe4t-linux/tegra-storage-layout-base/35.5.0/image/usr/share/tegraflash/flash_l4t_nvme_rootfs_ab.xml': No such file or directory
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/yocto/git/ext/meta-mender-community/build/../meta-tegra/recipes-bsp/tegra-binaries/tegra-storage-layout-base_35.5.0.bb:do_install) failed with exit code '1'

**To Reproduce**
```bash
git clone https://github.com/mendersoftware/meta-mender-community.git
git checkout scarthgap
kas build kas/jetson-agx-xavier-devkit.yml
```

**Additional context**

In order to fix the above bugs I think these changes need to be made:

**Bug Nr1**
in `kas/include/tegra-jetpack5.yml`
```
repos:
  meta-tegra:
    # refers to the scarthgap-l4t-r35.x branch
    # commit: 972488415a80afebbd652a983a46c2641eed99ad
    commit: 32d86844d857321e1ed137abbdd5a045c6f5cb3f # NOTE !!! (this commit has the global l4t_version)
```

and **Bug Nr2**
`meta-tegra/recipes-bsp/tegra-binaries/tegra-storage-layout-base_35.5.0.bb`
```
do_install() {
    install -d ${D}${datadir}/l4t-storage-layout
    install -d ${D}${datadir}/tegraflash # NOTE
    install -m 0644 ${PARTITION_FILE} ${D}${datadir}/l4t-storage-layout/${PARTITION_LAYOUT_TEMPLATE}
    install -m 0644 ${PARTITION_FILE} ${D}${datadir}/tegraflash/${PARTITION_LAYOUT_TEMPLATE} # NOTE
    [ -z "${PARTITION_LAYOUT_EXTERNAL}" ] || install -m 0644 ${PARTITION_FILE_EXTERNAL} ${D}${datadir}/tegraflash/${PARTITION_LAYOUT_EXTERNAL} # NOTE
    [ -z "${PARTITION_LAYOUT_EXTERNAL}" ] || install -m 0644 ${PARTITION_FILE_EXTERNAL} ${D}${datadir}/l4t-storage-layout/${PARTITION_LAYOUT_EXTERNAL}
}
```

Additionally there is a type in one of the PARTITION_LAYOUT_EXTERNAL used in the original code.

Please confirm.

### Many many thanks for building the meta-mender version for L4T 35.5 !!!!!

dwalkes, 2024-08-30:
Hi @tjhgit 
Please start a new PR or issue for this on https://github.com/mendersoftware/meta-mender-community since the meta-tegra repository is for the BSP layer only and these are specific to the mender integration.

closed 2024-08-30 by dwalkes

tjhgit, 2024-08-30:
@dwalkes but these are bugs in meta-tegra and not in mender. See needed correction in `meta-tegra/recipes-bsp/tegra-binaries/tegra-storage-layout-base_35.5.0.bb`

kekiefer, 2024-08-30:
I think we're just having trouble understanding how to help you, but if you've identified an issue (or multiple issues) that need to be fixed here in meta-tegra, then by all means please open a pull request (or multiple pull requests) so it can be fixed up.

dwalkes, 2024-08-30:
I see you've opened https://github.com/mendersoftware/meta-mender-community/pull/403 for one of these.
I'm not able to understand the second issue, but as @kekiefer mentioned, please open a PR with the change you are proposing to fix.

tjhgit, 2024-08-30:
Ok I did create a pull request.