OE4T/meta-tegra #1688: [question]how to disable serial debug logging on orin nano

AlistairKeiller, 2024-09-03:
I would like to use the UART port /dev/ttyTHS0 as a normal serial output to other UART devices.

Those devices are currently acting strangely, which is making me think that serial logging is enabled. Serial logging on Jetson should be enabled with the kernel flag console=ttyS0.

I'm not sure what would be enabling console=ttyS0, or how I could disable it.

Any advice for how to disable serial logging?

dwalkes, 2024-09-03:
Possibly related discussion at https://github.com/OE4T/meta-tegra/issues/84

madisongh, 2024-09-03:
There are some threads on the NVIDIA dev forum about this [here](https://forums.developer.nvidia.com/t/jetson-orin-nano-developer-board-did-not-see-outputs-from-uart1-dev-ttyths0/295276) and [here](https://forums.developer.nvidia.com/t/can-orin-nano-8gb-debug-port-be-a-normal-uart-port/259361).

Kernel console is normally routed to that UART via `/dev/ttyTCU0` on the Orins. If I'm understanding correctly from the forum threads, you tweak the boot configuration to switch that serial interface be "normal" instead of "debug", and that will route the UART through `/dev/ttyTHS0` instead, so you can use it for other things.

AlistairKeiller, 2024-09-08:
Thank you for the advice!! I am kinda in too much of a rush rn to be messing around with devicetree, but my question is resolved as there it will clearly work eventual with enough fiddling.

For future reference to others running into the issue, I used the adafruit USB to TTL Serial Cable to get a clean UART from /dev/ttyUSB0, which worked flawlessly in yocto by adding the kernel-module-cp210x package in IMAGE_INSTALL:append. I don't know if this is the best solution: it takes up a USB port, and I am having some issues with signal stability when running a 1k resistor from TX to RX (for TMC2209), but it worked first try and I haven't run into any issues yet.

closed 2024-09-08 by AlistairKeiller